realsenseai/librealsense #1424: Installing on Jetson Tx2

GustavZ, 2018-03-23:
I am trying to install librealsense for the **D435 on the nvidia jetson tx2 with ubuntu 16.04**
Therefore i am following the ubuntu 16.04 installation guide. https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
At one point it is said for arch architectures to do the following:
`You need to install the `base-devel `package group.`
But if you click on the link for base-devel there are plenty of different files and none of them is for aarch64.
So what is needed here?

when i skip this step, then `./patch-arch.sh` fails with this message:
```
Extract the kernel
RealSense patch...
patching file drivers/media/usb/uvc/Makefile
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 168 with fuzz 2 (offset 20 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 115 with fuzz 2 (offset -4 lines).
patching file include/uapi/linux/videodev2.h
Hunk #1 FAILED at 621.
1 out of 1 hunk FAILED -- saving rejects to file include/uapi/linux/videodev2.h.rej
```
and ofcourse after that cmake fails with:
```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GLFW_INCLUDE_DIR
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/software-device
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/software-device
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/software-device
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/software-device
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/software-device
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/software-device
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/software-device
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/capture
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/capture
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/capture
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/capture
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/capture
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/capture
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/capture
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/save-to-disk
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/save-to-disk
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/save-to-disk
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/save-to-disk
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/save-to-disk
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/save-to-disk
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/save-to-disk
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/multicam
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/multicam
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/multicam
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/multicam
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/multicam
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/multicam
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/multicam
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/pointcloud
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/pointcloud
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/pointcloud
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/pointcloud
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/pointcloud
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/pointcloud
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/pointcloud
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/align
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/align
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/align
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/align
[... 99 lines cut ...]
   used as include directory in directory /home/nvidia/workspace/intel_realsense/librealsense/tools/rosbag-inspector
GLFW_LIBRARIES
    linked by target "rs-software-device" in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/software-device
    linked by target "rs-capture" in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/capture
    linked by target "rs-save-to-disk" in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/save-to-disk
    linked by target "rs-multicam" in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/multicam
    linked by target "rs-pointcloud" in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/pointcloud
    linked by target "rs-align" in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/align
    linked by target "rs-sensor-control" in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/sensor-control
    linked by target "rs-measure" in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/measure
    linked by target "rs-depth" in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/C/depth
    linked by target "rs-color" in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/C/color
    linked by target "rs-distance" in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/C/distance
    linked by target "rs-post-processing" in directory /home/nvidia/workspace/intel_realsense/librealsense/examples/post-processing
    linked by target "rs-terminal" in directory /home/nvidia/workspace/intel_realsense/librealsense/tools/terminal
    linked by target "rs-fw-logger" in directory /home/nvidia/workspace/intel_realsense/librealsense/tools/fw-logger
    linked by target "rs-enumerate-devices" in directory /home/nvidia/workspace/intel_realsense/librealsense/tools/enumerate-devices
    linked by target "realsense-viewer" in directory /home/nvidia/workspace/intel_realsense/librealsense/tools/realsense-viewer
    linked by target "rs-data-collect" in directory /home/nvidia/workspace/intel_realsense/librealsense/tools/data-collect
    linked by target "rs-depth-quality" in directory /home/nvidia/workspace/intel_realsense/librealsense/tools/depth-quality
    linked by target "rs-rosbag-inspector" in directory /home/nvidia/workspace/intel_realsense/librealsense/tools/rosbag-inspector

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/workspace/intel_realsense/librealsense/build/CMakeFiles/CMakeOutput.log".
```

eleboss, 2018-03-23:
check that https://github.com/IntelRealSense/librealsense/blob/development/doc/installation_jetson.md

GustavZ, 2018-03-23:
@eleboss  i did the modifications but this guide should be updated as the referenced lines are not correct anymore.
This guide does not help with the current version of skripts.

arybalkin, 2018-03-23:
Gustav, I did compile the latest librealsense2 (2.10.2) from git repo on TX2 a few days ago and it did compile cleanly. I'm using L4T 28.1 kernel sources, here's the librealsense2 patches I did apply to my kernel tree:

`root@tx2:/usr/src/kernel-28.1/kernel-4.4# patch -p1 < realsense-camera-formats_ubuntu-xenial-master.patch
patching file drivers/media/usb/uvc/Makefile
patching file drivers/media/usb/uvc/uvc_driver.c
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 122 (offset 7 lines).
Hunk #2 succeeded at 137 (offset 7 lines).
patching file drivers/media/v4l2-core/v4l2-ioctl.c
Hunk #1 succeeded at 1236 (offset 7 lines).
patching file include/uapi/linux/videodev2.h
Hunk #1 succeeded at 632 (offset 8 lines).
root@tx2:/usr/src/kernel-28.1/kernel-4.4# 
root@tx2:/usr/src/kernel-28.1/kernel-4.4# 
root@tx2:/usr/src/kernel-28.1/kernel-4.4# patch -p1 < realsense-metadata-ubuntu-xenial-master.patch
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 2624 with fuzz 2 (offset -75 lines).
patching file drivers/media/usb/uvc/uvc_video.c
Hunk #1 succeeded at 1234 (offset 6 lines).
Hunk #2 succeeded at 1272 (offset 6 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 189 (offset 34 lines).
Hunk #2 succeeded at 209 (offset 34 lines).
root@tx2:/usr/src/kernel-28.1/kernel-4.4# patch -p1 < realsense-hid-ubuntu-xenial-master.patch
patching file drivers/iio/accel/hid-sensor-accel-3d.c
patching file drivers/iio/gyro/hid-sensor-gyro-3d.c
patching file include/linux/hid-sensor-ids.h
root@tx2:/usr/src/kernel-28.1/kernel-4.4# patch -p1 < realsense-powerlinefrequency-control-fix.patchpatching file drivers/media/usb/uvc/uvc_ctrl.c
Hunk #1 succeeded at 356 (offset -1 lines).
root@tx2:/usr/src/kernel-28.1/kernel-4.4# 
root@tx2:/usr/src/kernel-28.1/kernel-4.4#`

I used the following options to compile librealsense2:

`#!/bin/bash

cd build

cmake ..  -DCMAKE_BUILD_TYPE=Release -DBUILD_WITH_OPENMP=OFF -DBUILD_EXAMPLES=ON -DBUILD_GRAPHICAL_EXAMPLES=ON -DBUILD_PYTHON_BINDINGS=ON -DBUILD_CV_EXAMPLES=ON  

# -DBUILD_WITH_STATIC_CRT=OFF -DENABLE_ZERO_COPY=ON -DBUILD_EASYLOGGINGPP=OFF 
  
make clean && make -j4
`

-albertr

RealSense-Customer-Engineering, 2018-03-25:
[Realsense Customer Engineering Team Comment] 
Hi @GustavZ,

You can refer to below patch for Jetson TX2 and the latest BSP (L4T 28.2) used. It should be ok for SR300 and D415/D435 camera.
https://github.com/freemanlo/librealsense/blob/master/scripts/patch-realsense-ubuntu-xenial-jetson-tx2.sh

eleboss, 2018-03-25:
@RealSense-Customer-Engineering That looks great! I will check that immediately.

GustavZ, 2018-03-25:
Thank you!

And do have an idea where my 3Gb of disk space could have gone?
Because the librealsense directory is still just a few 100mbs.

Where could be the data junks?

eleboss, 2018-03-25:
@GustavZ  the buildJetsonTX2kernel will build a kernel under /usr/src maybe that is your 3GB space

lesreaper, 2018-03-25:
Is this using CUDA? I've seen videos like this:

https://www.youtube.com/watch?v=t9nOUVw2VzA

and the performance of the D435 is pretty awful on a TX2. It was the same for the SR300 I found as well, but had hoped it improved. 

Thoughts on improving performance?

eleboss, 2018-03-26:
@RealSense-Customer-Engineering 
I flash my TX2 and using the patch you provided. Install librealsense successfully but when I use ros warpper, I meet these errors. 
Do you have and ideas why？

ubuntu@tegra-ubuntu:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch 
... logging to /home/ubuntu/.ros/log/ab1a2a38-3090-11e8-b3fd-00044ba7d970/roslaunch-tegra-ubuntu-19827.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:43313/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/camera_aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/serial_no: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [19837]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ab1a2a38-3090-11e8-b3fd-00044ba7d970
process[rosout-1]: started with pid [19850]
started core service [/rosout]
[... 14 lines cut ...]
[ INFO] [1522025854.978723059]: Device Serial No: 801212070377
[ INFO] [1522025854.978780436]: Device FW version: 05.08.15.00
[ INFO] [1522025854.978840468]: Device Product ID: 0x0AD6
[ INFO] [1522025854.978896981]: Enable PointCloud: Off
[ INFO] [1522025854.978955158]: Align Depth: Off
[ INFO] [1522025854.979007159]: Sync Mode: Off
[ INFO] [1522025854.979764098]: Device Sensors: 
[ INFO] [1522025854.980343146]: Stereo Module was found.
[ INFO] [1522025854.980967667]: (Color, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1522025854.981057172]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1522025854.981124181]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1522025854.981184278]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1522025854.981337016]: setupPublishers...
[ INFO] [1522025855.012391825]: setupStreams...
 26/03 00:57:35,603 WARNING [547602657280] (sensor.cpp:313) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ WARN] [1522025855.604434086]: Given stream configuration is not supported by the device!  Stream: Depth, Stream Index: 0, Format: Z16, Width: 640, Height: 480, FPS: 30
[ WARN] [1522025855.604702730]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 1, Format: Y8, Width: 640, Height: 480, FPS: 30
[ WARN] [1522025855.604855532]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 2, Format: Y8, Width: 640, Height: 480, FPS: 30
[ INFO] [1522025855.605242450]: publishStaticTransforms...
[ERROR] [1522025855.605559990]: Given depth profile is not supported by current device!
[FATAL] [1522025855.757329825]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera_manager-2] process has died [pid 19867, exit code 1, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/ubuntu/.ros/log/ab1a2a38-3090-11e8-b3fd-00044ba7d970/camera-realsense2_camera_manager-2.log].
log file: /home/ubuntu/.ros/log/ab1a2a38-3090-11e8-b3fd-00044ba7d970/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has died [pid 19868, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/ubuntu/.ros/log/ab1a2a38-3090-11e8-b3fd-00044ba7d970/camera-realsense2_camera-3.log].
log file: /home/ubuntu/.ros/log/ab1a2a38-3090-11e8-b3fd-00044ba7d970/camera-realsense2_camera-3*.log

lesreaper, 2018-03-26:
I wouldn't even bother with all of this @eleboss . I used this to install a D435 on a clean install of Jetpack 3.2. Worked perfectly and I was up in 20 minutes.

https://github.com/syedharoonalam/installLibrealsenseTX2

arybalkin, 2018-03-26:
@eleboss looks like USB3 super-speed was not properly negotiated between D435 and J120/TX2...

[ INFO] [1522025854.978657746]: Device Name: Intel RealSense USB2

You can check kernel messages or run `lsusb -t` to confirm.

-albertr

eleboss, 2018-03-26:
@arybalkin   This my devices information. Seems the usb 3.0 works, but just like you said USB3 super-speed was not properly negotiated between D435 and J120/TX2, I am trying to reflash the TX2 and use the jetpack3.1 to try.

Hi, What kind of cmake setting you are using?@lesreaper

ubuntu@tegra-ubuntu:~$ rs-enumerate-devices 
 Device info: 
    Name                          : 	Intel RealSense USB2
    Serial Number                 : 	801212070377
    Firmware Version              : 	05.08.15.00
    Physical Port                 : 	/sys/devices/3530000.xhci/usb1/1-2/1-2:1.0/video4linux/video0
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0AD6

Stream Profiles supported by Stereo Module
 Supported modes:                                      stream     resolution fps   format   
    Infrared 2	  480x270	@ 30Hz	Y8
    Infrared 1	  480x270	@ 30Hz	Y8
    Infrared 2	  480x270	@ 15Hz	Y8
    Infrared 1	  480x270	@ 15Hz	Y8
    Infrared 2	  480x270	@ 6Hz	Y8
    Infrared 1	  480x270	@ 6Hz	Y8
    Infrared 2	  424x240	@ 30Hz	Y8
    Infrared 1	  424x240	@ 30Hz	Y8
    Infrared 2	  424x240	@ 15Hz	Y8
    Infrared 1	  424x240	@ 15Hz	Y8
    Infrared 2	  424x240	@ 6Hz	Y8
    Infrared 1	  424x240	@ 6Hz	Y8
    Depth	  480x270	@ 30Hz	Z16
    Depth	  480x270	@ 15Hz	Z16
    Depth	  480x270	@ 6Hz	Z16
    Depth	  424x240	@ 30Hz	Z16
    Depth	  424x240	@ 15Hz	Z16
    Depth	  424x240	@ 6Hz	Z16

ubuntu@tegra-ubuntu:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 8086:0ad6 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

ubuntu@tegra-ubuntu:~$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-tegra/3p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-tegra/4p, 480M
    |__ Port 3: Dev 10, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 3: Dev 10, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 3: Dev 10, If 2, Class=Video, Driver=uvcvideo, 480M
    |__ Port 3: Dev 10, If 3, Class=Vendor Specific Class, Driver=, 480M

GustavZ, 2018-03-26:
My Jetson won't boot due to the full disk.
Does anybody know how i can reflash it without it being able to boot?

I am able to do ctrl+alt+f1 and deleted some files, but still it is not able to boot.
Maybe its the damaged image the kernel created...

TheMikeyR, 2018-03-26:
@GustavZ just put it into recovery mode and push new os through the microusb on your host. It should all be in the jetpack "suite" installed on your host pc. [guide if needed](http://developer.download.nvidia.com/embedded/L4T/r27_Release_v1.0/Docs/Jetson_X2_Developer_Kit_User_Guide.pdf?g-7D5d-t6bbXpot3Uvsn4rNabb15mTrPsZzcDxkb655QA3zvUaX15XEUxkXVsE32bXSIzkR8QPC_SegSH3IboWIWZ8WM6ie-iO-zGo7oQNaYnd0f9X9Ql0rQA7OZcVMdUlWJ_YkfMJi2zxOaTz6C5nPmnR7clceV8eArKfsVz25rKP2sMTkSaQ)

GustavZ, 2018-03-26:
@TheMikeyR  alright, i am reflashing it. But thats a shame, will be a whole day's work to have everything up and running again....

Do you successfully use the D435 on your Jetson Tx2 with JetPack 3.2?

TheMikeyR, 2018-03-26:
@GustavZ I've only tested D415 and it works fine. My D435 is in backorder, and will be shipped soon I guess.

eleboss, 2018-03-26:
@arybalkin Thank you for pointing the USB problem!!! It's my USB cable problem, I change a new USB 3.1 cable and it work very well! 

@RealSense-Customer-Engineering 
I tested the script It work very well, except a little bug:
I comment this line: 	
`sudo rm ${tgt_ko}.bckup`
in https://github.com/freemanlo/librealsense/blob/master/scripts/patch-utils.sh line 138
otherwise it will report:  cp   no such file or directory.

My procedure as follow:
Flash the TX2 by using jetpack 3.2(with J120 USB patch, if you are using official board, just ignore)
Using jetpack3.0 to install CUDA8.0 and cudnn5.1
git clone https://github.com/freemanlo/librealsense
cd librealsense/scripts/
gedit patch-utils.sh
delete line 138
`sudo rm ${tgt_ko}.bckup`
save
sudo apt-get install libusb-1.0-0-dev pkg-config cmake git libglfw3-dev qtcreator cmake-curses-gui build-essential libgtk-3-dev openssl-dev
cd  ../
./scripts/patch-realsense-ubuntu-xenial-jetson-tx2.sh 
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && udevadm trigger
install opencv by using https://github.com/jetsonhacks/buildOpenCVTX2 (ignore if you have already done that)
mkdir build && cd build
cmake ../ -DBUILD_EXAMPLES=true
make && sudo make install

GustavZ, 2018-03-26:
@eleboss  why are you not cloning librealsense from the original repo?

RealSense-Customer-Engineering, 2018-03-28:
[Realsense Customer Engineering Team Comment] 
Just make sure you have the correct script, and then the kernel patches can work with librealsense 2.x + SR300/D415/D435. Still need any help?

eleboss, 2018-03-28:
@GustavZ 
 Highly suggest you to did that from scratch, Flash the jetson TX2 then install, otherwise more problems will come.
For the first, I think it is necessary for my TX2 cause I meet some error， so I think it is necessary for you too.
For the second, libssl-dev is right.
For the third and the rest, can you use:
`rs-enumerate-devices `
after make install action to see the parameter of you realsense ?

eleboss, 2018-03-28:
@GustavZ 
I see the image shows you are using Intel Realsense USB2, It not right, you shouldn't see the USB2 if you are using USB3 super speed.
Check your USB cable and port.
Make sure both the cable and port are USB3.0
That is important!

GustavZ, 2018-03-28:
@eleboss 
The thing is jetson only has one usb3 port and as ia m using it standalone, so with mouse and keyboard, i am using a usb hub and probably thats not usb3 able...

I will reflash and do the librealsense installation as the first action, lets see...

rs-enumerate-devices gives me:
```
rs-enumerate-devices
 Device info: 
    Name                          : 	Intel RealSense USB2
    Serial Number                 : 	728312070461
    Firmware Version              : 	05.08.15.00
    Physical Port                 : 	/sys/devices/3530000.xhci/usb1/1-2/1-2.1/1-2.1:1.0/video4linux/video1
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0AD6

Stream Profiles supported by Stereo Module
 Supported modes:                                      stream     resolution fps   format   
    Infrared 2	  480x270	@ 30Hz	Y8
    Infrared 1	  480x270	@ 30Hz	Y8
    Infrared 2	  480x270	@ 15Hz	Y8
    Infrared 1	  480x270	@ 15Hz	Y8
    Infrared 2	  480x270	@ 6Hz	Y8
    Infrared 1	  480x270	@ 6Hz	Y8
    Infrared 2	  424x240	@ 30Hz	Y8
    Infrared 1	  424x240	@ 30Hz	Y8
    Infrared 2	  424x240	@ 15Hz	Y8
    Infrared 1	  424x240	@ 15Hz	Y8
    Infrared 2	  424x240	@ 6Hz	Y8
    Infrared 1	  424x240	@ 6Hz	Y8
    Depth	  480x270	@ 30Hz	Z16
    Depth	  480x270	@ 15Hz	Z16
    Depth	  480x270	@ 6Hz	Z16
    Depth	  424x240	@ 30Hz	Z16
    Depth	  424x240	@ 15Hz	Z16
    Depth	  424x240	@ 6Hz	Z16

 Device info: 
    Name                          : 	Platform Camera
    Serial Number                 : 	1-2.2-4
    Physical Port                 : 	/sys/devices/3530000.xhci/usb1/1-2/1-2.2/1-2.2:1.2/video4linux/video3
    Product Id                    : 	0823

Stream Profiles supported by RGB Camera
 Supported modes:                                      stream     resolution fps   format   
    Color	  2592x1944	@ 2Hz	Y16
    Color	  2592x1944	@ 2Hz	BGRA8
    Color	  2592x1944	@ 2Hz	RGBA8
    Color	  2592x1944	@ 2Hz	BGR8
    Color	  2592x1944	@ 2Hz	RGB8
    Color	  2592x1944	@ 2Hz	YUYV
    Color	  2048x1536	@ 2Hz	Y16
    Color	  2048x1536	@ 2Hz	BGRA8
    Color	  2048x1536	@ 2Hz	RGBA8
    Color	  2048x1536	@ 2Hz	BGR8
    Color	  2048x1536	@ 2Hz	RGB8
    Color	  2048x1536	@ 2Hz	YUYV
    Color	  1920x1080	@ 2Hz	Y16
    Color	  1920x1080	@ 2Hz	BGRA8
    Color	  1920x1080	@ 2Hz	RGBA8
    Color	  1920x1080	@ 2Hz	BGR8
    Color	  1920x1080	@ 2Hz	RGB8
    Color	  1920x1080	@ 2Hz	YUYV
    Color	  1792x1008	@ 5Hz	Y16
    Color	  1792x1008	@ 5Hz	BGRA8
    Color	  1792x1008	@ 5Hz	RGBA8
    Color	  1792x1008	@ 5Hz	BGR8
    Color	  1792x1008	@ 5Hz	RGB8
    Color	  1792x1008	@ 5Hz	YUYV
[... 637 lines cut ...]
    Color	  160x120	@ 15Hz	BGRA8
    Color	  160x120	@ 15Hz	RGBA8
    Color	  160x120	@ 15Hz	BGR8
    Color	  160x120	@ 15Hz	RGB8
    Color	  160x120	@ 15Hz	YUYV
    Color	  160x120	@ 10Hz	Y16
    Color	  160x120	@ 10Hz	BGRA8
    Color	  160x120	@ 10Hz	RGBA8
    Color	  160x120	@ 10Hz	BGR8
    Color	  160x120	@ 10Hz	RGB8
    Color	  160x120	@ 10Hz	YUYV
    Color	  160x120	@ 7Hz	Y16
    Color	  160x120	@ 7Hz	BGRA8
    Color	  160x120	@ 7Hz	RGBA8
    Color	  160x120	@ 7Hz	BGR8
    Color	  160x120	@ 7Hz	RGB8
    Color	  160x120	@ 7Hz	YUYV
    Color	  160x120	@ 5Hz	Y16
    Color	  160x120	@ 5Hz	BGRA8
    Color	  160x120	@ 5Hz	RGBA8
    Color	  160x120	@ 5Hz	BGR8
    Color	  160x120	@ 5Hz	RGB8
    Color	  160x120	@ 5Hz	YUYV

```

eleboss, 2018-03-28:
It is very clear that your USB3.0 is not being used.
```
rs-enumerate-devices
 Device info: 
    Name                          : 	Intel RealSense USB2
```
I think it is the USB2 hub problem.

TheMikeyR, 2018-03-28:
@GustavZ my temporarily fix until I received my usb3 hub was to use desktop share and then remote desktop into the jetson with only the camera plugged in to the one usb port on the jetson. A fine work around while you wait for the hub to arrive. The remote desktop sharing does cost some resources of the jetson though.

GustavZ, 2018-03-28:
@eleboss @TheMikeyR  thanks guys, i will try it and tell you how it went afterwards.

one more thing @eleboss in your routine you are talking about installing openCV, but OpenCV 3.3.1 comes with JetPack 3.2 if im right ?

RealSense-Customer-Engineering, 2018-03-29:
[Realsense Customer Engineering Team Comment] 
Hi GustavZ,

Please refer to the attached picture "D415_USB3_Connection.png". D415 is connected to type-A USB3 port and one USB mouse/keyboard dongle is connected to the white OTG adaptor from Jetson TX2 Development Kit. Make sure to connect to the correct USB3 port.
And, run below command to show me the USB connection condition.
$ rs-enumerate-devices
$ lsusb
$ lsusb -t
Then, run "dmesg", like to see if its USB enumeration has any issue.

As I know, JetPack 3.2 includes OpenCV.

freemanlo, 2018-03-29:
The connection picture "D415_USB3_Connection.png" attached. Once this is correct, let's see if the librealsense issue there.
[image]

GustavZ, 2018-03-29:
@RealSense-Customer-Engineering  i already posted the rs-enumerate-devices output exactly the comment before your answer, but here is everything again:

```
Device info: 
    Name                          : 	Intel RealSense 435
    Serial Number                 : 	728312070461
    Firmware Version              : 	05.08.15.00
    Physical Port                 : 	/sys/devices/3530000.xhci/usb2/2-1/2-1.1/2-1.1:1.0/video4linux/video1
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B07

Stream Profiles supported by Stereo Module
 Supported modes:                                      stream     resolution fps   format   
    Infrared 2	  1280x800	@ 30Hz	Y8
    Infrared 1	  1280x800	@ 30Hz	Y8
    Infrared 1	  1280x800	@ 25Hz	Y16
    Infrared 2	  1280x800	@ 25Hz	Y16
    Infrared 2	  1280x800	@ 15Hz	Y16
    Infrared 1	  1280x800	@ 15Hz	Y16
    Infrared 1	  1280x800	@ 15Hz	Y8
    Infrared 2	  1280x800	@ 15Hz	Y8
    Infrared 1	  1280x720	@ 30Hz	Y8
    Infrared 2	  1280x720	@ 30Hz	Y8
    Infrared 2	  1280x720	@ 15Hz	Y8
    Infrared 1	  1280x720	@ 15Hz	Y8
    Infrared 1	  1280x720	@ 6Hz	Y8
    Infrared 2	  1280x720	@ 6Hz	Y8
    Infrared 2	  848x480	@ 90Hz	Y8
    Infrared 1	  848x480	@ 90Hz	Y8
    Infrared 2	  848x480	@ 60Hz	Y8
    Infrared 1	  848x480	@ 60Hz	Y8
    Infrared 2	  848x480	@ 30Hz	Y8
    Infrared 1	  848x480	@ 30Hz	Y8
    Infrared 2	  848x480	@ 15Hz	Y8
    Infrared 1	  848x480	@ 15Hz	Y8
    Infrared 2	  848x480	@ 6Hz	Y8
    Infrared 1	  848x480	@ 6Hz	Y8
    Infrared 2	  640x480	@ 90Hz	Y8
    Infrared 1	  640x480	@ 90Hz	Y8
    Infrared 2	  640x480	@ 60Hz	Y8
    Infrared 1	  640x480	@ 60Hz	Y8
    Infrared 1	  640x480	@ 30Hz	Y8
    Infrared 2	  640x480	@ 30Hz	Y8
    Infrared 1	  640x480	@ 15Hz	Y8
    Infrared 2	  640x480	@ 15Hz	Y8
    Infrared 2	  640x480	@ 6Hz	Y8
    Infrared 1	  640x480	@ 6Hz	Y8
    Infrared 1	  640x400	@ 25Hz	Y16
    Infrared 2	  640x400	@ 25Hz	Y16
    Infrared 1	  640x400	@ 15Hz	Y16
    Infrared 2	  640x400	@ 15Hz	Y16
    Infrared 2	  640x360	@ 90Hz	Y8
    Infrared 1	  640x360	@ 90Hz	Y8
    Infrared 1	  640x360	@ 60Hz	Y8
    Infrared 2	  640x360	@ 60Hz	Y8
    Infrared 1	  640x360	@ 30Hz	Y8
    Infrared 2	  640x360	@ 30Hz	Y8
    Infrared 2	  640x360	@ 15Hz	Y8
    Infrared 1	  640x360	@ 15Hz	Y8
    Infrared 1	  640x360	@ 6Hz	Y8
    Infrared 2	  640x360	@ 6Hz	Y8
    Infrared 2	  480x270	@ 90Hz	Y8
    Infrared 1	  480x270	@ 90Hz	Y8
    Infrared 1	  480x270	@ 60Hz	Y8
    Infrared 2	  480x270	@ 60Hz	Y8
    Infrared 1	  480x270	@ 30Hz	Y8
    Infrared 2	  480x270	@ 30Hz	Y8
    Infrared 2	  480x270	@ 15Hz	Y8
    Infrared 1	  480x270	@ 15Hz	Y8
[... 964 lines cut ...]
sudo apt-get update
git clone https://github.com/freemanlo/librealsense
cd librealsense
gedit scripts/patch-utils.sh
# comment line 138 "sudo rm ${tgt_ko}.bckup"
sudo apt-get install libusb-1.0-0-dev pkg-config cmake git libglfw3-dev qtcreator cmake-curses-gui build-essential libgtk-3-dev libssl-dev
./scripts/patch-realsense-ubuntu-xenial-jetson-tx2.sh
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && udevadm trigger
mkdir build && cd build
cmake ../ -DBUILD_EXAMPLES=true
sudo make uninstall && make clean && make -j4 && sudo make install
```
what also could be interesting, after the installation: 
inside the librealsense directory there is a` kernel-4.4` directory next to the` buildJetsonTx2Kernel` directory and also there is a `4.4.38-tegra` directory in the `/home` folder containing these files:
```
/home/nvidia/4.4.38-tegra/uvcvideo.ko
/home/nvidia/4.4.38-tegra/videobuf2-core.ko
/home/nvidia/4.4.38-tegra/videobuf2-memops.ko
/home/nvidia/4.4.38-tegra/videobuf2-v4l2.ko
/home/nvidia/4.4.38-tegra/videobuf2-vmalloc.ko
```
why is this directory inside home and not inside librealsense?

Thats all that can be said, the installation runs without errors but realsense-viewer always hangs teh whole system right after the start.

GustavZ, 2018-03-29:
what's strange is: after reflashing several times now i cant even build librealsense anymore it breaks at this point:

```
Scanning dependencies of target rs-pointcloud
[ 60%] Building CXX object examples/pointcloud/CMakeFiles/rs-pointcloud.dir/rs-pointcloud.cpp.o
[ 61%] Linking CXX executable rs-capture
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmCloseOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmMap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmUnmap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevices2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmOpenOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevice2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevices'
collect2: error: ld returned 1 exit status
examples/capture/CMakeFiles/rs-capture.dir/build.make:98: recipe for target 'examples/capture/rs-capture' failed
make[2]: *** [examples/capture/rs-capture] Error 1
CMakeFiles/Makefile2:315: recipe for target 'examples/capture/CMakeFiles/rs-capture.dir/all' failed
make[1]: *** [examples/capture/CMakeFiles/rs-capture.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 61%] Linking CXX executable rs-software-device
[ 61%] Linking CXX executable rs-multicam
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmCloseOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmMap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmUnmap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevices2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmOpenOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevice2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevices'
collect2: error: ld returned 1 exit status
examples/software-device/CMakeFiles/rs-software-device.dir/build.make:98: recipe for target 'examples/software-device/rs-software-device' failed
make[2]: *** [examples/software-device/rs-software-device] Error 1
CMakeFiles/Makefile2:260: recipe for target 'examples/software-device/CMakeFiles/rs-software-device.dir/all' failed
make[1]: *** [examples/software-device/CMakeFiles/rs-software-device.dir/all] Error 2
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmCloseOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmMap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmUnmap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevices2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmOpenOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevice2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevices'
collect2: error: ld returned 1 exit status
examples/multicam/CMakeFiles/rs-multicam.dir/build.make:98: recipe for target 'examples/multicam/rs-multicam' failed
make[2]: *** [examples/multicam/rs-multicam] Error 1
CMakeFiles/Makefile2:425: recipe for target 'examples/multicam/CMakeFiles/rs-multicam.dir/all' failed
make[1]: *** [examples/multicam/CMakeFiles/rs-multicam.dir/all] Error 2
[ 62%] Linking CXX executable rs-pointcloud
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmCloseOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmMap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmUnmap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevices2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmOpenOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevice2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevices'
collect2: error: ld returned 1 exit status
examples/pointcloud/CMakeFiles/rs-pointcloud.dir/build.make:98: recipe for target 'examples/pointcloud/rs-pointcloud' failed
make[2]: *** [examples/pointcloud/rs-pointcloud] Error 1
CMakeFiles/Makefile2:480: recipe for target 'examples/pointcloud/CMakeFiles/rs-pointcloud.dir/all' failed
make[1]: *** [examples/pointcloud/CMakeFiles/rs-pointcloud.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

RealSense-Customer-Engineering, 2018-03-30:
[Realsense Customer Engineering Team Comment] 
Hi @GustavZ,

To make sure you can run the librealsense apps, just summarize the installation&build steps based on the latest TX2 BSP 28.2 (JetPack 3.2):
1. Download JetPack 3.2 from Nvidia website: https://developer.nvidia.com/embedded/dlc/jetpack-l4t-3_2-ga
2. Run 'JetPack-L4T-3.2-linux-x64_xxxx.run' to install and reflash Jetson TX2 board
3. Install librealsense (ex: 2.9.1)

git clone https://github.com/IntelRealSense/librealsense.git

cd librealsense

git branch v2.9.1

git checkout -b v2.9.1

sudo apt-get update

sudo apt-get install git cmake

sudo apt-get install libssl-dev

sudo apt-get install libusb-1.0-0-dev pkg-config libgtk-3-dev

sudo apt-get install libglfw3-dev

sudo apt-get install libudev-dev

sudo apt-get install cmake-curses-gui

sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/

sudo udevadm control --reload-rules && udevadm trigger

reboot

4. Build and Install librealsense

mkdir build && mkdir install

Modify CMakeLists.txt: (if you have the issue: undefined reference to symbol 'pthread_create@@GLIBC_2.17')

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -pthread")

cd build

cmake ../ -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DBUILD_UNIT_TESTS=false

make -j4

sudo make install

5. Connect D415 or D435 to the USB3 type-A port
6. Here, you should be able to run "rs-capture" or other apps to stream D415

Hope this is clear to resolve your issue.
------------------------------------------------
If you like to support SR300 or other realsense camera formats, you can download the below patch and run the patch:
https://github.com/freemanlo/librealsense/tree/master/scripts/patch-realsense-ubuntu-xenial-jetson-tx2.sh
./scripts/patch-realsense-ubuntu-xenial-jetson-tx2.sh
After the patch, make sure the modules updated in /lib/modules/4.4.38-tegra/kernel/drivers/media/usb/uvc/ and /lib/modules/4.4.38-tegra/kernel/drivers/media/v4l2-core/.

RealSense-Customer-Engineering, 2018-03-30:
[Realsense Customer Engineering Team Comment] 
As for your question about a 'kernel-4.4' inside the librealsense directory, it's a temporary directory for the kernel module patches. After the patch is successful, you can delete it and also below modules. As I mentioned above, just make sure the modules updated in /lib/modules/4.4.38-tegra/kernel/drivers/media/usb/uvc/ and /lib/modules/4.4.38-tegra/kernel/drivers/media/v4l2-core/ after the successful patch.

    /home/nvidia/4.4.38-tegra/uvcvideo.ko/home/nvidia/4.4.38-tegra/videobuf2-core.ko/home/nvidia/4.4.38-tegra/videobuf2-memops.ko/home/nvidia/4.4.38-tegra/videobuf2-v4l2.ko/home/nvidia/4.4.38-tegra/videobuf2-vmalloc.ko

And, you can also upgrade your D435 FW to 5.9.2 (5.9.2.0 +) instead of 5.8.15.

GustavZ, 2018-03-31:
@RealSense-Customer-Engineering 
Thank you and three questions on that routine:

1. It seems my the building error i run into is due to openCV that comes with jetPack3.2 so your routine will not work with this routine as the building process will still fail like mine did (ref to https://devtalk.nvidia.com/default/topic/1007290/jetson-tx2/building-opencv-with-opengl-support-/post/5141945/?offset=13#5248633)
2. why are you checking out to release v2.9.1? why not using the latest verified release v2.10.1?
3. You are not applying any kernel patch. why is this not necessary anymore?
But still you say i can delete the "kernel-4.4" dir, but if i do not patch the kernel this dir will not be created. thats a bit confusing.
4. How do i upgrade the FW of the D435?

Thank you in advance!

RealSense-Customer-Engineering, 2018-04-02:
[Realsense Customer Engineering Team Comment] 
Hi @GustavZ,
For Q1: it’s interested that I didn’t see this issue when I did the clear reflash process based on JetPack 3.2.
For Q2: yes, you can use 2.10.1.
For Q3: Per check, The kernel of L4T BSP 28.2 (in JetPack 3.2) has supported Depth (Z16) format. Therefore, without patches, you still can stream D415 or D435. You can run below command to see the video device info.
$ sudo apt install v4l-conf
$ v4l-info /dev/video0
$ v4l-info /dev/video1
If you need other realsense camera format support, you can run the patch.
For Q4: please refer to below page to download and use the tool and firmware.
https://downloadcenter.intel.com/download/27522/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?v=t

GustavZ, 2018-04-03:
@RealSense-Customer-Engineering  when running your routine i run into this new error (i left out the kernel patch):

```
[ 23%] Building CXX object CMakeFiles/realsense2.dir/src/image.cpp.o
/home/nvidia/librealsense/src/image.cpp:23:19: fatal error: cpuid.h: No such file or directory
compilation terminated.
CMakeFiles/realsense2.dir/build.make:374: recipe for target 'CMakeFiles/realsense2.dir/src/image.cpp.o' failed
make[2]: *** [CMakeFiles/realsense2.dir/src/image.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
^CCMakeFiles/realsense2.dir/build.make:158: recipe for target 'CMakeFiles/realsense2.dir/src/context.cpp.o' failed
make[2]: *** [CMakeFiles/realsense2.dir/src/context.cpp.o] Interrupt
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/realsense2.dir/all' failed
make[1]: *** [CMakeFiles/realsense2.dir/all] Interrupt
Makefile:127: recipe for target 'all' failed
make: *** [all] Interrupt

```

RealSense-Customer-Engineering, 2018-04-03:
[Realsense Customer Engineering Team Comment] 
Hi @GustavZ,
You face the issue similar with the issue (#1446) which only happens on ARM platform. You can try the workaround/patch there first if you like to use the latest version of librealsense.

RealSense-Customer-Engineering, 2018-04-11:
[Realsense Customer Engineering Team Comment] 
Hi @GustavZ,

Did you try the librealsense 2.10.3 which fixes ARM build issue? Still need any help?

GustavZ, 2018-04-16:
@RealSense-Customer-Engineering does this mean it is now possible to install the dsk with `apt-get` on the jetson?

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
Hi @GustavZ,

Did you build form librealsense source and apply the proper script to resolve this issue?

RealSense-Customer-Engineering, 2018-06-22:
[Realsense Customer Engineering Team Comment]
Is this still an issue?  If not, the ticket will be closed in 5 days.

RealSense-Customer-Engineering, 2018-07-08:
[Realsense Customer Engineering Team Comment]
Hi @GustavZ,

I will close this first. If you still have any issue, you can re-open it or create another ticket.

closed 2018-07-08 by RealSense-Customer-Engineering