realsenseai/librealsense #1428: Illegal instruction on Up Squared computer with Pentium processor

yichenpan, 2018-03-23:
| Camera Model                       | {D400 } | 
| Firmware Version                   | 05.08.15.00| 
| Operating System & Version |   Linux Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.13.0                                        | 
| Platform                                 | [Up Squared computer with Pentium N4200 processor](https://up-shop.org/up-boards/97-up-squared-pentium-quad-core-8gb-memory64gb-emmc.html)  |
| SDK Version                            |  v2.10.2 | 

### Issue Description
I tried to compile and run the latest (v2.10.2) librealsense on a Up Squared computer with Pentium N4200 processor.  The compiling was smooth and successful, but I got an error message when I run any librealsense tools or examples:
>  Illegal instruction (core dumped)

Actually I have run into the same issue with my own program on Up Squared computer with Celeron processor before, and I solved that time by add -msse3 flag to compiler (since Celeron does not support AVX2 instructions).

However, this time I still got the error even after I added -msse3 flag... So I was wondering anyone knows how to solve this? or has the same issue?

Thanks

dorodnic, 2018-03-24:
Hi @yichenpan 
This is most likely related to the fact that in 2.10.2 we are trying to detect AVX support automatically and switch to it when possible. I assume the crash occurs after the CPUID identification, but if you could send us a stack-trace or a core-dump it would help a lot.

In the meantime I would recommend to roll-back to 2.10.1 for your use case and we will try to provide a solution ASAP. 

Also, if you come up with something, please consider contributing to this repository. I'm mentioning this because it would be very hard for us to test the fix since we do not have access to this hardware (and ordering it would take some time).

yichenpan, 2018-03-26:
Hi @dorodnic 
Thank you for your reply. I don't have my hands on Up Squared board during the weekend, so I gave it a try this morning.

I've roll-back to v2.10.0 and compiled again, but it's still giving the same illegal instruction error. Then I checked it with gdb, and here is what I got:

```
(gdb) run
Starting program: /usr/local/bin/rs-multicam 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGILL, Illegal instruction.
0x00007ffff75aaa6f in librealsense::color_map::color_map (
    this=0x7ffff7d92480 <librealsense::jet>, 
    values=std::vector of length 5, capacity 5 = {...}, steps=4000)
    at /home/yichen/Desktop/librealsense/src/proc/colorizer.h:28
28	                _map[(float)i / (values.size() - 1)] = values[i];
(gdb) disas
Dump of assembler code for function librealsense::color_map::color_map(std::vector<librealsense::float3, std::allocator<librealsense::float3> > const&, int):
   0x00007ffff75aaa04 <+0>:	push   %rbp
   0x00007ffff75aaa05 <+1>:	mov    %rsp,%rbp
   0x00007ffff75aaa08 <+4>:	push   %rbx
   0x00007ffff75aaa09 <+5>:	sub    $0x48,%rsp
   0x00007ffff75aaa0d <+9>:	mov    %rdi,-0x38(%rbp)
   0x00007ffff75aaa11 <+13>:	mov    %rsi,-0x40(%rbp)
   0x00007ffff75aaa15 <+17>:	mov    %edx,-0x44(%rbp)
   0x00007ffff75aaa18 <+20>:	mov    %fs:0x28,%rax
   0x00007ffff75aaa21 <+29>:	mov    %rax,-0x18(%rbp)
   0x00007ffff75aaa25 <+33>:	xor    %eax,%eax
   0x00007ffff75aaa27 <+35>:	mov    -0x38(%rbp),%rax
   0x00007ffff75aaa2b <+39>:	mov    %rax,%rdi
   0x00007ffff75aaa2e <+42>:	callq  0x7ffff727bb80 <_ZNSt3mapIfN12librealsense6float3ESt4lessIfESaISt4pairIKfS1_EEEC1Ev@plt>
   0x00007ffff75aaa33 <+47>:	mov    -0x38(%rbp),%rax
   0x00007ffff75aaa37 <+51>:	add    $0x30,%rax
   0x00007ffff75aaa3b <+55>:	mov    %rax,%rdi
   0x00007ffff75aaa3e <+58>:	callq  0x7ffff726bef0 <_ZNSt6vectorIN12librealsense6float3ESaIS1_EEC1Ev@plt>
   0x00007ffff75aaa43 <+63>:	movq   $0x0,-0x20(%rbp)
   0x00007ffff75aaa4b <+71>:	mov    -0x40(%rbp),%rax
   0x00007ffff75aaa4f <+75>:	mov    %rax,%rdi
   0x00007ffff75aaa52 <+78>:	callq  0x7ffff72a6e00 <_ZNKSt6vectorIN12librealsense6float3ESaIS1_EE4sizeEv@plt>
   0x00007ffff75aaa57 <+83>:	cmp    -0x20(%rbp),%rax
   0x00007ffff75aaa5b <+87>:	seta   %al
---Type <return> to continue, or q <return> to quit---
   0x00007ffff75aaa5e <+90>:	test   %al,%al
   0x00007ffff75aaa60 <+92>:	je     0x7ffff75aab23 <librealsense::color_map::color_map(std::vector<librealsense::float3, std::allocator<librealsense::float3> > const&, int)+287>
   0x00007ffff75aaa66 <+98>:	mov    -0x20(%rbp),%rax
   0x00007ffff75aaa6a <+102>:	test   %rax,%rax
   0x00007ffff75aaa6d <+105>:	js     0x7ffff75aaa7f <librealsense::color_map::color_map(std::vector<librealsense::float3, std::allocator<librealsense::float3> > const&, int)+123>
=> 0x00007ffff75aaa6f <+107>:	vxorps %xmm1,%xmm1,%xmm1
   0x00007ffff75aaa73 <+111>:	vcvtsi2ss %rax,%xmm1,%xmm1
   0x00007ffff75aaa78 <+116>:	vmovss %xmm1,-0x48(%rbp)
   0x00007ffff75aaa7d <+121>:	jmp    0x7ffff75aaa9d <librealsense::color_map::color_map(std::vector<librealsense::float3, std::allocator<librealsense::float3> > const&, int)+153>
   0x00007ffff75aaa7f <+123>:	mov    %rax,%rdx
   0x00007ffff75aaa82 <+126>:	shr    %rdx
   0x00007ffff75aaa85 <+129>:	and    $0x1,%eax
   0x00007ffff75aaa88 <+132>:	or     %rax,%rdx
   0x00007ffff75aaa8b <+135>:	vxorps %xmm0,%xmm0,%xmm0
   0x00007ffff75aaa8f <+139>:	vcvtsi2ss %rdx,%xmm0,%xmm0
   0x00007ffff75aaa94 <+144>:	vaddss %xmm0,%xmm0,%xmm2
   0x00007ffff75aaa98 <+148>:	vmovss %xmm2,-0x48(%rbp)
   0x00007ffff75aaa9d <+153>:	mov    -0x40(%rbp),%rax
   0x00007ffff75aaaa1 <+157>:	mov    %rax,%rdi

```
It looks to me the problem is still from AVX instructions... I'll try to add a flag and compiled again later.

yichenpan, 2018-03-27:
Sorry. I made a mistake, v2.10.0 does work on UP Square. Earlier after I compiled the new version and installed, I don't know why somehow the binaries under /usr/local/bin didn't get updated... 

Anyway, as for v2.10.2, I think the crush reason is just like what @dorodnic  said. The image.cpp made an assumption by setting do_avx to true once it detects ssse3 flag, but apparently this assumption does not apply to the Pentium processor on UP Square computer.

drhalftone, 2018-04-09:
I have a crazy question.  I installed librealsense on my Up Board following your instructions for install on Linux with apt-get.  How do I install this 2.10.0 version?

yichenpan, 2018-04-09:
@drhalftone Clone this librealsense repo to your computer, checkout to v2.10.0 and then build from source.

drhalftone, 2018-04-09:
Do I need to uninstall the packages installed with apt-get?

yichenpan, 2018-04-09:
@drhalftone You don't have to uninstall, but it's always good to make sure all binaries and libraries under /usr/local/ are the version you want before you use them.

drhalftone, 2018-04-16:
> We are planning to update the official PPA tomorrow if you prefer to use the package

Yes I’ll wait.  The earlier instructions worked and have no urgent need.

closed 2018-04-16 by yichenpan