realsenseai/librealsense #1429: Can't get color of point

VelazcoJD, 2018-03-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300                                                    | 
| Firmware Version                   | 2.10.2                                                     | 
| Operating System & Version |   Windows 10                                         | 
| Kernel Version (Linux Only)    |                                                               | 
| Platform                                 | PC                                                          |
| SDK Version                            |  2                                                          | 

I am trying to create a point cloud and store RGB data with its corresponding point (to use outside the SDK). I had this working using the previous library (RealSense), but I cannot get it to work with RealSense2. I tried both using intrinsic/extrinsic calculations to find corresponding color as well as using map_to method and use texture coordinates. I think I may be missing something small... This is the code using map_to method and texture coordinates.

```
rs2::pipeline pipe;
rs2::config cameraConfig;
cameraConfig.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16);
cameraConfig.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGB8);
rs2::pipeline_profile pipeProfile = pipe.start(cameraConfig);
float depthScale = pipeProfile.get_device().first<rs2::depth_sensor>().get_depth_scale();

rs2::frameset frames = pipe.wait_for_frames();
rs2::depth_frame depthFrame = frames.get_depth_frame();
rs_points = rs_pointCloud.calculate(depthFrame);
auto colorFrame = frames.get_color_frame();
rs_pointCloud.map_to(colorFrame);

rs2_intrinsics color_intrinsic = colorFrame.get_profile().as<rs2::video_stream_profile>().get_intrinsics();

auto rs_vertices = rs_points.get_vertices();
auto textureCoordinates = rs_points.get_texture_coordinates();
const unsigned char* colorData = static_cast<const unsigned char*>(colorFrame.get_data());

for (int i = 0; i < rs_points.size(); i++) {
    int x = static_cast<int>(textureCoordinates[i].u * color_intrinsic.width);
    int y = static_cast<int>(textureCoordinates[i].v * color_intrinsic.height);
    int colorLocation = y * color_intrinsic.width + x;

    float px = rs_vertices[i].x;
    float py = rs_vertices[i].y;
    float pz = rs_vertices[i].z;
    unsigned char r = colorData[colorLoation];
    unsigned char g = colorData[colorLocation + 1];
    unsigned char b = colorData[colorLocation + 2];
    save(px, py, pz, r, g, b);
}
```

I put in only relevant code. Basically, the texture coordinates' u and v are always 0 which then assigns the first color in colorData to every point. Am I missing something? Thanks in advance.

ev-mp, 2018-03-25:
Hi @VelazcoJD,
When you call for `rs_points  = rs_pointCloud.calculate(depthFrame);` the reference image should already be registered with `rs_pointCloud.map_to(someFrame);`, 
Otherwise no texture coordinates will be generated.

The order is less important if you're running the projection in a loop, i.e. you will get vertices with no texture mapping in the first iteration, but then it will reappear.
If, instead, it is only a `one-shot` application, as appear in the snippet, then the order is important.

You need to switch the order of operations in the code 
>rs2::frameset frames = pipe.wait_for_frames();
...
rs_pointCloud.map_to(colorFrame);
...
rs_points = rs_pointCloud.calculate(depthFrame);

In addition, when you calculate the location of the color pixel, note that each pixel is N-bytes wide (3 for RGB) when you access its raw data via `char*` pointer:
>unsigned char r = colorData[colorLoation  *__**size_of_the_reference_pixel**__];

VelazcoJD, 2018-03-25:
Ah! I was trying to set this up and was using a single capture to test. That was my major mistake, thank you.

For any one that may read this, you should also wait for a couple of frames before starting to collect data.

closed 2018-03-25 by VelazcoJD