realsenseai/librealsense #1430: rs2_intrinsics coeffs[] all 0 by default

beyondmetis, 2018-03-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
|Firmware Version | 05.08.15.00 |
|Operating System & Version | Ubuntu 16 |
|Kernel Version (Linux Only) |  4.8.14 |
|Platform | PC |
| SDK Version | 2.10.2 |

I ran the following code after initializing the device:

auto depthstream = profile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();
rs2_intrinsics depthint = depthstream.get_intrinsics();

I know according to the documentation that rs2_intrinsics contains elements such as fx, fy, ppx, ppy, and coeffs (for the 5 lens undistort parameters). When inspecting the values contained by these elements, I found:

fx = 940.924011
fy = 940.924011
ppx = 636.445312
ppy = 359.391479
coeffs[0] = 0
coeffs[1] = 0
coeffs[2] = 0
coeffs[3] = 0
coeffs[4] = 0

Are these coeffs supposed to be all 0? I figured there should at least be some reasonable default based on factory calibration.

dorodnic, 2018-03-25:
Hi @beyondmetis 
Yes, these are supposed to be zero for the D400. We consider adding coefficient estimation to the RGB calibration to reduce the distortion (by about 1 pixel at extremes), but at the moment projection without coefficients is the most accurate you can do (we are not calibrating and then ignoring the coefficients, we estimate fx, fy, ppx and ppy without them)

kazoo-kmt, 2018-04-03:
@dorodnic I was wondering the same thing, so let me confirm. There are similar discussion at #1380, #837, #780. but it's unclear. Is it correct tha D435/D415 with a factory calibration setup always returns the RGB image without any rectification or undistortion technique, though all the coefficient parameters are zero?

dorodnic, 2018-04-03:
@kazoo-kmt , that is correct, at least for now. Research on it shows very little improvement to texture-mapping accuracy, while it would require more CPU to process. That's why for now the recommendation is to use all-zero coefficients.