realsenseai/librealsense #1431: architecture

phaggio, 2018-03-25:
Do RealSense cameras have any support for ARM architecture?

BeLikeH2O, 2018-03-25:
NOPE. 

I've been tracking and using RealSense for the past 5 years, and it's been nothing but disappointment. All of their demos are just that - demos. Nothing ever goes into production with their customers or partners. 

I have no doubt that in a couple of years, RealSense is going to get canned. You can tell because at least they used to do demos (back in CES 2015 and 2016). After that, they couldn't even showcase demos, which is particularly sad. SR300 is dead, R200 is gone. And this D series was promised for a long time and only recently rolled out... god.

The depth map has so many damned holes... you can't do any real high fidelity work. I wish PrimeSense was still around!

phaggio, 2018-03-25:
I see. I can see the same as they already shoved their project Alloy under the rug. 
Thanks for the tip.

RealSense-Customer-Engineering, 2018-03-26:
[Realsense Customer Engineering Team Comment] 
Hello @phaggio,

RealSense supports Windows, Linux, MacOs and soon Android.  We do not test on competitors SoCs but if they support these operating systems or the wrappers we offer, the D400 series RealSense products should work as long as the board has a USB3 port.  Feel free to experiment and post results.

Regards,
Jesus G.
Intel Customer Support

closed 2018-04-04 by dorodnic