realsenseai/librealsense #1432: RealSense Website is a Joke

BeLikeH2O, 2018-03-25:
The Intel RealSense website is a complete insult to humanity. The content there is utterly useless and uninformative. Why would you spend so much screen real estate showing an "unboxing" of literally a pair of hands opening a packaging box for the camera? 

Why not put some useful content to tell us what EXACTLY this camera is good for? Generic statements about how it has many applications in depth sensing is vague. FAQs are equally vague and unhelpful. 

I'd like to be able to see some actual examples of people who actually used RealSense to build some useful applications? This whole product line has made no progress over the last 5 years. What happened to the R200? 

Why is your product always out of stock?
Why can't I order this through Amazon?
What companies or customers actually use RealSense?

This site is a clear indication that Intel is resting on its brand - which has been declining over the years - and not making anything truly innovative.

ppzgenome, 2018-03-25:
As harsh as these questions and comments are... I have to agree. RealSense has been going downhill for a while now. I think some of their top talent has taken flight (or fired from the layoffs in the past couple of years) and now they're probably a bucket for keeping all the incompetent. You can tell from the presentation of their product - lazy, ambiguous, and completely disrespectful to the customer in every manner.

phaggio, 2018-03-25:
GitHub may not be the right place to post these questions and comments, but I guess there's no where else to ask these questions.

MachLearn4ever, 2018-03-25:
Seems like the right place to ask to me. After all, who says Github is only for technical concerns? The overall RealSense product is really just low quality and poorly explained to the public. They've been hiding behind a generic phrasing about "sensification" for years and made nothing truly game changing.

phaggio, 2018-03-25:
I think ZED has a much better product and support.
https://www.stereolabs.com/zed/

ProfFan, 2018-03-26:
ZED also does not state that all its depth info is calculated on the host, not by the camera.

closed 2018-04-04 by dorodnic