realsenseai/librealsense #1433: Where is Person tracking feature?

gunnicus, 2018-03-25:
Is there any plan to bring back the person tracking feature? I have to use the old Realsense R200 instead of the D series because of this.

phaggio, 2018-03-25:
As far as I can see, there's no real feature support for the D-series cameras.

TomOskam, 2018-03-27:
You can make it yourself by using OpenCv examples? 

Use the wrapper to obtain the image from the camera (see included examples) and use OpenCv for future tracking.

gunnicus, 2018-03-27:
Which OpenCv example? I couldn't fine any OpenCv example same as person tracking in old R200 SDK.

TomOskam, 2018-03-28:
Depends on what features you want included. OpenCv offers things like hand and gesture tracking as pedestrian or car tracking also face recognition.

gunnicus, 2018-03-28:
I want to have body tracking with x,y,z of joints same as old R200 sdk.

TomOskam, 2018-03-28:
You're looking for 'skeleton tracking'. I found no example codes for this feature on OpenCv. You'll have to invest time to combine some clever OpenCv tricks to accomplice this I ques.

gunnicus, 2018-03-28:
Yes 'skeleton tracking', but it was in the old R200 sdk I wonder why Intel can't include that feature in.

TomOskam, 2018-03-28:
Either Intel will include it in the coming patches or the community will make a function like it in the public wrappers

RealSense-Customer-Engineering, 2018-04-13:
[Realsense Customer Engineering Team Comment] 
Hello,

Our SDK is focused on providing the best depth data across multiple wrappers and operating systems.  We wish to grow our open source community and encourage them to do something wonderful with our products.  We have no plan to add middleware libraries to our product but we may partner with 3rd parties in the future.  All updates on these items will be via github and our website: http://realsense.intel.com.

Regards,
Jesus G.
Intel Customer Support

closed 2018-05-11 by RealSense-Customer-Engineering