realsenseai/librealsense #1435: Request to add support for USB2 high-speed transfers in the next DXX firmware

arybalkin, 2018-03-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415 | 
| Firmware Version                   | 5.9.2  | 
| Operating System & Version |   Linux Ubuntu 16.04.3 LTS  | 
| Kernel Version (Linux Only)    |  4.4.38                                        | 
| Platform                                 | NVIDIA Jetson TX2 |
| SDK Version                            |  2.10.2 }                          | 

We are planning to use a single video stream from RGB sensor in 1280x720 @ 30 FPS and USB2 should be capable of providing sufficient bandwidth for this application. Would it be possible to add support for USB2 in the next firmware release of this camera? Not sure if it matters, but we are using Nvidia Jetson TX2 and USB3 is quite quirky when used with D415 camera (latest firmware).

-albertr

RealSense-Customer-Engineering, 2018-03-26:
[Realsense Customer Engineering Team Comment] 
hi @arybalkin,

do you mean you want to use RGB video from D415 only and not using depth?

arybalkin, 2018-03-26:
That's correct, currently only streaming one stream at the time (mostly from color sensor) at 1280x720 @ 30fps. I think even streaming two streams (both color and depth) at that resolution and frame rate would be within bandwidth limits of USB2 high-speed?

Also, I don't need raw, MJPEG compression will be fine.

arybalkin, 2018-03-29:
I'm on a fence right now deciding between DXX and ZED.  Please let me know if Intel would consider to include this requested functionality in the new release of DXXX firmware...

JasonChang13, 2018-04-25:
hi @arybalkin.

I just tested with Librealsense V2.10.4 with latest FWv5.9.9, and I can have stream for both RGB and depth, but resolution and framerate are limited due to bandwidth.
please let me know if this also works for you, thanks.

Regards,
Jason
Intel Customer Platform Engineering

RealSense-Customer-Engineering, 2018-05-11:
[Realsense Customer Engineering Team Comment]
hi @arybalkin,

would like to know if you still need support on this topic?
thanks.

dongxuanlb, 2018-12-29:
HI i think the question is would d4XX support the V4L2_PIX_FMT_MJPEG format by hardware for the color stream?