realsenseai/librealsense #1441: Export a PLY with a the cropped backgroud

baptiste-mnh, 2018-03-26:
Hi,

I know how to get the point cloud in a PLY format however, from the rs-align example, I want to get the PLY without the trimmed part.
So, I want to choose a maximum distance to capture and generate a PLY recording only the points in the chosen distance.

I have tried several methods starting from the rs-align example but I am running out of ideas.

I'm using D400 models.
Thank you in advance.

Alex9812, 2018-03-26:
I solved this issue using a the box crop feature from PCL.

If you do not want to deal with PCL, you can use cloudcompare and crop out regions from your ply file(s)

baptiste-mnh, 2018-03-26:
Thank you for your fast answer, I will try this option. I will come back if I find something more convinient.

baptiste-mnh, 2018-03-29:
For those who didn't resolve it, here is my code working well to crop a PLY with PCL lib 

  ```
#include <iostream>
#include <pcl/io/pcd_io.h>
#include <pcl/io/ply_io.h>
#include <pcl/point_types.h>
#include <pcl/filters/crop_box.h>

int
main (int argc, ch
```ar** argv)
{
  pcl::PointCloud<pcl::PointXYZRGB>::Ptr cloud (new pcl::PointCloud<pcl::PointXYZRGB>);

  if (pcl::io::loadPLYFile<pcl::PointXYZRGB> ("/home/baptiste-mnh/Documents/test2.ply", *cloud) == -1) 
  {
    PCL_ERROR ("Couldn't read file\n");
    return (-1);
  }

    Eigen::Vector4f minPoint;
    minPoint[0]=-0.5f;  // define minimum point x
    minPoint[1]=-0.75f;  // define minimum point y
    minPoint[2]=0;  // define minimum point z

    Eigen::Vector4f maxPoint;
    maxPoint[0]=5;  // define max point x
    maxPoint[1]=0.75f;  // define max point y
    maxPoint[2]=1;  // define max point z

    Eigen::Vector3f boxTranslatation;
    boxTranslatation[0]=0;
    boxTranslatation[1]=0;
    boxTranslatation[2]=0;

    Eigen::Vector3f boxRotation;
    boxRotation[0]=0;  // rotation around x-axis
    boxRotation[1]=0;  // rotation around y-axis
    boxRotation[2]=0;

    pcl::PointCloud<pcl::PointXYZRGB>::Ptr cloudOut (new pcl::PointCloud<pcl::PointXYZRGB>);

    pcl::CropBox<pcl::PointXYZRGB> cropFilter;
    cropFilter.setInputCloud (cloud);
    cropFilter.setMin(minPoint);
    cropFilter.setMax(maxPoint);
    cropFilter.setTranslation(boxTranslatation);
    cropFilter.setRotation(boxRotation);

    cropFilter.filter (*cloudOut);

    pcl::io::savePLYFileBinary<pcl::PointXYZRGB>("/home/baptiste-mnh/Documents/test2_cropped.ply", *cloudOut);

    return (0);
}

```

> source : http://www.pcl-users.org/template/NamlServlet.jtp?macro=print_post&node=3889666

closed 2018-03-29 by baptiste-mnh