realsenseai/librealsense #1442: librealsense2-utils Packet not found; SDK 2.0 on PC, Ubuntu 16.04

ChristophSchwanke, 2018-03-26:
| Camera Model                       |  D400  | 
| Firmware Version                   | not installed | 
| Operating System & Version |   Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  Linux 4.13.0-36-generic    | 
| Platform                                 | PC  |
| SDK Version                            |  not installed   | 

I tried to install the SDK 2.0 on my PC as described here:
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md
Everything runs fine until the step
"sudo apt-get install librealsense2-utils". The packet could not be found. What did I do wrong?

Error message (with minimal translation):
>sudo apt-get install librealsense2-utils
Paketlisten werden gelesen... Fertig  (list read)
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig (... done)
E: Paket librealsense2-utils kann nicht gefunden werden.  (Packet could not be found)

kind regards

ev-mp, 2018-03-27:
Hello @ChristophSchwanke ,
Please verify that you have the public key received, and rerun `sudo apt update` to make sure your apt cache is current.
The latest released version of this package is (2.10.2-0~realsense0.64).

Additionally, the specific package supports 64bit Debian distributions only.
Additionally - the debian packages are for 64bit Ubuntu/Debian distributions. In case you're running 32 bit distribution it w

ChristophSchwanke, 2018-04-08:
> Hi @ChristophSchwanke
> Are you still experiencing this issue? Could you please comment if you are using x86 or x64 version of Ubuntu?

Dear Sergey,
Thank you for the fast help! I did not yet figure out what the problem was, since a friend installed the SDK without problems on his pc and we use  this one. I guess that I installed a wrong Ubuntu version. I will check it in a week or so.

Kind regards

RealSense-Customer-Engineering, 2018-04-18:
[Realsense Customer Engineering Team Comment] 
@ChristophSchwanke are you still having this issue?

ChristophSchwanke, 2018-04-25:
> Hello @ChristophSchwanke, as you haven't replied and it appeared in your last comment that you had a working system, I will close this issue. If you need more support or have additional issues, please feel free to open a new issue(s).
> Regards,
> Ted,
> Intel RealSense Support

Hello Ted,
Sorry, that I didn't find the time. Thank you for the support.

Kind regards
Christoph

Rahul-Venugopal, 2018-07-23:
I am also having the same issue . Please let me know how I can solve this.

closed 2018-11-06 by dorodnic