realsenseai/librealsense #1443: Failure to launch viewer after install

kevin-j-miller, 2018-03-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | unknown | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  16.04   | 
| Platform                                 | PC  |
| SDK Version                            |  2  | 

### Issue Description
I am attempting to install the SDK following the instructions given [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md). Everything seems to install correctly, but when I go to launch the realsense-viewer, I get an error "Illegal instruction (core dumped)". 

Any hints on what I'm I doing wrong?

Thanks!

bu3w, 2018-03-27:
Which kind of CPU do you use? Core series? Or atom series?
I guess, you have to rebuild the project from source code?

dorodnic, 2018-03-27:
Hi @kevin-j-miller 
This is most likely related to #1428. A work-around is to fall back to v2.10.1. More information about your system would be very helpful

kevin-j-miller, 2018-03-29:
Hi @dorodnic 
Thanks for the pointer! The machine I initially encountered this problem on has a Pentium G4400, which I think is indeed missing AVX. I tried again today with another machine (with a Core i5-3470), and ran into the same problem -- maybe this cpu also doesn't support AVX?  

I'll try installing v2.10.1 as you suggest, and report back on whether that works.

RealSense-Customer-Engineering, 2018-04-05:
[Realsense Customer Engineering Team Comment] 
@kevin-j-miller did you try v2.10.1?

RealSense-Customer-Engineering, 2018-04-18:
[Realsense Customer Engineering Team Comment] 
@kevin-j-miller
This issue should be fixed and merged into development branch, can you have a try?

dorodnic, 2018-04-18:
Should be resolved with [v2.10.4](https://github.com/IntelRealSense/librealsense/releases/tag/v2.10.4)

closed 2018-04-18 by dorodnic