realsenseai/librealsense #1445: C# Wrapper TextureCoordinates are always 0.

telemetrieTP23, 2018-03-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435                                                      | 
| Firmware Version                   |05.08.15.00                                              | 
| Operating System & Version |  Win 10                                                   | 
| Platform                                 | PC                                                           |
| SDK Version                            |Intel® RealSense™ SDK 2.0 (Build 2.10.2)             | 

Dear intel realsense developers !

I want to get back the TextureCoordinates of the DepthImage, but instead u,v for all TextureCoordinates are always 0.

Points worldPoints;
PointCloud world = new PointCloud();
public Points.TextureCoordinate[] texturepoints = new Points.TextureCoordinate[640 * 480];
 using (var f = q.WaitForFrame() as DepthFrame)
                {
                    worldPoints = world.Calculate(f);
                    worldPoints.CopyTo(texturepoints);
}

=====> texturepoints[i].u, texturepoint[i].v ---> are alway 0.

Getting back the Vertex Coordinates with nearly the same code is no problem:

Points worldPoints;
PointCloud world = new PointCloud();
 public Points.Vertex[] worldpoints= new Points.Vertex[640*480];
 using (var f = q.WaitForFrame() as DepthFrame)
                {                 
                     worldPoints = world.Calculate(f);
                    worldPoints.CopyTo(worldpoints);
}

====> worldpoints[i].x,worldpoints[i].y,worldpoints[i].z  ---->are always have the correct Pointcloud WorldCoordinates.

dorodnic, 2018-03-27:
Hi @telemetrieTP23 
Texture coordinates only make sense when you specify a second stream to map the texture from. 
Given color or IR frame, please try adding `world.MapTo(color);`. [pointcloud](https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud) C++ example can offer the high-level reference. If it doesn't work for you please post your code here and we will do our best to help.

telemetrieTP23, 2018-04-10:
Hey @dorodnic  !

I read the pointcloud c++ example and tried the same in c#:

// INITALIZING STREAMS START //
 depthSensor = dev.Sensors[0];
            sp = depthSensor.VideoStreamProfiles
                                    .Where(p => p.Stream == Intel.RealSense.Stream.Depth)
                                    .OrderByDescending(p => p.Framerate)
                                    .Where(p => p.Width == 640 && p.Height == 480)
                                    .First();
            depthSensor.Open(sp);
            depthSensor.Start(q);

            rgbSensor = dev.Sensors[1];
            spRGB = rgbSensor.VideoStreamProfiles
                                    .Where(p => p.Stream == Intel.RealSense.Stream.Color)
                                    .OrderByDescending(p => p.Framerate)
                                    .Where(p => p.Width == 640 && p.Height == 480 && p.Format == Format.Rgb8)
                                    .First();
            rgbSensor.Open(spRGB);
            rgbSensor.Start(q);
// INITALIZING STREAMS END //

 using (var f = q.WaitForFrames() as FrameSet)
                {
                    DepthFrame depthFrame = f.DepthFrame;
                    worldPoints = world.Calculate(depthFrame);
                    VideoFrame colorFrame = f.ColorFrame;
                    world.MapTexture(colorFrame);
                    worldPoints.CopyTo(texturepoints);
                  }

After the line "DepthFrame depthFrame = f.DepthFrame" in the "using block"  this runtimerror happens:

"Object does not support  "librealsense::composite_frame" interface! 
rs2_embedded_frames_count(composite:000000002F0087E8)"

Greetings

dorodnic, 2018-04-10:
What if you change the initialising block to:
```cpp
var p = new Pipeline();
p.Start();
```
and then loop on `using (var f = p.WaitForFrames() as FrameSet)`?
I think the frame queue object should not expose `WaitForFrames` method, will look into it.

RealSense-Customer-Engineering, 2018-05-23:
[Realsense Customer Engineering Team Comment]
@telemetrieTP23
have you tried the suggestion from dorodnic?

closed 2018-05-28 by RealSense-Customer-Engineering