realsenseai/librealsense #1446: cpuid.h is missing. Compiling for ODROID XU4

deadundead, 2018-03-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |   4.9.38                                         | 
| Platform                                 | ODROID XU4  |
| SDK Version                            | 2.0 }                          | 

### Issue Description
I stumbled across a strange issue building librealsense 2.0 SDK.
"cpuid.h : No such file or directory." That's true, locate cpuid.h gives blank result.
I solved it the dirty way - 
I found file image.cpp that includes cpuid.h and found out there is only one bool function that uses it. And i made it always return false. That did the job for me, i rebuilt the SDK, and realsense2camera also built just fine. So I`m leaving it here in case anyone gets this issue, since I found nothing helpful on the internet. 
I also kindly suggest that you make a "light" version of the SDK, as it is clear that everything works like a charm without additional functionality. I get depth information, and RGBD just fine.

Chunhai, 2018-03-27:
I did the same way! I think it is reasonable since the function always return FALSE for Android platform.

BTW, what do you mean "light" version of SDK? Is it just building librealsense core library but not examples?

deadundead, 2018-03-28:
It is unclear that stuff like kernel patches is unnecessary for core functionality. Thus, one may stumble upon the fact that neither of official odroid ubuntu kernels are supported by the librealsense2 kernel patches.  
By "light" version I mean the core functionality version only required to run the realsense2_camera node.
Maybe it should just be clearly stated on the instructions page what cmake options one has to use to get that result.

jallum, 2018-04-02:
I have a [patch](https://gist.github.com/jallum/85439ded2c9a78c8948fdceac179d695) for this.  It's a more general fix.

dorodnic, 2018-04-03:
Hi @jallum 
Thank you, you are welcomed to submit a pull-request with your patch if you'd like.

GustavZ, 2018-04-03:
i face the same error on the jetson tx2

dorodnic, 2018-04-03:
Hi @GustavZ 
The following PR broke the build: #1472 
Currently we don't have ARM platforms in our CI pool that are not Android