realsenseai/librealsense #1447: Request to save XYZ for each pixel during alignment

danielw256, 2018-03-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.08.15 | 
| Operating System & Version | Win 10 |          
| Platform                                 | PC |
| SDK Version                            | 2.9.0   | 

Z is calculated for each pixel in the aligned depth image. X and Y are calculated for 2 corners of a rectangle and can be interpolated for each pixel. Please create a new frame type can store the XYZ for each pixel and add an option to create this frame during alignment..

dorodnic, 2018-03-28:
Hello @danielw256 
We will try to incorporate this functionality. If I understand correctly, your goal is to save CPU by doing this calculation just once if you need both align and point-cloud. 
Is this correct?

danielw256, 2018-03-28:
Correct. Thanks

closed 2020-09-09 by RealSenseCustomerSupport