realsenseai/librealsense #1449: RealSense Viewer improvements

electro-logic, 2018-03-27:
Hello all,

- Sliders of parameters don't support keyboard UP/DOWN key or insertion of manual value so reaching a desired value is very difficult with only mouse support

- In 3D view zoom is limited, why not unlock it? I want to zoom far objects

Thank you

dorodnic, 2018-03-28:
Hi @electro-logic 

Thank you for your feedback. Both items are in our backlog and we are planning to address soon. 

I already did some work on item 1 (#1421) but was not satisfied with the code quality. I will rewrite it more cleanly, but if you would like to take a look at this PR, I would appreciate feedback on the user-experience. 

On the second item, navigation in the 3D viewer is powered by an invisible arc-ball and the zoom stops when you reach the center of the ball. I think we need to map mid-scroll to panning the arc-ball along the viewing vector, rather then scaling relative to ball center.

RealSense-Customer-Engineering, 2018-05-02:
[Realsense Customer Engineering Team Comment]
@electro-logic
The manual editing for parameter is implemented in the latest version 2.10.4, please have a test.

electro-logic, 2018-05-02:
Hello,

I have tested it and manual editing of parameters works well. 

Thank you

closed 2018-05-14 by RealSense-Customer-Engineering