realsenseai/librealsense #1450: cannnot register the server's public key

agari1229, 2018-03-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   |  | 
| Operating System & Version |   Linux (Ubuntu 16.04)   | 
| Kernel Version (Linux Only)    |  4.10.0-28-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |   | 

### Issue Description
hi, I attempted to install by referring to [this site](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md), but an error occurred.

when i run `sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE`
i got the error 
?: [fd 4]: read error: Connection reset by peer
gpgkeys: HTTP fetch error 7: couldn't connect: eof
↓
cannnot register the server's public key...

ev-mp, 2018-03-28:
Hello @agari1229 , 
even though the symptoms are not identical - please check if this could be due to firewall #1276

willemremie, 2018-04-03:
Experiencing similar issues, trying solution suggested by @ev-mp results in following output:

sudo apt-key adv --keyserver hkp://keys.gnupg.net:80 --recv-key 6F3EFCDE
Executing: /tmp/tmp.VMct9acML8/gpg.1.sh --keyserver
hkp://keys.gnupg.net:80
--recv-key
6F3EFCDE
gpg: requesting key 6F3EFCDE from hkp server keys.gnupg.net
gpgkeys: key 6F3EFCDE not found on keyserver
gpg: no valid OpenPGP data found.
gpg: Total number processed: 0
gpg: keyserver communications error: key not found
gpg: keyserver communications error: bad public key
gpg: keyserver receive failed: bad public key

RealSense-Customer-Engineering, 2018-04-17:
[Realsense Customer Engineering Team Comment] 
@willemremie, @agari1229
Do you still have problem retrieving the key? I tested on my Ubuntu 16.04 with both address start with "hkp://" or without it, both can retrieve the key without problem.

willemremie, 2018-04-17:
@RealSense-Customer-Engineering, the problem is still persistent, however, I've managed to retrieve the key at my home network. I wasn't able to do so on company network, even with the fix provided by @ev-mp.

willemremie, 2018-04-18:
Sure, I only wanted to mention it. Apparently some networks have dicculties retrieving the key. Maybe make a note in the documentation?

RealSense-Customer-Engineering, 2018-04-18:
[Realsense Customer Engineering Team Comment] 
Okay, I will follow up to add this into the document. Thank you.

closed 2018-04-18 by RealSense-Customer-Engineering

ArifSohaib, 2019-03-21:
sorry for opening this again but I had the same issue at my home network, I fixed it by going from the 5 GHz wifi to 2.4 Ghz