realsenseai/librealsense #1453: Increase point cloud density using librealsense

Alex9812, 2018-03-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Operating System & Version |   {Win (10)} | 
| Platform                                 | PC  |

### Issue Description

Is there a way to increase point cloud density using librealsense? I know it is possible to do using the viewer that is packaged but is there a way to do it through C++ and visual studio?

vladsterz, 2018-03-29:
@Alex9812 you can load high density preset I guess..

Alex9812, 2018-03-29:
@vladsterz How? I saw a post saying how to do it in python but I'm completely lost trying to figure out how to do it in C++

vladsterz, 2018-03-29:
Download presets from [here](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets) (or save them from viewer). So far you have a JSON file.

Now try this
`rs2::context cxt;`
`rs2::config cfg;`
`auto device = cxt.query_devices().back();`
`auto advanced_mode_dev = device_.as<rs400::advanced_mode>();`
`if (device_.is<rs400::advanced_mode>())`
		`if (!advanced_mode_dev.is_enabled())`
			`advanced_mode_dev.toggle_advanced_mode(true);`
`advanced_mode_dev.load_json(JSON.json);`
`pipe_.start(cfg);`

Alex9812, 2018-03-29:
@vladsterz Thanks for your help. I have having errors in the following places currently

[image]

Does it matter where I save my json file as long as I am loading it from a defined position? Also, I'm not sure why there is an error on rs400? I have these libraries loaded on
#include <librealsense2/rs_advanced_mode.h>
#include <librealsense2/rs.h>

Not too sure what's going on. It's a little difficult to configure for a beginner

vladsterz, 2018-03-29:
Since you are on windows, always give paths with \\ ( as C:\\blah\\blah.blah).
Hmmm, kinda weird. Try to include the .hpp files (If I remember correctly). 
Btw there is an underscore after device on line 4, change it to device. Sorry for this one, it was copy paste from a project of mine.

Alex9812, 2018-03-30:
@vladsterz That solved the issue but now it seems the stream isn't different at all from the default stream. Not sure what is going on.

vladsterz, 2018-03-30:
@Alex9812  Ok. I guess its the pipe. Try to get data like [this](https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md)

RealSense-Customer-Engineering, 2018-04-17:
[Realsense Customer Engineering Team Comment] 
@Alex9812 Is the issue resolved? Please update if you still need help.

closed 2018-05-24 by RealSense-Customer-Engineering