realsenseai/librealsense #1455: Export an organized point cloud

baptiste-mnh, 2018-03-29:
Hi,

I'm able to generate a PLY from my D415 but I need to get an organized point cloud such as a PCD, PTX or E57.
Is there a way to do it directly with th SDK 2.0 or PCL library?

Thanks in advance

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400                                                             | 
| Operating System & Version |   Linux (Ubuntu 16.04)                        | 
| Platform                                 | PC                                                            |
| SDK Version                            |             2.0                                          |

Alex9812, 2018-03-29:
use the rs-pcl example to convert realsense data to PCL. Once you're in PCL you can convert to PCD

dorodnic, 2018-04-01:
Hi @baptiste-mnh @Alex9812 
Adding the link to [pcl example](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/pcl). It's not perfect, but should give you a point to get started.

baptiste-mnh, 2018-04-03:
Thank you for your answers. I successfully generated a .pcd but I still having issues to convert it.

My steps
----------------
- Wait for a new frame
- Get the points with this code 
```
auto depth = frames.get_depth_frame();
auto points = pc.calculate(depth);
```
- Transform it PCL point cloud like in the exemple [rs-pcl](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/pcl)
`            ptr_cloud cloud = points_to_pcl(points);
`
- Save it
`pcl::io::savePCDFile(filename+".pcd", * cloud);
`

PCD Header :
--------------------
```
# .PCD v0.7 - Point Cloud Data file format
VERSION 0.7
FIELDS x y z rgb
SIZE 4 4 4 4
TYPE F F F U
COUNT 1 1 1 1
WIDTH 395805
HEIGHT 1
VIEWPOINT 0 0 0 1 0 0 0
POINTS 395805
DATA ascii
```
I easily can open it with CloudCompare.
Furthermore, my .pcd contain a lot of (-0, -0, 0)  lines.

But when I try to convert it as E57, I can't open it in Cloudcompare, it says that the box dimension is 0,0,0.
Is there an issue in my .pcd header or maybe I miss a step?

So I still can't open my laser capture in Capturing Reality. May be there is an other way to convert it in PTX for exemple?

Thanks!

RealSense-Customer-Engineering, 2018-04-27:
[Realsense Customer Engineering Team Comment]

@baptiste-mnh
Do you think this is still issue of LibRS? since you are now able save the PCD file from RealSense camera. For PCD file converts to E57, I read some topics few mentioned that E57 has richer feature saved than PCD, don't know if that cause the problem of converting, so maybe you should ask the question in PCL community.

baptiste-mnh, 2018-04-27:
Hi,
Yes, I still have the issues.
I will try to find the solution with PCL but it will be great if we could export directly in E57 format which is standard for laser scans.

RealSense-Customer-Engineering, 2018-05-14:
[Realsense Customer Engineering Team Comment]
@baptiste-mnh
Since this is additional feature request, I will discuss internally if it will be implemented.

baptiste-mnh, 2018-05-14:
Ok thank you. It could be a very useful feature for 3D scan.

closed 2018-08-06 by baptiste-mnh

vik-timmersion, 2019-03-07:
But here, PCD is not organized point cloud as HEIGHT 1
Is there an option to obtain organized point cloud from intelrealsense D415 and also intelrealsense SR300 ?

Thanks

> Thank you for your answers. I successfully generated a .pcd but I still having issues to convert it.
>
> ## My steps
> * Wait for a new frame
> * Get the points with this code
>
> ```
> auto depth = [...]

AndreV84, 2022-01-28:
@RealSense-Customer-Engineering @MartyG-RealSense 
so what is the status on e57 support? closed feature request?
How do we create e57 from realsense outputs?
Thanks

MartyG-RealSense, 2022-01-28:
@AndreV84   I checked the progress of the feature request.  As suggested in https://github.com/IntelRealSense/librealsense/issues/1455#issuecomment-388717544 there was an official Intel feature request created but it was closed in October 2021 because research determined that there was little public interest in the E57 format compared to the ply format and so it was decided not to proceed with creating the feature. 

In regard to how you might convert RealSense data into E57 format, **MeshLab** apparently introduced E57 import / export support in 2021.

https://github.com/cnr-isti-vclab/meshlab/issues/582#issuecomment-880862636
https://www.meshlab.net/

The link below has a guide to using a ply exported from RealSense in MeshLab to create a solid mesh.

https://www.andreasjakl.com/capturing-3d-point-cloud-intel-realsense-converting-mesh-meshlab/

AndreV84, 2022-01-28:
@MartyG-RealSense 
Thank you for following up
Yes, the recent Meshlab release supports e57 to a limited extent 
for example saving [exporting] .ply without faces as e57 will result in e57 that won't work
so as for now the solution that seems to work is to convert .ply with mesg to e57 using cloud compare
```
cloudcompare.CloudCompare -o examplePointcloud.ply -C_EXPORT_FMT E57 -EXTRACT_VERTICES
``` 
then I was able to import resulting e57 from realsense camera into isaac sim
but if there was e57 direct support it would be more convinient indeed
[image]

MartyG-RealSense, 2022-01-28:
It's excellent to hear that CloudCompare can provide the E57 conversion that you needed.

As an official librealsense feature request for E57 has already been declined for creation, it would not be considered again a second time unfortunately.