realsenseai/librealsense #1458: Error building librealsense from source linux

GustavZ, 2018-03-29:
i try to instal librealsense from source on the jetson tx2 for the D435. (ubuntu 16.04)

When building it stops at this point:
```
Scanning dependencies of target rs-pointcloud
[ 60%] Building CXX object examples/pointcloud/CMakeFiles/rs-pointcloud.dir/rs-pointcloud.cpp.o
[ 61%] Linking CXX executable rs-capture
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmCloseOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmMap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmUnmap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevices2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmOpenOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevice2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevices'
collect2: error: ld returned 1 exit status
examples/capture/CMakeFiles/rs-capture.dir/build.make:98: recipe for target 'examples/capture/rs-capture' failed
make[2]: *** [examples/capture/rs-capture] Error 1
CMakeFiles/Makefile2:315: recipe for target 'examples/capture/CMakeFiles/rs-capture.dir/all' failed
make[1]: *** [examples/capture/CMakeFiles/rs-capture.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 61%] Linking CXX executable rs-software-device
[ 61%] Linking CXX executable rs-multicam
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmCloseOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmMap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmUnmap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevices2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmOpenOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevice2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevices'
collect2: error: ld returned 1 exit status
examples/software-device/CMakeFiles/rs-software-device.dir/build.make:98: recipe for target 'examples/software-device/rs-software-device' failed
make[2]: *** [examples/software-device/rs-software-device] Error 1
CMakeFiles/Makefile2:260: recipe for target 'examples/software-device/CMakeFiles/rs-software-device.dir/all' failed
make[1]: *** [examples/software-device/CMakeFiles/rs-software-device.dir/all] Error 2
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmCloseOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmMap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmUnmap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevices2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmOpenOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevice2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevices'
collect2: error: ld returned 1 exit status
examples/multicam/CMakeFiles/rs-multicam.dir/build.make:98: recipe for target 'examples/multicam/rs-multicam' failed
make[2]: *** [examples/multicam/rs-multicam] Error 1
CMakeFiles/Makefile2:425: recipe for target 'examples/multicam/CMakeFiles/rs-multicam.dir/all' failed
make[1]: *** [examples/multicam/CMakeFiles/rs-multicam.dir/all] Error 2
[ 62%] Linking CXX executable rs-pointcloud
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmCloseOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmMap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmUnmap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevices2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmOpenOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevice2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevices'
collect2: error: ld returned 1 exit status
examples/pointcloud/CMakeFiles/rs-pointcloud.dir/build.make:98: recipe for target 'examples/pointcloud/rs-pointcloud' failed
make[2]: *** [examples/pointcloud/rs-pointcloud] Error 1
CMakeFiles/Makefile2:480: recipe for target 'examples/pointcloud/CMakeFiles/rs-pointcloud.dir/all' failed
make[1]: *** [examples/pointcloud/CMakeFiles/rs-pointcloud.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

what could be the reason?

ev-mp, 2018-03-29:
Hello @GustavZ ,
There is a known issue with Jetson TX2 and OpenGl installation, see the following discussion: 
https://devtalk.nvidia.com/default/topic/1026823/libgl-so-undefined-reference-error-when-building-opencv-3-3-0/?offset=2
Please check and update.

GustavZ, 2018-03-29:
So it seems opencv that comes flashed with jetPack3.2 is the problem.
Deleting it and rebuilding from source should do the trick.
Will try next week and report.

closed 2018-04-03 by GustavZ

jafet-chaves, 2021-02-12:
For someone in the future, I solved it in a TX2 with JetPack-3.2.1 by applying the following commands:

```
cd /usr/lib/aarch64-linux-gnu/
sudo ln -sf tegra/libGL.so libGL.so
```