realsenseai/librealsense #1459: Depth value of given pixel in RGB image

MathiasDePaepe, 2018-03-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | R200 | 
| Firmware Version                   | 1.0.71.06 | 
| Operating System & Version |   Linux (Ubuntu 16  | 
| SDK Version                            |  legacy | 

### Issue Description
Hello,

How to get the depth value of the RGB camera according to a given pixel?
I visited https://software.intel.com/en-us/articles/robotics-development-kit-r200-depth-data-interpretation, but I have no idea how to implement it.

Thanks on advance

RealSense-Customer-Engineering, 2018-03-30:
[Realsense Customer Engineering Team Comment] 
Hi @MathiasDePaepe
You can build and review the legacy librealsense example apps to show the depth image.
https://github.com/IntelRealSense/librealsense/tree/v1.12.1
https://github.com/IntelRealSense/librealsense/tree/v1.12.1/examples

MathiasDePaepe, 2018-04-10:
Hello thanks for asking,

I figured out how to get easily the depth information from the depthimage. When i was writing a python program to detect multiple objects, i got an output like that:

[image]

My program investigates the keypoints of the SURF algoritme and then look at the corresponding depth of it. After determining the location of the objects, i draw a bounding box. Now the problem is that i have a kind of a tunneling vision at the objects. You don't see i realy clear at the image, but when i take a picture of smaller object who are closer to the camera, you see it more clear. My question now is what is the cause of this?

From the DAC stream i noticed a little shift in pixels between the colorstream and the depthstream. The cause is the place of the seperate camera's on the R200. I solved this by triangular geometry and i have a correction of **a*2*d*tan(theta/2) / x** where a=58.2mm (distance between colorcamera/depthcamera), d=mean depth of object, theta is the horizontal FOV in radials (70 degrees). Is this the best way to solve this problem or is there an other manner?

Thanks on advance!