realsenseai/librealsense #1462: How to calculate invalid depth band?

UnaNancyOwen, 2018-03-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400                                                       |
| SDK Version                            |  2.x.x                                                      | 

### Issue Description
This is question.
The calculation method of Invalid Depth Band is written in [realsense D400 series data sheet](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Intel-RealSense-D400-Series-Datasheet.pdf) (pp.55-56).
I calculated according to this. But, It seems smaller value than actual invalid band.

Is my calculation wrong? What do you think?
```cpp
#include <librealsense2/rs.hpp>
#include <librealsense2/rsutil.h>
#include <array>
#include <cmath>
```
```cpp
rs2::pipeline pipeline;
rs2::pipeline_profile pipeline_profile = pipeline.start();

// BaseLine
rs2::frameset frameset = pipeline.wait_for_frames();
rs2::depth_frame depth_frame = frameset.get_depth_frame();
rs2::disparity_transform disparity_transform( true );
rs2::disparity_frame disparity_frame = disparity_transform.process( depth_frame );
float baseline = disparity_frame.get_baseline(); // mm

// Distance (e.g. 2.0m)
float distance = 2000; // mm

// Horizontal FOV
rs2::video_stream_profile video_stream_profile = pipeline_profile.get_stream( rs2_stream::RS2_STREAM_DEPTH ).as<rs2::video_stream_profile>();
rs2_intrinsics intrinsics = video_stream_profile.get_intrinsics();
std::array<float, 2> fov;
rs2_fov( &intrinsics, &fov[0] );
float horizontal_fov = fov[0] * M_PI / 180.0; // radian

// DBR
float distance_band_ratio = baseline / std::floor( 2.0 * distance * std::tan( horizontal_fov / 2.0 ) );

// IDB
float invalid_depth_band = depth_width * distance_band_ratio;
```

UnaNancyOwen, 2018-04-03:
This is depth image with IDB area painted in red based above calculation.
It seems smaller than the actual invalid area.

[image]

UnaNancyOwen, 2018-05-11:
@RealSense-Customer-Engineering Thanks for reply!
The actually calculated results using above source code is as follows

RealSense D415 (1280x720)
* baseline : 54.8575 [mm]
* distance : 2000 [mm]
* horizontal_fov : 1.19014 [radian] ( 68.19 [degree] )
* distance_band_ratio : 0.0202651
* invalid_depth_band : 25.9393 [pixel]

I think it improved a little bit than above screen shot.
(It was maybe affected by updates of Firmware or SDK.)
However, It seems a little small yet. Is it inevitable?

BTW, Why baseline and fov that retrieved using SDK is a little different with datasheet?

RealSense-Customer-Engineering, 2018-06-20:
[Realsense Customer Engineering Team Comment]
hi @UnaNancyOwen,

sorry for taking a long time to return to this topic, i do more test that confirm the value reading is different from each unit and within tolerance from spec. however, with more test, i think the IDB is a bit larger than our calculation, we will look into this.

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
hi @UnaNancyOwen,
per checked, the formula is correct, the equations represent absolute minimum limits; in many cases there may be additional invalid pixels appearing adjacent to the band region. The size of the additional invalid pixel region depends on the specific depth parameters and the scene.

UnaNancyOwen, 2018-08-10:
Thanks, RealSense Customer Engineering Team.
I understood it.

closed 2018-08-10 by UnaNancyOwen