realsenseai/librealsense #1463: Ubuntu 16.04 Kernel patch fails to build

ChrisHughes, 2018-03-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (N/A) | 
| Operating System & Version |   {Ubuntu 16.04 LTS  | 
| Kernel Version (Linux Only)    |  (4.4.76-aero-1.2)                                         | 
| Platform                                 | Aero RTF Drone  |
| SDK Version                            |  { 2.10.2 }                          | 

When trying to build 2.10.2 from source, the kernel patch fails with

`make: Leaving directory '/home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master'
cp: cannot stat '/home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/iio/accel/hid-sensor-accel-3d.ko': No such file or directory
`

The full log is here

`
ubuntu@ubuntu-CherryTrail:~/dev-sw/librealsense$ ./scripts/patch-realsense-ubuntu-xenial.sh

Remove all RealSense cameras attached. Hit any key when ready

[sudo] password for ubuntu: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.1ubuntu2).
git is already the newest version (1:2.7.4-0ubuntu1.3).
linux-headers-generic is already the newest version (4.4.0.116.122).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
The kernel has modified files:
	modified:   drivers/iio/accel/hid-sensor-accel-3d.c
	modified:   drivers/iio/gyro/hid-sensor-gyro-3d.c
	modified:   drivers/media/usb/uvc/Makefile
	modified:   drivers/media/usb/uvc/uvc_ctrl.c
	modified:   drivers/media/usb/uvc/uvc_driver.c
	modified:   drivers/media/usb/uvc/uvc_video.c
	modified:   drivers/media/usb/uvc/uvcvideo.h
	modified:   drivers/media/v4l2-core/v4l2-ioctl.c
	modified:   include/linux/hid-sensor-ids.h
	modified:   include/uapi/linux/videodev2.h
Proceeding will reset all local kernel changes. Press 'n' within 10 seconds to abort the operation
Do you want to proceed? [Y/n]

Update the folder content with the latest from mainline branch
remote: Total 0 (delta 0), reused 0 (delta 0)
From git://kernel.ubuntu.com/ubuntu/ubuntu-xenial
 * branch            master     -> FETCH_HEAD
Resetting local changes in ubuntu-xenial-master folder
 HEAD is now at 855cff5 UBUNTU: Ubuntu-4.4.0-116.140
Applying realsense-uvc patch
patching file drivers/media/usb/uvc/Makefile
patching file drivers/media/usb/uvc/uvc_driver.c
patching file drivers/media/usb/uvc/uvcvideo.h
patching file drivers/media/v4l2-core/v4l2-ioctl.c
patching file include/uapi/linux/videodev2.h
Applying realsense-metadata patch
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 2739 with fuzz 2 (offset 40 lines).
patching file drivers/media/usb/uvc/uvc_video.c
Hunk #1 succeeded at 1233 (offset 5 lines).
Hunk #2 succeeded at 1271 (offset 5 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 182 (offset 27 lines).
Hunk #2 succeeded at 202 (offset 27 lines).
Applying realsense-hid patch
patching file drivers/iio/accel/hid-sensor-accel-3d.c
patching file drivers/iio/gyro/hid-sensor-gyro-3d.c
[... 267 lines cut ...]
make: Entering directory '/home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master'
  CC [M]  /home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/media/v4l2-core/videobuf2-v4l2.o
  CC [M]  /home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/media/v4l2-core/videobuf2-memops.o
  CC [M]  /home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/media/v4l2-core/videobuf2-vmalloc.o
  Building modules, stage 2.
  MODPOST 4 modules
WARNING: "__x86_indirect_thunk_rax" [/home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/media/v4l2-core/videobuf2-vmalloc.ko] undefined!
WARNING: "__x86_indirect_thunk_rdx" [/home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/media/v4l2-core/videobuf2-v4l2.ko] undefined!
WARNING: "__x86_indirect_thunk_rax" [/home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/media/v4l2-core/videobuf2-v4l2.ko] undefined!
WARNING: "__x86_indirect_thunk_rax" [/home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/media/v4l2-core/videobuf2-memops.ko] undefined!
WARNING: "__x86_indirect_thunk_rcx" [/home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/media/v4l2-core/videobuf2-core.ko] undefined!
WARNING: "__x86_indirect_thunk_rdx" [/home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/media/v4l2-core/videobuf2-core.ko] undefined!
WARNING: "__x86_indirect_thunk_r8" [/home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/media/v4l2-core/videobuf2-core.ko] undefined!
WARNING: "__x86_indirect_thunk_rax" [/home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/media/v4l2-core/videobuf2-core.ko] undefined!
  CC      /home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/media/v4l2-core/videobuf2-core.mod.o
  CC      /home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/media/v4l2-core/videobuf2-memops.mod.o
  CC      /home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/media/v4l2-core/videobuf2-v4l2.mod.o
  CC      /home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/media/v4l2-core/videobuf2-vmalloc.mod.o
  LD [M]  /home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/media/v4l2-core/videobuf2-core.ko
  LD [M]  /home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/media/v4l2-core/videobuf2-v4l2.ko
  LD [M]  /home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/media/v4l2-core/videobuf2-vmalloc.ko
  LD [M]  /home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/media/v4l2-core/videobuf2-memops.ko
make: Leaving directory '/home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master'
cp: cannot stat '/home/ubuntu/dev-sw/librealsense/ubuntu-xenial-master/drivers/iio/accel/hid-sensor-accel-3d.ko': No such file or directory
`

ev-mp, 2018-04-01:
Hello @ChrisHughes ,
The kernel patches are meant for  Ubuntu distributions with LTS kernels.
The board you're using runs a different dedicated `4.4.76-aero-1.2` kernel as you mention.

Please also see a related #1320 with instructions on how to integrate D400/Librealsense2 with Aero board

ev-mp, 2018-04-05:
Hello @ChrisHughes , please update whether you need a further assistance with this.

ChrisHughes, 2018-04-05:
Thanks, the patch on #1320 worked well and solved the issue for me. Would be helpful to have it included officially and in the build docs.

closed 2018-04-05 by ChrisHughes