realsenseai/librealsense #1464: realsense-viewer Illegal instruction (core dumped)

smishraasu, 2018-03-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
| Camera Model                       | { D415 } | 
| Firmware Version                   | | 
| Operating System & Version |Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |4.13.0-36-generic| 
| Platform                                 |UP board|

Hi,

I have a D415 and am trying to use realsense-viewer but I get "Illegal instruction (core dumped)" everytime I run "realsense-viewer". I used the following page to install dependencies: 

https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md

Please let me know what could be done to resolve this. 

Thanks.

ev-mp, 2018-04-01:
Hello @smishraasu, this error is probably has to do with #1428.
Providing the core dump would allow to look deeper in to it though.

smishraasu, 2018-04-01:
hi @ev-mp,

Thank you for the response. Please find the gdb output below for more information.

Reading symbols from realsense-viewer...(no debugging symbols found)...done.
(gdb) run
Starting program: /usr/bin/realsense-viewer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGILL, Illegal instruction.
0x00007ffff787703b in librealsense::logger_type::logger_type() ()
   from /usr/lib/x86_64-linux-gnu/librealsense2.so.2
(gdb)

ev-mp, 2018-04-02:
Hi @smishraasu , the debug symbols are part of a [separate ](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) 'librealsense2-dbg' package.

Can you rerun gdb with the symbols installed ?
Alternatively, you can upload and/or share a link to the full dump file.

smishraasu, 2018-04-02:
`Starting program: /usr/bin/realsense-viewer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGILL, Illegal instruction.
librealsense::logger_type::logger_type (this=0x7ffff7d8f640 <librealsense::logger>) at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/src/log.cpp:81
81	/home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/src/log.cpp: No such file or directory.
(gdb) disas
Dump of assembler code for function librealsense::logger_type::logger_type():
   0x00007ffff7877010 <+0>:	push   %r15
   0x00007ffff7877012 <+2>:	push   %r14
   0x00007ffff7877014 <+4>:	push   %r13
   0x00007ffff7877016 <+6>:	push   %r12
   0x00007ffff7877018 <+8>:	push   %rbp
   0x00007ffff7877019 <+9>:	push   %rbx
   0x00007ffff787701a <+10>:	lea    0x130(%rdi),%rbp
   0x00007ffff7877021 <+17>:	mov    %rdi,%rbx
   0x00007ffff7877024 <+20>:	sub    $0x308,%rsp
   0x00007ffff787702b <+27>:	movq   $0x0,0x10(%rdi)
   0x00007ffff7877033 <+35>:	movq   $0x0,0x18(%rdi)
=> 0x00007ffff787703b <+43>:	vmovdqa 0x22570d(%rip),%xmm0        # 0x7ffff7a9c750
   0x00007ffff7877043 <+51>:	mov    %fs:0x28,%rax
   0x00007ffff787704c <+60>:	mov    %rax,0x2f8(%rsp)
   0x00007ffff7877054 <+68>:	xor    %eax,%eax
   0x00007ffff7877056 <+70>:	vmovups %xmm0,(%rdi)
   0x00007ffff787705a <+74>:	lea    0x38(%rdi),%rax
   0x00007ffff787705e <+78>:	movq   $0x0,0x20(%rdi)
   0x00007ffff7877066 <+86>:	movq   $0x0,0x28(%rdi)
   0x00007ffff787706e <+94>:	mov    %rax,%r15
   0x00007ffff7877071 <+97>:	movq   $0x0,0x30(%rdi)
   0x00007ffff7877079 <+105>:	mov    %rbp,%rdi
   0x00007ffff787707c <+108>:	mov    %rax,0x20(%rsp)
   0x00007ffff7877081 <+113>:	callq  0x7ffff776e5a0 <_ZNSt8ios_baseC2Ev@plt>
   0x00007ffff7877086 <+118>:	mov    0x50cba3(%rip),%rax        # 0x7ffff7d83c30
   0x00007ffff787708d <+125>:	movb   $0x0,0x210(%rbx)
   0x00007ffff7877094 <+132>:	mov    %r15,%rdi
   0x00007ffff7877097 <+135>:	movq   $0x0,0x208(%rbx)
   0x00007ffff78770a2 <+146>:	movb   $0x0,0x211(%rbx)
   0x00007ffff78770a9 <+153>:	xor    %esi,%esi
   0x00007ffff78770ab <+155>:	movq   $0x0,0x218(%rbx)
   0x00007ffff78770b6 <+166>:	movq   $0x0,0x220(%rbx)
   0x00007ffff78770c1 <+177>:	add    $0x10,%rax
   0x00007ffff78770c5 <+181>:	movq   $0x0,0x228(%rbx)
   0x00007ffff78770d0 <+192>:	movq   $0x0,0x230(%rbx)
   0x00007ffff78770db <+203>:	mov    %rax,0x130(%rbx)
   0x00007ffff78770e2 <+210>:	mov    0x50d587(%rip),%rax        # 0x7ffff7d84670
   0x00007ffff78770e9 <+217>:	mov    0x8(%rax),%r12
   0x00007ffff78770ed <+221>:	mov    0x10(%rax),%r14
   0x00007ffff78770f1 <+225>:	add    -0x18(%r12),%rdi
   0x00007ffff78770f6 <+230>:	mov    %r12,0x38(%rbx)
   0x00007ffff78770fa <+234>:	mov    %r14,(%rdi)
   0x00007ffff78770fd <+237>:	callq  0x7ffff7774290 <_ZNSt9basic_iosIcSt11char_traitsIcEE4initEPSt15basic_streambufIcS1_E@plt>
   0x00007ffff7877102 <+242>:	mov    0x50e5a7(%rip),%rax        # 0x7ffff7d856b0
   0x00007ffff7877109 <+249>:	lea    0x40(%rbx),%r13
---Type <return> to continue, or q <return> to quit---
   0x00007ffff787710d <+253>:	mov    %r13,%rdi
   0x00007ffff7877110 <+256>:	lea    0x18(%rax),%rdx
   0x00007ffff7877114 <+260>:	add    $0x40,%rax
   0x00007ffff7877118 <+264>:	mov    %rax,0x130(%rbx)
   0x00007ffff787711f <+271>:	mov    %rdx,0x38(%rbx)
   0x00007ffff7877123 <+275>:	callq  0x7ffff77728e0 <_ZNSt13basic_filebufIcSt11char_traitsIcEEC1Ev@plt>
   0x00007ffff7877128 <+280>:	mov    %r13,%rsi
   0x00007ffff787712b <+283>:	mov    %rbp,%rdi
   0x00007ffff787712e <+286>:	callq  0x7ffff7774290 <_ZNSt9basic_iosIcSt11char_traitsIcEE4initEPSt15basic_streambufIcS1_E@plt>
   0x00007ffff7877133 <+291>:	mov    0x50d516(%rip),%rax        # 0x7ffff7d84650
   0x00007ffff787713a <+298>:	movq   $0x0,0x240(%rbx)
   0x00007ffff7877145 <+309>:	mov    %rax,0x238(%rbx)
   0x00007ffff787714c <+316>:	lea    0x70(%rsp),%rax
   0x00007ffff7877151 <+321>:	mov    %rax,%rdi
   0x00007ffff7877154 <+324>:	mov    %rax,(%rsp)
[... 766 lines cut ...]
   0x00007ffff7877fb6 <+4006>:	add    $0x10,%rax
   0x00007ffff7877fba <+4010>:	mov    %rax,0x188(%rsp)
   0x00007ffff7877fc2 <+4018>:	lea    0x60(%r15),%rax
   0x00007ffff7877fc6 <+4022>:	cmp    %rax,%rdi
   0x00007ffff7877fc9 <+4025>:	je     0x7ffff787801f <librealsense::logger_type::logger_type()+4111>
   0x00007ffff7877fcb <+4027>:	vzeroupper 
   0x00007ffff7877fce <+4030>:	callq  0x7ffff77721b0 <_ZdlPv@plt>
   0x00007ffff7877fd3 <+4035>:	mov    0x50c4fe(%rip),%rax        # 0x7ffff7d844d8
   0x00007ffff7877fda <+4042>:	lea    0x40(%r15),%rdi
   0x00007ffff7877fde <+4046>:	add    $0x10,%rax
   0x00007ffff7877fe2 <+4050>:	mov    %rax,0x188(%rsp)
   0x00007ffff7877fea <+4058>:	callq  0x7ffff7774a60 <_ZNSt6localeD1Ev@plt>
   0x00007ffff7877fef <+4063>:	mov    -0x18(%r13),%rax
   0x00007ffff7877ff3 <+4067>:	mov    %r14,0x180(%rsp,%rax,1)
   0x00007ffff7877ffb <+4075>:	mov    0x50bc2e(%rip),%rax        # 0x7ffff7d83c30
   0x00007ffff7878002 <+4082>:	lea    0x70(%r15),%rdi
   0x00007ffff7878006 <+4086>:	add    $0x10,%rax
   0x00007ffff787800a <+4090>:	mov    %rax,0x1f0(%rsp)
   0x00007ffff7878012 <+4098>:	vzeroupper 
   0x00007ffff7878015 <+4101>:	callq  0x7ffff776e9d0 <_ZNSt8ios_baseD2Ev@plt>
   0x00007ffff787801a <+4106>:	jmpq   0x7ffff7877ee2 <librealsense::logger_type::logger_type()+3794>
   0x00007ffff787801f <+4111>:	vzeroupper 
   0x00007ffff7878022 <+4114>:	jmp    0x7ffff7877fd3 <librealsense::logger_type::logger_type()+4035>
End of assembler dump.
`

smishraasu, 2018-04-03:
Hi @ev-mp ,

I managed to build the v2.10.0. Now I can open up realsense-viewer. Please take a look at the screenshot below.
[image]

 Something which concerns me is that, I dont see a tab for RGB camera like we see one in the snapshot below which is from the following link: https://software.intel.com/en-us/realsense/d400/get-started

dorodnic, 2018-04-03:
Hi @smishraasu 
Please note that the camera was identified as a USB2 device. If you are using the UpBoard, please use the USB OTG connector rather then the 4 regular ones on the side of the board. 
USB2 imposes additional constraints on what can and what cannot be streamed.

smishraasu, 2018-04-03:
Hi @dorodnic ,

Thank you for the response. I connected the USB OTG connector and I connected the D415 to the side of the board but still I see the output at a USB2 device. Please find the screenshot of the realsense viewer and a photo of the setup below. Please let me know if any drivers need to be installed before using the USB 3.0 port on the upboard.

Screenshot of RealsenseViewer - 

[image]

Photo of setup with D415 connected to USB 3.0 port through the USB-OTG (blue) connector - 
[image]

smishraasu, 2018-04-03:
Hi @dorodnic ,

I had 1 more question. The specs say that D415 provides depth stream at a resolution of Up to 1280 x 720 . But I am getting a resolution of 480 x 270. Is that because the device is detected as usb2 ?