realsenseai/librealsense #1465: The bug of DisparityTransform

HungLearner, 2018-03-31:
Required Info |  
-- | --
Camera Model | D415
Operating System & Version | Windows 10
Platform | PC
Compiler Enviorment| C#
SDK Version | SDK 2.0 (Build 2.10.1)

When I use "Align" fix Depth FOV, and then use "DisparityTransform" to apply filter depth frame.

The output size will half.

For example, if input of the size is 640*480, output will be 320*240.

This is my code：

```
Align align = new Align(Stream.Color);
FrameSet _frames = align.Process(f);

DecimationFilter Decim = new DecimationFilter();
Decim.Options[Option.FilterMagnitude].Value = 2.0f;
var Dec_Depth = Decim.ApplyFilter(_frames.DepthFrame);

var Colorized_Depth = Colorize.Colorize(Dec_Depth);
```

dorodnic, 2018-04-01:
Hi @HungLearner 
I'm not sure what you mean. Reducing spatial resolution is the intent of `DecimationFilter`. Also, I'm not sure how is this related to `DisparityTransform`. 
Please give more information on what you need to happen.

HungLearner, 2018-04-02:
I'm sorry, the topic is not correct, I modified.
I mean where my depth resolution is 1280 x 720 and color resolution is 640 x 480.
Then use align to convert depth with color, the new resolution of depth will be 640 x 480.
But when I use "DecimationFilter" to apply this depth image, resolution will half(320 x 240).

HungLearner, 2018-04-03:
@dorodnic 
oh.....
I mistake this function mean, sorry.
So, this problem can be closed, thanks.

dorodnic, 2018-04-03:
@HungLearner glad this helped. I suspect there still might be issues with combining post-processing and alignment in C#, so feel free to open more issues.

closed 2018-04-03 by dorodnic