realsenseai/librealsense #1466: How can I get pointclouds as numpy object?

jtoyama4, 2018-04-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Operating System & Version |  Linux (Ubuntu 16 | 
| Kernel Version (Linux Only)    |  (e.g. 4.10.17)                                         | 
| Platform                                 | PC/ |
| SDK Version                            | 2.0                         | 
|Python Version                  | 3.5                             |

I want to get pointcloud as numpy object.

I supposed that the pointcloud can be obtained as below,

```
pc = rs.pointcloud()
points = rs.points()
pipe = rs.pipeline()
pipe.start()

frames = pipe.wait_for_frames()
depth = frames.get_depth_frame()
color = frames.get_color_frame()

pc.map_to(color)
points = pc.calculate(depth)
points_data = points.get_data()
np_points = np.asanyarray(points_data)

print(np_points)
```

but this code returns an empty list.

I checked that the points are successfully obtained by using export_ply_example.py.

Do you have any ideas to solve it?

zivsha, 2018-04-02:
You need to take the points not data of the points.
```py
np_points = np.asanyarray(points)
```

zivsha, 2018-04-02:
You should use `get_vertices`:
```py
pc = rs.pointcloud()
pc.map_to(color)
points = pc.calculate(depth)
vtx = np.asanyarray(points.get_vertices())
tex = np.asanyarray(points.get_texture_coordinates())
```

jtoyama4, 2018-04-02:
@zivsha 

Thank you so much! It successfully worked.

closed 2018-04-03 by dorodnic