realsenseai/librealsense #1467: No device found that satisfies all requirements && Frame did not arrive in time!

GuoJiechang, 2018-04-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | RS410| 
| Firmware Version                   | 05.08.06.00 | 
| Operating System & Version |   {Win (10)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  RealSense SDK 2.0                      | 

### Issue Description
Hi all,
I'm a beginner to the RealSense, and my devices is RS410, I tried to work on the unity wrapper of the RealSense SDK2.0 however it not work well.

The first problem exists is when I start the game, exception existed.
`Exception: Failed to resolve request. No device found that satisfies all requirements
rs2_config_resolve(config:000000005FC90D30, pipe:000000005FC90E90)
Intel.RealSense.Helpers+ErrorMarshaler.MarshalNativeToManaged (IntPtr pNativeData)
Intel.RealSense.Config.Resolve (Intel.RealSense.Pipeline pipe)
RealSenseDevice.Awake () (at Assets/RealSenseSDK2.0/Scripts/RealSenseDevice.cs:89)
`

I tried to fixed it by annotation the code (related to Stream.Color) in RealSenseDevices.cs, because I think the device I use cannot caputure RGB data(fix me if I wrong)

`public RealSenseConfiguration DeviceConfiguration = new RealSenseConfiguration
    {
        Profiles = new VideoStreamRequest[] {
            new VideoStreamRequest {Stream = Stream.Depth, StreamIndex = -1, Width = 640, Height = 480, Format = Format.Z16 , Framerate = 30 },
            new VideoStreamRequest {Stream = Stream.Infrared, StreamIndex = -1, Width = 640, Height = 480, Format = Format.Y8 , Framerate = 30 },
           // new VideoStreamRequest {
                //Stream = Stream.Color, StreamIndex = -1, Width = 640, Height = 480, Format = Format.Rgb8 , Framerate = 30 
           // }
        }
    };
`

and change the device configuration property on the inspector of RealsenseDevices, change the Profiles size from 3 to 2.

The problem seems fixed. However another problem exist,

`Frame did not arrive in time!
rs2_wait_for_frame(queue:000000006F531E00)
UnityEngine.Debug:LogError(Object)
RealSenseDevice:ProcessFrames(FrameSet) (at Assets/RealSenseSDK2.0/Scripts/RealSenseDevice.cs:169)
RealSenseDevice:Worker_DoWork(Object, DoWorkEventArgs) (at Assets/RealSenseSDK2.0/Scripts/RealSenseDevice.cs:202)
System.ComponentModel.BackgroundWorker:ProcessWorker(Object, AsyncOperation, SendOrPostCallback)`

I can see some pictures, but it not smooth, seems not real time.

PLS, help me work with these problems.

Thank you!

Cheers,
Jiechang

dorodnic, 2018-04-02:
Hi @GuoJiechang 
It seems you are working with some sort of a pre-production device (based on the model and firmware version). Assuming you received it from Intel representative, perhaps you could ask him to replace it with the D415 (same but with RGB camera)? 
The Unity wrapper was designed and tested with the peripherals (D415 & D435 & SR300). While it is certainly possible to make it work with other RealSense devices it is likely to require some troubleshooting. 
If replacement is not an option, first thing I would do is check if the camera is functioning properly in the [RealSense Viewer](https://github.com/IntelRealSense/librealsense/releases/download/v2.10.2/Intel.RealSense.Viewer.exe). You should also consider updating to the [latest firmware](https://downloadcenter.intel.com/download/27514?v=t).

GuoJiechang, 2018-04-02:
Hi @dorodnic 
Sure, I think the device I work with is pre-production device and I just figure out the way to fixed these problems. I'll report it later. 
Anyway, Thank you very much.

GuoJiechang, 2018-04-02:
The devices I used is a pre-production which with no RGB camera, if someone who have the same problem,  maybe you could try the following steps,
1. Annotation the code in the RealSenseDevices.cs 
`public RealSenseConfiguration DeviceConfiguration = new RealSenseConfiguration
    {
        Profiles = new VideoStreamRequest[] {
            new VideoStreamRequest {Stream = Stream.Depth, StreamIndex = -1, Width = 640, Height = 480, Format = Format.Z16 , Framerate = 30 },
            new VideoStreamRequest {Stream = Stream.Infrared, StreamIndex = -1, Width = 640, Height = 480, Format = Format.Y8 , Framerate = 30 },
           // new VideoStreamRequest {
                //Stream = Stream.Color, StreamIndex = -1, Width = 640, Height = 480, Format = Format.Rgb8 , Framerate = 30 
           // }
        }
    };`
2、Change the Size（= 2） of Profiles on the RealsenseDevice's inspector
3、Inactive some gameobjects(which related to the Color stream): 
RealsenseDevice::Color,
RealsenseDevice::Alignment, 
Images::ColorImage.
Then....the project works well.

closed 2018-04-02 by GuoJiechang