realsenseai/librealsense #1477: lastest Unity DEMO pointclound  ：out of memory and unity crash！

Zhangliuwei, 2018-04-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)  |                                    | 
| Platform                                 | unity |
| SDK Version                            |  latest }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

dorodnic, 2018-04-04:
Hi @Zhangliuwei 
We are looking into the issue, but did not find decent solution yet. It is related to recent textured pointcloud feature. Disabling it can offer a work-around while we find a better fix.

pdm1302, 2018-04-09:
Hi @Zhangliuwei 
I also experienced the same issue. After isolating and testing different parts of [PointCloudGenerator.cs](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/unity/Assets/RealSenseSDK2.0/Scripts/PointCloudGenerator.cs), I have found that the reason may be how `FrameQueue pointsQueue` manages its frames. At the line 69 (`pointsQueue.Enqueue(points);`), the reference count of `points` is increased by `pointsQueue` but I could not find where it is decreased in the native code.

So I think that garbage collector is not able to clean up unused memory of `pointsQueue`. I have decided not to use this queue and implemented two alternative queues to store the vertices and texture coordinates instead. It seems to work well so far **if running on Unity Thread**. I haven't found how to solve the issue when processing the frames with Multithread.

Here is my alternative implementation of **PointCloudGenerator.cs**

    public class PointCloudGenerator : MonoBehaviour
    {
        private ParticleSystem.Particle[] particles = new ParticleSystem.Particle[0];
        private PointCloud pc = new PointCloud();
        //FrameQueue pointsQueue = new FrameQueue(1);
        Points.Vertex[] vertices;
        Points.TextureCoordinate[] textureCoordinate;
        private byte[] lastColorImage;
        Texture2D colorTexture;
        private int colorFrameWidth;
        private int colorFrameHeight;

        public bool mirrored;
        public float pointsSize = 0.01f;
        public int skipParticles = 2;
        public ParticleSystem pointCloudParticles;

        Queue textureQueue = Queue.Synchronized(new Queue());
        Queue vertexQueue = Queue.Synchronized(new Queue());

        // Use this for initialization
        void Start()
        {
            //RealSenseDevice.Instance.onNewSample += OnFrame;
            RealSenseDevice.Instance.onNewSampleSet += OnFrames;

        }
        object l = new object();
        private void OnFrames(FrameSet frames)
        {
            if(frames.DepthFrame == null)
            {
                Debug.Log("No depth frame in frameset, can't create point cloud");
                return;
            }

            if (!UpdateParticleParams(frames.DepthFrame.Width, frames.DepthFrame.Height))
            {
                Debug.Log("Unable to create point cloud");
                return;
            }
        
            using (var points = pc.Calculate(frames.DepthFrame))
            {
                if (frames.ColorFrame != null)
                {
                    if (frames.ColorFrame.BitsPerPixel == 24)
                    {
                        pc.MapTexture(frames.ColorFrame);
                        colorFrameWidth = frames.ColorFrame.Width;
                        colorFrameHeight = frames.ColorFrame.Height;
                        var newSize = frames.ColorFrame.Stride * colorFrameHeight;
                        lock (l)
                        {
                            if (lastColorImage == null || lastColorImage.Length != newSize)
                                lastColorImage = new byte[newSize];

                            frames.ColorFrame.CopyTo(lastColorImage);
                        }
                    }
                }

                Points.Vertex[] tmpvertices = new Points.Vertex[points.Count];
[... 104 lines cut ...]
                Debug.Assert(vertices.Length == particles.Length);
                int mirror = mirrored ? -1 : 1;
                for (int index = 0; index < vertices.Length; index += skipParticles)
                {
                    var v = vertices[index];
                    if (v.z > 0)
                    {
                        particles[index].position = new Vector3(v.x * mirror, v.y, v.z);
                        particles[index].startSize = pointsSize;
                        particles[index].startColor = colorTexture.GetPixelBilinear(textureCoordinate[index].u, textureCoordinate[index].v);
                    }
                    else //Required since we reuse the array
                    {
                        particles[index].position = Vector3.zero;
                        particles[index].startSize = 0;
                        particles[index].startColor = Color.black;
                    }
                }

            }

            //Either way, update particles
            pointCloudParticles.SetParticles(particles, particles.Length);
        }
    }

RealSense-Customer-Engineering, 2018-06-11:
[Realsense Customer Engineering Team Comment]
hi @Zhangliuwei
would like to confirm if solution for #1394 works for you?
thanks.

RealSense-Customer-Engineering, 2018-06-21:
[Realsense Customer Engineering Team Comment]
let me close this ticket here and please feel free to create new ticket if any question.

closed 2018-06-21 by RealSense-Customer-Engineering