realsenseai/librealsense #1480: Export IR and RGB image from D400 to PNG

baptiste-mnh, 2018-04-04:
Hi,
I need to get the images of the 2 IR cameras and the rgb camera. 
For now, I successfully export the IR left and right image using OpenCV but I have issue with the RGB one.

I think the format CV_8UC1 doesn't work with the colored, I tried different format but the image still weird.

Thanks for your help. 

Code
---------------
```
rs2::pipeline p;
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_INFRARED, 1, 1280, 720, RS2_FORMAT_Y8, 30);
cfg.enable_stream(RS2_STREAM_INFRARED, 2, 1280, 720, RS2_FORMAT_Y8, 30);
cfg.enable_stream(RS2_STREAM_COLOR, 1920, 1080, RS2_FORMAT_BGR8, 30);

rs2::frameset frames = pipeline.wait_for_frames();

auto ir_frame_left = frames.get_infrared_frame(1);
auto ir_frame_right = frames.get_infrared_frame(2);
auto depth = frames.get_depth_frame();
auto colored_frame = frames.get_color_frame();

//save files
cv::Mat dMat_left = cv::Mat(cv::Size(1280, 720), CV_8UC1, (void*)ir_frame_left.get_data());
cv::Mat dMat_right = cv::Mat(cv::Size(1280, 720), CV_8UC1, (void*)ir_frame_right.get_data());
cv::Mat dMat_colored = cv::Mat(cv::Size(1920, 1080), CV_8UC1, (void*)colored_frame.get_data());

cv::imwrite( "/home/baptiste-mnh/Documents/irLeft0.png", dMat_left );
cv::imwrite( "/home/baptiste-mnh/Documents/irRight0.png", dMat_right );
cv::imwrite( "/home/baptiste-mnh/Documents/coloredCV0.png", dMat_colored);
```
IR result
-------------
> [image]
> Left image

>[image]
> Right Image

Color result
-------------------
> [image]
>Colored result

UnaNancyOwen, 2018-04-04:
<code>RS2_FORMAT_BGR8</code> is 8-bits, 3-channels (<code>CV_8UC3</code>).
```patch
- cv::Mat dMat_colored = cv::Mat(cv::Size(1920, 1080), CV_8UC1, (void*)colored_frame.get_data());
+ cv::Mat dMat_colored = cv::Mat(cv::Size(1920, 1080), CV_8UC3, (void*)colored_frame.get_data());
```

baptiste-mnh, 2018-04-04:
Thanks!
Do you know the format for Depth?

UnaNancyOwen, 2018-04-04:
If you specified <code>RS2_FORMAT_Z16</code> (16-bits, 1-channels), You can use <code>CV_16SC1</code>.

baptiste-mnh, 2018-04-04:
It works! Thanks a lot.

baptiste-mnh, 2018-04-04:
Here is what I get for the color
---------------------------------------------------------------
```cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8, 30);```

```
cv::Mat dMat_colored = cv::Mat(cv::Size(1920, 1080), CV_8UC3, (void*)colored_frame.get_data());
cv::imwrite( "/home/baptiste-mnh/Documents/coloredCV"+std::to_string(nCapture)+".png", dMat_colored);
```

[image]

Here is what I get for the depth
-----------------------------------------------------------------
```cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);```

```
cv::Mat dMat_depth = cv::Mat(cv::Size(1280, 720), CV_16SC1, (void*)colored_frame.get_data());
cv::imwrite( "/home/baptiste-mnh/Documents/depth"+std::to_string(nCapture)+".png", dMat_depth);
```

[image]

baptiste-mnh, 2018-04-04:
Depth looks now ok but I still have an issue with the colored one.

Code
---------------
Stream parameters : 
```
cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);
cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8, 30);
```
Reccuperation code : 
```
auto depth = frames.get_depth_frame();
auto colored_frame = frames.get_color_frame();

cv::Mat dMat_colored = cv::Mat(cv::Size(1920, 1080), CV_8UC3, (void*)colored_frame.get_data());
cv::Mat dMat_depth = cv::Mat(cv::Size(1280, 720), CV_16SC1, (void*)depth.get_data());
```
Result
-------------
> [image]
> dMat_colored

> [image]
> depth1

baptiste-mnh, 2018-04-05:
It works on another PC, thanks.

closed 2018-04-05 by baptiste-mnh

Sreedhar-P, 2018-11-19:
Can anyone help me, how can proceed further to get individual images from the frames which are came from IR cameras using OpenCV-python

Code:
import cv2
import pyrealsense2 as rs
import numpy as np
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.infrared, 1, 1280, 720, rs.format.y8, 30)
config.enable_stream(rs.stream.infrared, 2, 1280, 720, rs.format.y8, 30)
pipeline.start(config)

frames = pipeline.wait_for_frames()
'''ir1_frame = frames.get_infrared_frame(1) # Left IR Camera, it allows 0, 1 or no input
ir2_frame = frames.get_infrared_frame(2) # Right IR camera'''

for frame in pipeline.wait_for_frames():
    print (frame.profile)

for frame in pipeline.wait_for_frames():
    print (frame.profile.stream_type(), frame.profile.stream_index())

ir1_frame = frames.get_infrared_frame(1)
ir2_frame=frames.get_infrared_frame(2)