realsenseai/librealsense #1481: SR300 depth sensor is not working: Frame didn't arrived within 5000

fspindle, 2018-04-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300  | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |  Ubuntu 16.04.4 LTS  | 
| Kernel Version (Linux Only)    |  4.13.0-38-generic x86_64                                       | 
| Platform                                 | HP EliteBook laptop  |
| SDK Version                            |  2.10.2 from master, commit 3c1743cc6b0ac6f91                          | 

### Issue Description

I followed the [tutorial](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) including the patch installation using `./scripts/patch-realsense-ubuntu-xenial.sh`.

After installation and reboot I have the following issue:

````
$ ./rs-capture 
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x276af50):
    Frame didn't arrived within 5000
````
As described in #892 I tried to connect my SR300 to a USB3 powered hub, but it doesn't solve the issue.

When I launch `realsense-viewer` I got the following logs  in the terminal:
````
$ ./realsense-viewer 
 04/04 15:14:55,962 WARNING [139966568441600] (backend-v4l2.cpp:1098) Pixel format 435a4e49-b602-480f-978c-e4e88ae likely requires patch for fourcc code INZC!
 04/04 15:14:55,962 WARNING [139966568441600] (backend-v4l2.cpp:1098) Pixel format 52494150-8536-4841-b6bf-8fc6ffb likely requires patch for fourcc code PAIR!
 04/04 15:14:55,962 WARNING [139966568441600] (sensor.cpp:313) Unregistered Media formats : [ INZC PAIR Y10  YUYV ]; Supported: [ ]
 04/04 15:15:01,180 WARNING [139966568441600] (types.cpp:57) set_pu(id=3) failed! Last Error: Input/output error
 04/04 15:15:01,180 WARNING [139966568441600] (sensor.cpp:691) Exception was thrown when inspecting properties of a sensor
 04/04 15:15:06,271 WARNING [139966568441600] (types.cpp:57) set_pu(id=9) failed! Last Error: Input/output error
 04/04 15:15:06,271 WARNING [139966568441600] (sensor.cpp:691) Exception was thrown when inspecting properties of a sensor
````

Now when in  `realsense-viewer` I enable the RGB camera, I can see the images. But when I try to enable the Coded-Light Depth Sensor, I got additional logs in the terminal:

````
04/04 15:17:11,764 ERROR [139966457829120] (types.h:178) xioctl(VIDIOC_DQBUF) failed Last Error: No such device
 04/04 15:17:11,764 ERROR [139966457829120] (backend-v4l2.cpp:1185) xioctl(VIDIOC_DQBUF) failed Last Error: No such device
 04/04 15:17:11,764 ERROR [139966449436416] (types.h:178) xioctl(VIDIOC_DQBUF) failed Last Error: No such device
 04/04 15:17:11,764 ERROR [139966449436416] (backend-v4l2.cpp:1185) xioctl(VIDIOC_DQBUF) failed Last Error: No such device
 04/04 15:17:15,864 WARNING [139966863592000] (backend-v4l2.cpp:1098) Pixel format 435a4e49-b602-480f-978c-e4e88ae likely requires patch for fourcc code INZC!
 04/04 15:17:15,865 WARNING [139966863592000] (backend-v4l2.cpp:1098) Pixel format 52494150-8536-4841-b6bf-8fc6ffb likely requires patch for fourcc code PAIR!
 04/04 15:17:15,865 WARNING [139966863592000] (sensor.cpp:313) Unregistered Media formats : [ INZC PAIR Y10  YUYV ]; Supported: [ ]
 04/04 15:17:15,872 ERROR [139966863592000] (types.h:178) xioctl(VIDIOC_STREAMOFF) failed Last Error: No such device
 04/04 15:17:15,872 ERROR [139966863592000] (sensor.cpp:238) An error has occurred while stop_streaming()!
 04/04 15:17:15,872 ERROR [139966863592000] (types.h:178) xioctl(VIDIOC_STREAMOFF) failed Last Error: No such device
 04/04 15:17:15,872 ERROR [139966863592000] (sensor.cpp:238) An error has occurred while stop_streaming()!
 04/04 15:17:15,872 WARNING [139966863592000] (archive.cpp:316) The user was holding on to 1 frames after stream 0x0x153a0c0 stopped
 04/04 15:17:15,872 WARNING [139966863592000] (archive.cpp:316) The user was holding on to 1 frames after stream 0x0x153a0c0 stopped
 04/04 15:17:15,872 WARNING [139966863592000] (archive.cpp:326) All frames from stream 0x0x153a0c0 are now released by the user
````
 

### dmesg log

````
$ dmesg | tail -n 50
[  586.338424] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[  586.339124] uvcvideo 3-3.3.1:1.0: Entity type for entity Processing 2 was not initialized!
[  586.339126] uvcvideo 3-3.3.1:1.0: Entity type for entity Camera 1 was not initialized!
[  586.339128] uvcvideo 3-3.3.1:1.0: Entity type for entity Extension 3 was not initialized!
[  586.339210] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.3/3-3.3.1/3-3.3.1:1.0/input/input32
[  586.339628] uvcvideo: Unknown video format 435a4e49-b602-480f-978c-e4e88ae89b89
[  586.339630] uvcvideo: Unknown video format 52494150-8536-4841-b6bf-8fc6ffb083a8
[  586.339728] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[  586.340348] uvcvideo: Unable to create debugfs 3-7 directory.
[  586.340417] uvcvideo 3-3.3.1:1.2: Entity type for entity INV_EU_STRING was not initialized!
[  586.340420] uvcvideo 3-3.3.1:1.2: Entity type for entity Camera 5 was not initialized!
[  586.340544] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.3/3-3.3.1/3-3.3.1:1.2/input/input33
[  749.490334] perf: interrupt took too long (2555 > 2500), lowering kernel.perf_event_max_sample_rate to 78250
````

ev-mp, 2018-04-04:
Hello @fspindle ,
According to the log the patches have been properly set.
But you have a device disconnect event when the camera started streaming:
>4/04 15:17:11,764 ERROR [139966457829120] (types.h:178) **xioctl(VIDIOC_DQBUF)** failed Last Error: No such device
 04/04 15:17:11,764 ERROR [139966457829120] (backend-v4l2.cpp:1185) **xioctl(VIDIOC_DQBUF)** failed Last Error: No such device.

In most cases this error is attributed ti insufficient power provided by the host. (Running a search with `VIDIOC_DQBUF` keyword will collect the relevant cases for review).
Please try the following steps to mitigate this:
1. Switch USB port on Laptop and/or remove additional Power consumers from the specific USB hub .
2. Use AC power when running on Laptop.
3. Use a **self-powered** USB3 hub as a mediator between host and device.

fspindle, 2018-04-04:
Thanks for your feedback. I tryed all what you suggest without success before opening the issue

1. I tried to connect the camera on each USB3 port. My laptop has 4 ports
2. the laptop is connected to an AC power supply
3. I tried also with a self-powered USB3 hub

fspindle, 2018-04-04:
I was using the cable provided with the camera.
Changing the cable makes `rs-capture`working.

Just to be sure before closing this issue, do you confirm that:
- patching the kernel is not mandatory if I'm interested only in color images
- if I want the depth and infrared I need to patch the kernel either with kernel 4.13 or 4.14

ev-mp, 2018-04-05:
Good to hear that it has been resolved!

For SR300 the patches are required in two cases:
1. Kernel version below 4.12.
2. Retrieving frame metadata attributes (hw timestamp, actual exposure, etc;)

With kernel 4.12+ streaming all supported (Depth/IR/Color) sensors is provided out of the box and the patches are not needed.

dorodnic, 2018-04-06:
Just a small correction - hardware timestamp for the SR300 actually does not require a Kernel patch, since the SR300 was encoding timestamp into first two pixels of the frame.

RealSense-Customer-Engineering, 2018-05-02:
[Realsense Customer Engineering Team Comment]
hi @fspindle,

do you need more help on this topic?
thanks.

fspindle, 2018-05-02:
Thanks, I was able to make the device working with an other cable then the one delivered with the camera.

Some suggestions:
- Modify the patch installation process description in order to install the patches only if requested. From what is explained [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md), it sounds that the V4L2 backend patches are requested all the time.
- Could be nice also to extend the installtion process for Ubuntu 18.04.

RealSense-Customer-Engineering, 2018-05-25:
[Realsense Customer Engineering Team Comment]
hi @@fspindle,
thanks for your suggestion, I think we will keep the patch in doc, it's not necessary but make no problem here.
and Ubuntu 18.04 is on the way.
so let me close this ticket here and feel free to open new ticket if any question.

closed 2018-05-25 by RealSense-Customer-Engineering