realsenseai/librealsense #1483: Question) Hello, I want fix this problem about old SDK installation.

kingmugum, 2018-04-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hello, i got trouble on installing Intel RealSense SDK (old version).

Error code: 1601
File name : intel_rs_sdk_runtime_intel64_hand_10_0_26_396.msi

How can i fix it?

RealSense-Customer-Engineering, 2018-04-16:
[Realsense Customer Engineering Team Comment] 
hi @kingmugum,

please refer to this page, https://software.intel.com/en-us/realsense-sdk-windows-eol
old SDK is discontinued, suggest to move on with Realsense SDK 2.0.

some suggestion I have.
please make sure you SR300 connected with Intel Gen6 or above platform with USB3.
please make sure you have DCM installed.
please make sure you have Realsense SDK installed, then install the hand SDK you want to install.
do you have see any other error message?
if you follow above and still not working, please move on SDK2.0

RealSense-Customer-Engineering, 2018-05-02:
[Realsense Customer Engineering Team Comment]
hi @kingmugum,

Do you need more help on this topic?

kingmugum, 2018-05-07:
it's solved. thank you!