realsenseai/librealsense #1484: realsense-viewer Illegal instruction (core dumped)

tutranv, 2018-04-05:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   |  | 
| Operating System & Version |   16.04  | 
| Kernel Version (Linux Only)    |  4.13.0-38-generic                                        | 
| Platform                                 | PC |
| SDK Version                            |  2.10.3                          | 

### Issue Description
I've followed the instruction here to install realsense:
https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md

But whenever trying to run realsense-viewer, I get this error:

> Illegal instruction (core dumped)

Seems it's same to #1464 but I cannot fixed it. Please see attached dump file.
I also try to install v2.10.0 (and v2.10.1) but another error appears:

> realsense-viewer: symbol lookup error: /usr/lib/x86_64-linux-gnu/librealsense2.so.2: undefined symbol: libusb_get_parent

[_usr_bin_realsense-viewer.1000.crash.tar.gz](https://github.com/IntelRealSense/librealsense/files/1878958/_usr_bin_realsense-viewer.1000.crash.tar.gz)

ev-mp, 2018-04-05:
Hello @tutranv ,
Can you please share `ldd realsense-viewer | grep libusb` on your PC ?

tutranv, 2018-04-05:
Here you are:

> ldd /usr/bin/realsense-viewer | grep libusb
> libusb-1.0.so.0 => /usr/local/lib/libusb-1.0.so.0 (0x00007f43e6005000)

P/S: ldd realsense-viewer does not work

ev-mp, 2018-04-05:
Are you building the project from sources, or using debian package ?

It seem that the libusb that is picked up is a privately-build version of the SO file: (/usr/**local**/lib/...)
instead of the expected /lib/x86_64-linux-gnu/....

tutranv, 2018-04-05:
realsense was installed by apt-get

Regarding libusb, I tried both methods:

> sudo apt-get install libusb-1.0-0

Download libusb-1.0.9
> ./configure
> make
> sudo make install

None of them worked

ev-mp, 2018-04-05:
When you install librealsense from debians all the depended packages, including the required libusb version are installed automatically.
You don't need to build it locally, it is not in the installation guide and also creates a conflict.

I'd suggest to remove the locally build version of libusb (or otherwise disable it) and reinstall librealsense packages to realign the system

tutranv, 2018-04-05:
I didn't install libusb manually until I got the error msg.
In other words, if I just install realsense, it did NOT work.
I will tried to remove all (libusb, realsense) then install from the beginning without libusb, then let you know

tutranv, 2018-04-05:
I remove all realsense & libusb and install realsense again from scratch, then it works.
Don't know why?

Anyway, thank you ver much

ev-mp, 2018-04-05:
It could have been the case of older versions of the library being not properly removed , but it is only one of the things that might have go wrong.
Anyway - good to hear that it has been resolved.

Thanks for cooperation.

closed 2018-04-05 by ev-mp

tn0432, 2018-05-27:
Hello,

I encounter similar issue. I also try to remove all realsense & libusb and install again. But still not work! Any suggestion?
Thanks

darknight-007, 2018-08-17:
I had this error after installing another camera drivers (matrixvision bluefox) 

$ ldd realsense-viewer | grep libusb
	libusb-1.0.so.0 => /opt/mvIMPACT_Acquire/lib/x86_64/libusb-1.0.so.0 (0x00007f0f82865000)

upon removing /opt/mvIMPACT_Acquire/lib/x86_64/libusb-1.0.so.0 (specific to my case), things worked again

pjrambo, 2019-02-24:
> I had this error after installing another camera drivers (matrixvision bluefox)
>
> $ ldd realsense-viewer | grep libusb
> libusb-1.0.so.0 => /opt/mvIMPACT_Acquire/lib/x86_64/libusb-1.0.so.0 [...]

@darknight-007 How do you removing /opt/mvIMPACT_Acquire/lib/x86_64/libusb-1.0.so.0 clearly?
It seems that I have met the save issue as you. After I install bluefox driver on my computer, the realsense driver can not works well, If I just remove file as you said. It still can not work. So how can I remove the bluefox driver completly