realsenseai/librealsense #1485: SAVE depth data into file

martijnkusters, 2018-04-05:
I like to save the depth data into a file for the D400 camera.
I work in c++. Is there some example available that I can use.

Best Regards,
Martijn

martijnkusters, 2018-04-09:
I can save the depth and rgb as png files, but what I want to do is to save the Raw depth data in a data file format likes csv or txt.

dorodnic, 2018-04-09:
```cpp
bool save_frame_raw_data(const std::string& filename, rs2::frame frame)
{
    bool ret = false;
    auto image = frame.as<video_frame>();
    if (image)
    {
        std::ofstream outfile(filename.data(), std::ofstream::binary);
        outfile.write(static_cast<const char*>(image.get_data()), image.get_height()*image.get_stride_in_bytes());

        outfile.close();
        ret = true;
    }

    return ret;
}
```

martijnkusters, 2018-04-09:
And is there already code to save this depth data in a matlab format?

GitwellAnyohub, 2018-04-09:
@dorodnic can you briefly explain the code and the context in which it has to be inserted?

dorodnic, 2018-04-10:
Of course. `rs2::frame` is the object SDK uses to reference a single frame. You can add `save_frame_raw_data("1.bin", depth);` to [rs-capture.cpp:32](https://github.com/IntelRealSense/librealsense/blob/master/examples/capture/rs-capture.cpp#L32) to save the depth frame (or change it to color if you want).
Inside the function, we check if the frame indeed comes from video stream (and not some other supported sensor). Then, you can access the underlying pixels byte-array using `image.get_data()` and calculate size of the frame in bytes using `image.get_height()*image.get_stride_in_bytes()`. From there its pretty much copy-paste from stack overflow to write bytes to file.

@martijnkusters I'm not sure what is matlab format, however there must be ways to import raw byte array in matlab. Generating giant CSV with depth values in meters is also not very complicated, please let us know if you need help.

GitwellAnyohub, 2018-04-10:
@dorodnic @martijnkusters Ok so just to tie the knot I'm pasting the code I'm using here. First, I started with the rs.capture.cpp and removed all the opengl nonsense because something isnt working there for me (which is fine I don't need to see the video as it's streaming I can do that in the downstream analysis). When I run the code below, the camera simply streams (I can tell that by the ir):

#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <iostream>

// Capture Example demonstrates how to
// capture depth and color video streams and render them to the screen
int main(int argc, char * argv[]) try
{

    // Declare depth colorizer for pretty visualization of depth data
    rs2::colorizer color_map;

    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    // Start streaming with default recommended configuration
    pipe.start();

    while(true) // Application still alive?
    {
        rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera

        rs2::frame depth = color_map(data.get_depth_frame()); // Find and colorize the depth data
        rs2::frame color = data.get_color_frame();            // Find the color data
    }

    return EXIT_SUCCESS;
}
catch(const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch(const std::exception & e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}

NOW, I want to simply save the depth stream as a binary (little endian is preferred but it doesnt really matter) text file. I want each pixel to be a number that represents distance from the camera. You're saying to do that I need to copy and paste the code above like so:

#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <iostream>

// Capture Example demonstrates how to
// capture depth and color video streams and render them to the screen

bool save_frame_raw_data(const std::string& filename, rs2::frame frame) //this function
	{
   		bool ret = false;
   		auto image = frame.as<video_frame>();
    		if (image)
    		{
       			std::ofstream outfile(filename.data(), std::ofstream::binary);
       			outfile.write(static_cast<const char*>(image.get_data()), image.get_height()*image.get_stride_in_bytes());
			outfile.close();
       			ret = true;
   		 }

   		 return ret;

int main(int argc, char * argv[]) try
{

    // Declare depth colorizer for pretty visualization of depth data
    rs2::colorizer color_map;

    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    // Start streaming with default recommended configuration
[... 49 lines cut ...]
                 from /usr/include/librealsense2/hpp/rs_record_playback.hpp:8,
                 from /usr/include/librealsense2/hpp/rs_context.hpp:8,
                 from /usr/include/librealsense2/rs.hpp:9,
                 from rs-recorddepth.cpp:4:
/usr/include/librealsense2/hpp/rs_frame.hpp:361:11: note: candidate: template<class T> T rs2::frame::as() const
         T as() const
           ^
/usr/include/librealsense2/hpp/rs_frame.hpp:361:11: note:   template argument deduction/substitution failed:
rs-recorddepth.cpp:15:41: error: template argument 1 is invalid
      auto image = frame.as<video_frame>();
                                         ^
rs-recorddepth.cpp:18:33: error: variable ‘std::ofstream outfile’ has initializer but incomplete type
           std::ofstream outfile(filename.data(), std::ofstream::binary);
                                 ^
rs-recorddepth.cpp:18:50: error: incomplete type ‘std::ofstream {aka std::basic_ofstream<char>}’ used in nested name specifier
           std::ofstream outfile(filename.data(), std::ofstream::binary);
                                                  ^
rs-recorddepth.cpp:29:35: error: a function-definition is not allowed here before ‘try’
 int main(int argc, char * argv[]) try
                                   ^
rs-recorddepth.cpp:61:1: error: expected ‘}’ at end of input
 }
 ^

Also thanks for your help. I know I'm a lot of trouble!

dorodnic, 2018-04-10:
I think `using namespace rs2;` would solve the compilation problem. 

To save you some time I modified `rs-capture` to dump depth and colour frames to CSV, you can use the code here: [rs-capture.cpp](https://github.com/dorodnic/librealsense/blob/text_sample/examples/capture/rs-capture.cpp)
It has the extra benefit of converting depth values to meters and colour pixels to RGB. Hope this will do the trick :)

GitwellAnyohub, 2018-04-10:
@dorodnic Thanks so much Sergey!! I can't wait to get to my local machine and try it out! I'll let you know how it goes as soon as possible. @martijnkusters once I have this working I can help you with the Matlab part (if you haven't done it already)

GitwellAnyohub, 2018-04-10:
@dorodnic so something is def working! I was able to get multiple depth AND color csv files. I need to look into it more (I only had 5 minutes at my machine and I ran it quickly) and see why it's breaking it up into multiple files (but this may not be a problem). Thanks so much! 

Here is what it looked like

[image]

dorodnic, 2018-04-11:
I'm appending the frame number to the filename at lines 19 and 44, if you just want to save just the last frame remove the `<< "_" << frame.get_frame_number()`.

martijnkusters, 2018-04-11:
Thank you for the work. I will try the code. Currently I also changed the align example with some lines to capture depth and pointclouds and write it into a txt file to read it with matlab.

Best Regards,
Martijn Kusters

void remove_background(rs2::video_frame& other_frame, const rs2::depth_frame& depth_frame, float depth_scale, float clipping_dist)
{
    const uint16_t* p_depth_frame = reinterpret_cast<const uint16_t*>(depth_frame.get_data());
    uint8_t* p_other_frame = reinterpret_cast<uint8_t*>(const_cast<void*>(other_frame.get_data()));

    int width = other_frame.get_width();
    int height = other_frame.get_height();
    int other_bpp = other_frame.get_bytes_per_pixel();
	rs2::pointcloud pc;
	rs2::points points;
	j = j + 1;
	//float array[];
	ofstream myfile("c:/tmp/depth" + IntToStr(j) +".txt");
	ofstream myfile2("c:/tmp/pointcloud" + IntToStr(j) + ".txt");
	points = pc.calculate(depth_frame);
	const uint16_t* p_cloud_frame = reinterpret_cast<const uint16_t*>(points.get_data());
    #pragma omp parallel for schedule(dynamic) //Using OpenMP to try to parallelise the loop
    for (int y = 0; y < height; y++)
    {
        auto depth_pixel_index = y * width;
        for (int x = 0; x < width; x++, ++depth_pixel_index)
        {
            // Get the depth value of the current pixel
            auto pixels_distance = 1000*depth_scale * p_depth_frame[depth_pixel_index];
			//auto depth = frames.get_depth_frame();
			//points = pc.calculate(depth_frame);

			//auto pointcloudpunt = points;

			//array[depth_pixel_index] = pixels_distance;
			if (myfile.is_open())
			{
				myfile << pixels_distance << " ";
			}

			if (myfile2.is_open())
			{
				myfile2 << p_cloud_frame[depth_pixel_index] << " ";
			}
            // Check if the depth value is invalid (<=0) or greater than the threashold
            if (pixels_distance <= 0.f || pixels_distance > clipping_dist)
            {
                // Calculate the offset in other frame's buffer to current pixel
                auto offset = depth_pixel_index * other_bpp;

                // Set pixel to "background" color (0x999999)
                std::memset(&p_other_frame[offset], 0x99, other_bpp);
            }
        }
    }
	myfile.close();
	myfile2.close();

martijnkusters, 2018-04-11:
Next step is to calibrate the camera.

GitwellAnyohub, 2018-04-12:
@dorodnic  something that i noticed when running the rs-capture.cpp script is that a lot of frames were skipped. Is there anything I can do about that?

GitwellAnyohub, 2018-04-12:
I'm basically only saving ~10% of the frames. Does it take too long to open and write and close a file each time?

dorodnic, 2018-04-12:
There are two simple things you can do:
1. Run `cmake .. -DCMAKE_BUILD_TYPE=Release` before rebuilding the SDK and the examples to enable compiler optimisations 
2. Run it on a partition with faster IO, like SSD

However, saving data to CSV is extremely not efficient by definition. 
We worked hard to offer a decent quality recorder with `*.bag` format. It should be fairly easy to use (just add `config. enable_record_to_file("filename.bag");`) and reasonably optimised. 

You can choose a hybrid approach of recording into `bag` file and then extract CSVs in post-processing. I will be happy to help with the technical details.

GitwellAnyohub, 2018-04-12:
@dorodnic Ok I will try both options. (I really hope the second one works especially since im just using my laptop now). I do have a script for extracting the bagfiles but it’s super slow. In fact, the slowness is one the reason I wanted to try and write to disk. Would it be faster to write a binary file instead? Also, Im not thrilled about doing the bag exteactiin because it requires me to use ROS and thats something I could not download without sudo access. I therefore cannot process it using my universities cluster. Is there a very fast way to extract the bagfiles? If I wantd to do some real time feature detections, would it be too slow to analyze frames in the bag file?

GitwellAnyohub, 2018-04-12:
@dorodnic I was talking with some members of my lab about this issue and it seems like the best solution for us would be to save the depth information as a long string of bytes. If each pixel is 16 bytes, then we can save a really long string of bytes. Since we know the resolution we can do whatever we want afterwards.

GitwellAnyohub, 2018-04-12:
ok im a fool i just needed to change <video_frame> to <rs::video_frame>

GitwellAnyohub, 2018-04-12:
ok so basically I've gotten it to "work". The script:
`

#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API

#include <fstream>
#include <iostream>
#include <sstream>
#include <iomanip>

bool save_frame_raw_data(const std::string& filename, rs2::frame frame)
{
    bool ret = false;
    auto image = frame.as<rs2::video_frame>();
    if (image)
    {
        std::ofstream outfile(filename.data(), std::ofstream::binary);
        outfile.write(static_cast<const char*>(image.get_data()), image.get_height()*image.get_stride_in_bytes());

        outfile.close();
        ret = true;
    }

    return ret;
}

// Capture Example demonstrates how to
// capture depth and color video streams and render them to the screen
int main(int argc, char * argv[]) try
{
    rs2::colorizer color_map;
    
    auto depth_units = rs2::context().query_devices().front()
        .query_sensors().front().get_option(RS2_OPTION_DEPTH_UNITS);
    
    rs2::pipeline pipe;
    rs2::config myconfig; //I am declaring the configuration object
    myconfig.enable_stream(RS2_STREAM_DEPTH, 848, 480, RS2_FORMAT_Z16, 90); //This is the configuration that I want. 
    pipe.start(myconfig);

    while(true) // Application still alive?
    {
        rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera
	rs2::frame depth = data.get_depth_frame(); // Find and colorize the depth data     
        save_frame_raw_data("1.bin", depth);      
    }

    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}
`
Gives me 1 file called 1.bin containing one frame of data. Now I just need it to record all the frames of data. Sorry I'm also using this thread as a living notepad!

RealSense-Customer-Engineering, 2018-04-16:
[Realsense Customer Engineering Team Comment] 
hi @GitwellAnyohub,

thanks for sharing your experience, do you need more help on this topic?
thanks.

RealSense-Customer-Engineering, 2018-05-02:
[Realsense Customer Engineering Team Comment]
closing this ticket here, please feel free to create another new ticket if you still have question.

closed 2018-05-02 by RealSense-Customer-Engineering

ywjia, 2018-09-11:
I have a question about saving data to file.

Is it possible to save frameset as whole, as below:

      rs2::frameset frameset_w = pipe.wait_for_frames();
      save_frameset("1.bin", frameset_w);      

then, in a separate post-processing step, we can simply do the following to retrieve all the data

   rs2::frameset frameset_r = read_frameset("1.bin");
   rs2::video_frame color = frameset_r.get_color_frame();
   rs2::depth_frame depth = frameset_r.get_depth_frame();
 ...
  
Note that this is different from recording, since here the user can choose which framesets to save.

Thanks.

gayanbrahmanage, 2018-10-23:
You can use,
`  Mat M;`
 `M.create(height,width,CV_32FC1);`
Then, save using .exr file type.

Resays, 2019-04-29:
I'm using RealSense 2.20 and task is to capture image with D415 and convert that image to rgb-d data and convert that data into xyz for pointcloud 3D demonstration 
any solution ?

Resays, 2019-04-29:
i'm actually looking to implement this research http://rgbd.cs.princeton.edu/
using realsense and c++
anyone can help ? @dorodnic @gayanbrahmanage

gayanbrahmanage, 2019-04-30:
Follow this example
https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud

Resays, 2019-06-10:
Here is an example to get RGB data from point how to get [XYZ RGB data from RealSense](https://github.com/Resays/xyz_rgb_realsense/tree/master)