realsenseai/librealsense #1487: Basic Question: Simple Streaming of Frames and Saving to Disk

GitwellAnyohub, 2018-04-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

GitwellAnyohub, 2018-04-05:
When it comes to high level programming (python, java, Matlab, etc) I'm relatively new but I can do it. However, I am quite new to Ubuntu. I'm using Ubuntu 16.04 and I simply want to record and save a depth stream. I don't want to use the BAG format (or the real-sense viewer), I just want to record directly from my D415 camera and spit out a file that contains depth values for each pixel for each frame. I have two questions, one about installation and one about implementing the librealsense SDK.

1. How should I install the SDK? Should I install from source or not? I have given up trying to use the python wrapper since I can't simply import a realsense package. There's python2 and python3 and every package I download tries to use their own python and it's different so a package installed in one python doesn't exist in another python and when I tried to use ROS packages to read the "bag" format they had their own python and nothing seems compatible. Also, I needed to be in a specific folder to get the python wrapper to work which would be fine if I knew how to install other packages to that specific version and instance of python. 

2. What are the step by step instructions for implementing these 
https://github.com/IntelRealSense/librealsense/wiki/API-How-To

For instance, this wiki says I need to run this code to stream

rs2::pipeline pipe;
pipe.start();

Where am I running this code?

Am I running this directly in a terminal?
Do I need to copy this into some sort of .txt file and run it with some "header" or "flag" (whatever those mean)

If I want to run a python script called 'script.py' located in my current working directory, I would type this in the terminal
$ python script.py
It wouldn't matter which folder that working directory was because python works everywhere on my computer.

For using the librealsense2 sdk, Do I need to be in a specific directory?
Is this different if I install from source vs not installing from source?

I know this is a lot so thanks in advance to whoever answers my questions. I know the sdk is relatively new, but there just aren't any step by step instructions that make these things completely idiot proof for total beginners. Too many assumptions are made about prior knowledge.  

I'd also like to add that I've tried many things for several weeks and I do have working code, it's just that now I'd like to scale up my project and so I'm trying to do things properly. The realsense viewer isn't feasible for my project. It's still buggy and we need to record for a very long time. Perhaps if it didn't spit out bag files that could ONLY be processed with ROS (another thing I have an issue with since it's not a simple sudo apt install) I'd be more inclined to use it.

TheMikeyR, 2018-04-06:
If you want to use the python wrapper I would make a clone this repository `git clone https://github.com/IntelRealSense/librealsense.git` and then follow this [guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#building-librealsense2-sdk) to compile. 

You need to learn and read about cmake, it is basically an easy way to tell when you compile the program what it should compile e.g. python wrapper. This way you don't get a lot of things you don't need. 

I'm using anaconda (python library, which makes it easy to make new python environments) as you mentioned you have trouble with package x needs python version y etc. With anaconda environments you can easily change the python version you are running and packages you have installed. [More details about environments with anaconda.](https://conda.io/docs/user-guide/tasks/manage-environments.html)

For CMAKE with a custom python environment I'm running this cmake line to get what I need I'm using this line `cmake ../ -DBUILD_EXAMPLES=bool:true -DBUILD_PYTHON_BINDINGS=bool:true -DBUILD_CV_EXAMPLES=bool:true -DPYTHON_EXECUTABLE=/home/ubuntu/anaconda3/envs/realsense/bin/python3.6`, should replace the cmake .. line in the installation notes listed in the top. Else you can read up on how cmake works etc. 

When you have build, maked and installed from the source, you can in the folder `build/wrappers/python/` a file called `pyrealsense2.cpython-36m-x86_64-linux-gnu.so` you need to put this file into your working directory (I believe this work) or you can put it into your python library, since I'm working in a custom environment I'm cd into `~/anaconda3/envs/realsense/lib/python3.6/site-packages` and from there I make a symbolic link to the package I compiled by doing `ln -s /home/ubuntu/librealsense/build/wrappers/python/pyrealsense2.cpython-36m-x86_64-linux-gnu.so ./` this will spawn a file inside my python library and then I can access it from all directories as long as I'm inside my environment. 

You can use the BAG file and play it back in python, the way it works is that the bag file "simulate" a live camera, so you handle it the same way as you will do with a live stream of the camera. In python you just need two extra lines for it to load a bag file compared to streaming from camera.
```
    config = rs.config()
    rs.config.enable_device_from_file(config, "path-to/bag-file.bag")
    pipeline = rs.pipeline()
    config.enable_all_streams()
    pipeline.start(config)
    while True:
        for frame in pipeline.wait_for_frames():
            print(frame.profile.stream_type(), frame.profile.stream_index())
```

dorodnic, 2018-04-08:
Hi @GitwellAnyohub 
The realsense-viewer should be sufficiently stable for what you are describing (or at least as stable as the rest of the code). 
We are also planning to add a dedicated code sample on the topic of BAG recording.

closed 2018-04-21 by GitwellAnyohub