realsenseai/librealsense #1488: Cannot load preset in the quality tool

ValerioBiscione, 2018-04-05:
| Required Info 
|---------------------------------|-------------
| Camera Model                       | {  D435 } | 
| Firmware Version                   | (05.08.05.00 | 
| Operating System & Version |   {Win 10| 
| Platform                                 | PC |

### Issue Description
I cannot load preset within the depth-quality tool as it says "Disabled when Streaming" (see image): 
[image]

Note that I  think I have stopped the streaming by pressing the top right button indicated in the red square in the image. 

Am I doing something wrong? How would I load a preset? 
Thanks

dorodnic, 2018-04-08:
Hi @ValerioBiscione 
You are doing everything right, this is unfortunately a known UI bug we were planning to address very soon.

ValerioBiscione, 2018-04-17:
great thanks