realsenseai/librealsense #1490: Does librealsense work for real-time kernel ?

nathanlem1, 2018-04-06:
The current librealsense works for Linux Kernel 4.4, 4.8 ,4.10 and 4.13. I tested this and its ros-version and it works fine, however, it doesn't work for real-time kernel. Doesn't it support real-time kernel (rts)? I am using SR300 with the current version of librealsense.  

Thanks!

dorodnic, 2018-04-08:
Hi @nathanlem1 
At the moment we are not targeting or testing librealsense with the real-time kernel. This feature exists in our internal backlog, but yet to gain sufficient customer interest. 
Once approved, we most certainly will have to publish new patches, and possibly revisit some of the code under `src/v4l2-backend.cpp`.
Until then feel free to publish any findings you discover on the subject.

dorodnic, 2018-04-18:
Hi @nathanlem1 
Do you have follow-up questions on the subject?

RealSense-Customer-Engineering, 2018-05-11:
[Realsense Customer Engineering Team Comment]
le me closed this ticket here, please feel free to open new ticket if any question.

closed 2018-05-11 by RealSense-Customer-Engineering