realsenseai/librealsense #1491: After update to 2.10.3 everything stopped working. Ubuntu 16.04 (4.13)

manusovich, 2018-04-07:
After update lib to the latest version:
http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 librealsense2-utils amd64 2.10.3-0~realsense0.66 [1,319 kB]
http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 librealsense2-dev amd64 2.10.3-0~realsense0.66 [207 kB]         
http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 librealsense2-dbg amd64 2.10.3-0~realsense0.66 [43.2 MB]        

Everything is crashing during start:
alex@alex:~$ realsense-viewer
Illegal instruction (core dumped)
alex@alex:~$ rs-terminal
Illegal instruction (core dumped)

System info:
alex@alex:~$ uname -a
Linux alex 4.13.0-37-generic #42~16.04.1-Ubuntu SMP Wed Mar 7 16:03:28 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

manusovich, 2018-04-07:
Just in case if you need gdb info
```
alex@alex:~$  gdb realsense-viewer
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from realsense-viewer...Reading symbols from /usr/lib/debug/.build-id/cf/c97ace73a3ab0c840548fdeeabee84e2f4742f.debug...done.
done.
(gdb) run
Starting program: /usr/bin/realsense-viewer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGILL, Illegal instruction.
librealsense::logger_type::logger_type (
    this=0x7ffff7d8fb60 <librealsense::logger>)
    at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/src/log.cpp:81
81	/home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/src/log.cpp: No such file or directory.

```

manusovich, 2018-04-07:
Rolback to 2.10.1-0~realsense0.62 has fixed the issue

dorodnic, 2018-04-08:
Hi @manusovich 
We are looking into the issue and going to publish a fix soon, thank you for reporting it.

dorodnic, 2018-04-18:
Should be resolved in [v2.10.4](https://github.com/IntelRealSense/librealsense/releases/tag/v2.10.4)

closed 2018-04-18 by dorodnic