realsenseai/librealsense #1493: 2.10.3 patch-realsense-ubuntu-xenial.sh fails with

grepteam, 2018-04-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | n/a, still building | 
| Operating System & Version |  Ubuntu 16.04 Xenial  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | Ubuntu Laptop  |
| SDK Version                            | 2.10.3                          | 

### Issue Description

When following https://github.com/IntelRealSense/librealsense/blob/v2.10.3/doc/installation.md#ubuntu-build-dependencies , and I run `./scripts/patch-realsense-ubuntu-xenial.sh`

I get:

```
patching file drivers/media/usb/uvc/uvc_ctrl.c
Prepare kernel modules configuration
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --silentoldconfig Kconfig
  SYSTBL  arch/x86/include/generated/asm/syscalls_32.h
  SYSHDR  arch/x86/include/generated/asm/unistd_32_ia32.h
  SYSHDR  arch/x86/include/generated/asm/unistd_64_x32.h
  SYSTBL  arch/x86/include/generated/asm/syscalls_64.h
  HYPERCALLS arch/x86/include/generated/asm/xen-hypercalls.h
  SYSHDR  arch/x86/include/generated/uapi/asm/unistd_32.h
  SYSHDR  arch/x86/include/generated/uapi/asm/unistd_64.h
  SYSHDR  arch/x86/include/generated/uapi/asm/unistd_x32.h
  HOSTCC  scripts/basic/bin2c
  HOSTCC  arch/x86/tools/relocs_32.o
  HOSTCC  arch/x86/tools/relocs_64.o
  HOSTCC  arch/x86/tools/relocs_common.o
  HOSTLD  arch/x86/tools/relocs
  CHK     include/config/kernel.release
  UPD     include/config/kernel.release
  WRAP    arch/x86/include/generated/asm/clkdev.h
  WRAP    arch/x86/include/generated/asm/dma-contiguous.h
  WRAP    arch/x86/include/generated/asm/early_ioremap.h
  WRAP    arch/x86/include/generated/asm/mcs_spinlock.h
  WRAP    arch/x86/include/generated/asm/mm-arch-hooks.h
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  CC      arch/x86/purgatory/purgatory.o
/bin/bash: ./scripts/ubuntu-retpoline-extract-one: No such file or directory
scripts/Makefile.build:316: recipe for target 'arch/x86/purgatory/purgatory.o' failed
make[2]: *** [arch/x86/purgatory/purgatory.o] Error 127
arch/x86/Makefile:257: recipe for target 'archprepare' failed
make[1]: *** [archprepare] Error 2
Makefile:552: recipe for target '__build_one_by_one' failed
make: *** [__build_one_by_one] Error 2
```

grepteam, 2018-04-08:
@ev-mp thanks for the reply!

Here:
```
Linux greeppypy 4.13.0-38-generic #43~16.04.1-Ubuntu SMP Wed Mar 14 17:48:43 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.4 LTS
Release:	16.04
Codename:	xenial
```

yf225, 2018-04-08:
@grepteam It seems that one way to fix it is to add `cp debian/scripts/retpoline-extract-one scripts/ubuntu-retpoline-extract-one` after https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-realsense-ubuntu-xenial.sh#L85. Normally this copying should be done in one of the build rules of ubuntu-xenial-hwe, but I am not sure why it isn't working in this case.

compiaffe, 2018-04-13:
I encountered the same issue on 

Ubuntu 17.10 with kernel 4.13.0-38-generic

@yf225 's instruction worked here, too.

RealSense-Customer-Engineering, 2018-04-16:
[Realsense Customer Engineering Team Comment] 
hi @grepteam,
would like to know if yf225's instruction works for you too?
thanks.

moorage, 2018-04-17:
@RealSense-Customer-Engineering  I just tried this, and I can confirm @yf225 's workaround worked for me.  Also worked for 2.10.2 which was having the same issue as well.

TheMikeyR, 2018-04-18:
@RealSense-Customer-Engineering @yf225 doesn't work for me, 
I `cp /ubuntu/xenial/hwe/debian/scripts/retpoline-extract-one ./scripts/ubuntu-retpoline-extract-one` and the same error still occur with kernel `4.13.0-38-generic`

dnandha, 2018-04-18:
@yf225 Works for me!

kirillin, 2018-04-18:
@TheMikeyR it works for me, try exactly as @yf225 said.

TheMikeyR, 2018-04-19:
@kirillin Arh, my bad I read it wrong, after adding 
```
cp /ubuntu/xenial/hwe/debian/scripts/retpoline-extract-one ./scripts/ubuntu-retpoline-extract-one
```
to https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-realsense-ubuntu-xenial.sh#L85 as described by @yf225 it also worked for me.

RealSense-Customer-Engineering, 2018-05-11:
[Realsense Customer Engineering Team Comment]
let me close this ticket, and please feel free to open new ticket if any question.

closed 2018-05-11 by RealSense-Customer-Engineering

yf225, 2018-05-13:
@ev-mp , please feel free to do so, thanks!

CapOM, 2019-09-27:
Still happen when trying to compile https://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/bionic/