realsenseai/librealsense #1497: missing RSSDK when building pcl

chiaho903, 2018-04-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | ZR300  | 
| Firmware Version                   | | 
| Operating System & Version |  Linux (Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | NVIDIA Jetson TX2 |
| SDK Version                            |  legacy                        | 

### Issue Description
Hi there, 
I got error when I'm trying to build pcl with RSSDK, it shows cannot find directory for RSSDK. So I change the dir to "/librealsense/build" but then it shows file "/librealsense/build/include/pxcversion.h" cannot be read. However, there doesn't have any file named "pxcversion.h" in the repositories. Where can I find it, or is there anything I can do to make the pcl work right?
Thanks.

dorodnic, 2018-04-09:
Hi @chiaho903 
I know it's confusing but RSSDK in-question is not librealsense, but rather the old RealSense SDK. 
For librealsense we do not yet provide direct integration with PCL, but we do offer a getting started guide [here](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/pcl)

chiaho903, 2018-04-09:
Hi @dorodnic thanks for replying me, but the link you gave is based on librealsense2, which is use for the new cameras (D400 series and the SR300) and my ZR300 cannot run it.
I've searched the old RealSense SDK for ZR300 [here](https://github.com/IntelRealSense/realsense_sdk_zr300), which can build according to [this tutorial](https://software.intel.com/sites/products/realsense/sdk/getting_started.html), but I still cannot find the "pxcversion.h" file.

dorodnic, 2018-04-09:
I see. To the best of my understanding RSSDK parameter only refers to the older version of RealSense Windows SDK. I don't think the ZR300 software offered any official integration with PCL, but I hope the document for librealsense2 can still be useful (since librealsense 1 and 2 are not that different)
If this direction can be relevant for you, we can help adapt librealsense2 code in the pcl snippet.

chiaho903, 2018-04-09:
okay then, I may try to build an integration between librealsense 1 and pcl on my own.  Thanks for your help anyway!

closed 2018-04-09 by chiaho903