realsenseai/librealsense #1500: Jerky framerate in skeleton opencv application

visez, 2018-04-09:
RealSense D415, SDK version 2.1.3, MacOS 10.13.3

I have modified the opencv samples imshow and grabcut in order to show the colorized depth and rgb frames in two opencv windows as shown in the code below. 

However, even with this very simple structure, the stream is quite jerky as if there were frequent frame drops. 

Moreover, if I remove the guard if(!depth) continue; then the program crashes as it receives null depth frames, while I thought this issue was fixed in a previous release.

I also tried to add a queue to the code, but the stream is still jerky. I was not expecting the stream to be this jerky even with no processing involved, am I doing anything wrong?

```
#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <opencv2/opencv.hpp>   // Include OpenCV API

using namespace rs2;
using namespace cv;

cv::Mat frame_to_mat(const rs2::frame& f)
{
    using namespace cv;
    using namespace rs2;
    
    auto vf = f.as<video_frame>();
    const int w = vf.get_width();
    const int h = vf.get_height();
    
    if (f.get_profile().format() == RS2_FORMAT_BGR8)
    {
        return Mat(Size(w, h), CV_8UC3, (void*)f.get_data(), Mat::AUTO_STEP);
    }
    else if (f.get_profile().format() == RS2_FORMAT_RGB8)
    {
        auto r = Mat(Size(w, h), CV_8UC3, (void*)f.get_data(), Mat::AUTO_STEP);
        cv::cvtColor(r, r, CV_BGR2RGB);
        return r;
    }
    else if (f.get_profile().format() == RS2_FORMAT_Z16)
    {
        return Mat(Size(w, h), CV_16UC1, (void*)f.get_data(), Mat::AUTO_STEP);
    }
    else if (f.get_profile().format() == RS2_FORMAT_Y8)
    {
        return Mat(Size(w, h), CV_8UC1, (void*)f.get_data(), Mat::AUTO_STEP);;
    }
    
    throw std::runtime_error("Frame format is not supported yet!");
}

int main(int argc, char * argv[]) try
{
    //Declare image window
    const auto rgbWindow = "RGB";
    const auto depthWindow = "Depth";
    namedWindow(rgbWindow, WINDOW_AUTOSIZE);
    namedWindow(depthWindow, WINDOW_AUTOSIZE);

    // Declare depth colorizer for pretty visualization of depth data
    colorizer color_map;
    color_map.set_option(RS2_OPTION_COLOR_SCHEME, 1);
    color_map.set_option(RS2_OPTION_HISTOGRAM_EQUALIZATION_ENABLED,true);
    align align_to(RS2_STREAM_COLOR);

    // Declare RealSense pipeline, encapsulating the actual device and sensors
    pipeline pipe;
    
    pipe.start();
    
    // Skip initial frames to allow auto-exposure stabilization
    Size rgbSize, depthSize;
    for (int i = 0; i < 20; i++) {
        frameset data = pipe.wait_for_frames();
        frameset aligned_data = align_to.process(data);
        frame rgb = aligned_data.get_color_frame();
        frame depth = aligned_data.get_depth_frame();
        if (!depth)
            continue;
        
        rgbSize.width = rgb.as<video_frame>().get_width();
        rgbSize.height = rgb.as<video_frame>().get_height();
        depthSize.width = depth.as<video_frame>().get_width();
        depthSize.height = depth.as<video_frame>().get_height();
    }

    while (waitKey(1) < 0)
    {
        frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera
        frameset aligned_data = align_to.process(data);
        
        //get depth and rgb frames
        frame depth = aligned_data.get_depth_frame();
        if (!depth)
            continue;
        
        //create opencv matrices
        frame depthColor = color_map(depth);
        Mat depthMat(depthSize, CV_8UC3, (void*)depthColor.get_data(), Mat::AUTO_STEP);
        auto rgbMat = frame_to_mat(aligned_data.get_color_frame());
        
        // Update the window with new data
        imshow(rgbWindow, rgbMat);
        imshow(depthWindow, depthMat);
    }

    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
    }
    catch (const std::exception& e)
    {
        std::cerr << e.what() << std::endl;
        return EXIT_FAILURE;
    }
```

dorodnic, 2018-04-09:
Hi @visez 
First, consider enabling compiler optimization. If you are using XCode, go to `edit scheme` > `Run` > `Build Configuration`. If building using clang, add `-DCMAKE_BUILD_TYPE=Release`. This somewhat improves the situation. (from 120ms to around 70ms for single while loop iteration on my Mac).
Next, I commented out the following lines:
```cpp
Mat depthMat(depthSize, CV_8UC3, (void*)depthColor.get_data(), Mat::AUTO_STEP);
auto rgbMat = frame_to_mat(aligned_data.get_color_frame());
// Update the window with new data
imshow(rgbWindow, rgbMat);
imshow(depthWindow, depthMat);
```
This reduced iteration time to stable 33ms (as expected for default configuration).
I'm not sure if OpenCV version on my laptop itself was compiled with optimizations, but I'd say this is a good direction to take.

visez, 2018-04-10:
Thank you very much!

closed 2018-04-10 by visez