realsenseai/librealsense #1502: Post processing / filters with Python wrapper

TheMikeyR, 2018-04-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Linux (Ubuntu 16.04.4)  | 
| Kernel Version (Linux Only)    |  4.14.13-041413-generic                                 | 
| Platform                                 | PC  |
| SDK Version                            |  v2.10.0                       | 
| Language                            |  Python                       | 

### Issue Description
It seems like the post processing steps / filters are not available to the python wrapper, any plans on porting it?

dorodnic, 2018-04-09:
Hi @TheMikeyR 
Yes, we should add it.

RealSense-Customer-Engineering, 2018-05-08:
[Realsense Customer Engineering Team Comment]
Hello @TheMikeyR, the post-processing via Python API has been added in release v2.11.0.

TheMikeyR, 2018-05-09:
Thanks @RealSense-Customer-Engineering

closed 2018-05-09 by TheMikeyR

phojjat, 2018-08-28:
Hello @TheMikeyR and @RealSense-Customer-Engineering ,

Could you please point me to the v.2.11.0 release, where the filters have been added? I don't seem to be able to find it.

Many thanks!

TheMikeyR, 2018-09-04:
@phojjat https://github.com/IntelRealSense/librealsense/blob/a8e38426bf2bf0fc4233bca054cec32caaf60bc8/wrappers/python/python.cpp#L534-L544

phojjat, 2018-09-04:
This is what I have tried:
aligned_frames = align.process(frames)
decimated=rs.decimation_filter(aligned_frames)

but I get a TypeError: __init__(): incompatible constructor arguments.

lauer2356, 2018-09-18:
@phojjat - did you figure out how to use these filters in python? I am having a similar issue.