realsenseai/librealsense #1506: Duplicated wrong serial number on MacOS

emilk, 2018-04-09:
**My setup**: four 430 cameras connect to a MacBook Pro (Retina, 15-inch, Mid 2015) running macOS High Sierra 10.13.4.

**The bug**: When I list the connected realsense devices I get the serial number '750612070575' repeated four times.

If I check which USB devices are connected to my computer (using other means), I find all four cameras with different serials.

Here is the code I run to list the serials:

```
#include <iostream>
#include <librealsense2/rs.hpp>

int main()
{
    rs2::context ctx;
    for (auto&& device : ctx.query_devices())
    {
        std::cout << "Camera serial: " << device.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER) << std::endl;
    }
}
```

And here is the output:

```
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
Camera serial: 750612070575
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
Camera serial: 750612070575
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
Camera serial: 750612070575
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
Camera serial: 750612070575
```

And here are the actual serial numbers: 

```
§ system_profiler SPUSBDataType | grep "RealSense(TM) 430:" -A 10 | grep "Serial Number:"
              Serial Number: 751313020148
              Serial Number: 734613020958
              Serial Number: 751313020574
              Serial Number: 736613021581
```

emilk, 2018-04-09:
Btw, connecting just _two_ cameras directly into the Macbook yields the same results, but with the same serial repeated twice instead of four times.

dorodnic, 2018-04-12:
I tried to reproduce this issue with various combinations of D435 and D415, with v2.10.3 and v2.10.2 and different cable layouts but was not able to.
What about the RealSense Viewer? Are you able to work with the different devices together using that App? Did you try with the latest update (v2.10.3)?

emilk, 2018-04-12:
RealSense Viewer shows only one camera. I’ve tried v2.10.3 and latest master.

dorodnic, 2018-04-13:
Any other USB devices connected to the machine? Could you share how the cameras are connected?

@EladZucker, do you have any idea what might cause such behavior?

emilk, 2018-04-13:
No other USB devices. In the simplest reproducible setup I just have two cameras plugged in directly to my Macbook in two separate sockets.

fonskuijk, 2018-05-14:
I confirm having the exact same issue as Emil reported

RealSense-Customer-Engineering, 2018-05-14:
[Realsense Customer Engineering Team Comment]
yes, we can reproduce this issue and trying to fix this issue.

emilk, 2018-06-12:
I've reported the USB hub issue as a separate issue at https://github.com/IntelRealSense/librealsense/issues/1875

RealSense-Customer-Engineering, 2018-06-12:
[Realsense Customer Engineering Team Comment]
hi @emilk,

thanks for confirm the fix, so I think we can close this one and continue discussion on#1875.

closed 2018-06-19 by RealSense-Customer-Engineering