realsenseai/librealsense #1508: 410 Depth Module not being recognized

viprav07, 2018-04-09:
| Camera Model                       | D410 depth module + D4 vision processor card | 
| Firmware Version                   | (05.08.15.00) | 
| Operating System & Version |   {Win (10)   |                        
| Platform                                 | PC  |
| SDK Version                            |  2.0     | 

Hello, 

I have the Intel Real sense 410 depth module and the d4 vision processor card. I have connected them both using the 50 pin interposer. I am trying to set up the module on my computer.

For some reason the Intel Real Sense Viewer is not recognizing my device. It is being detected but it shows the following error:

 

Invalid Value in rs2_get_option_range(options:000002B3641140C0, option:Stereo Baseline, min:0.0001, max:0.01, step:1e-06, def:0.001):

OpCodes do not match! Sent 21 but received -6!

 

Can someone please help me out?

dorodnic, 2018-04-10:
Hi @viprav07 
Is [updating camera firmware](https://downloadcenter.intel.com/download/27408/?v=t) an option for you? Alternatively, I would try going back as far as [v2.7.7](https://github.com/IntelRealSense/librealsense/releases/tag/v2.7.7) to check if it helps.

viprav07, 2018-04-10:
-I tried updating the firmware 
It failed.
[image]

- I tried the older SDK. It doesn't even detect the device.
- The latest SDK does detect the device. It shows the error code as mentioned above. 

When I tried the default setting in the realsense viewer it says:
Invalid Value in rs2_open_multiple(sensor:0000024417FD4BD0, profiles:0000024417EF2B68, count:1):
apply_preset(...) failed! Invalid product ID (2774)

dorodnic, 2018-04-10:
Based on the product ID, it seems that the camera was recognized as USB2 device. Could you please make sure it is connected to a super-speed (blue) USB port, check the cable and the connection?

viprav07, 2018-04-10:
@dorodnic Thank you for your reply. 

I connected it using a different cable. I have a USB type C cable and Type C USB 3 port on the pc. I am using this and was able to update the driver. I can see a stream from the camera. But the error still persists and it prints this: 
[image]
[image]

I have two questions:
1) Do I have to use a Type C to Type A USB 3.0 cable and use it in the regular USB 3.0 port on the PC instead of the C to C cable + C port on pc.
If so, what is the logic behind this? 
 
2) Do I have to calibrate the device before using? 

Here is the procedure I followed:
- Updated Windows 10.
- Got the Dfu Tool ; updated the device using the latest firmware 
- Tried both SDK versions, the latest as well as the one you suggested. 

I really appreciate your time helping me.

dorodnic, 2018-04-11:
Hi @viprav07 
The images suggest that the device was not calibrated. Perhaps dynamic calibration software can help, but usually we don't send customers uncalibrated units (I'm referring to stereo calibration required to generate depth). If you got the camera from Intel representative, perhaps you could reach out for more direct assistance? I would love to help but most of the engineers who respond here have less experience with the calibration process. Alternatively, perhaps @RealSense-Customer-Engineering could help.
Regarding the cable, I'm not aware of any limitation that would prevent you from using Type C to Type C cable, but we do most of our tests with Type A to Type C cables.

RealSense-Customer-Engineering, 2018-04-11:
[Realsense Customer Engineering Team Comment] 
hi @viprav07,

could you please provide the serial number of your D410 so I can check the status of you camera, and your windows version please?
thanks.

viprav07, 2018-04-12:
Thank You  @dorodnic .
That's what I thought. For example, the baseline parameter in the setting s of the viewer is showing up as 0mm and it's not letting me change it ( says read only).

@RealSense-Customer-Engineering  The details are as follows:
Windows Version : 14393 ( 1607 ) 
Intel 410:
J32106 100 951913
747512021062
I appreciate your help.

RealSense-Customer-Engineering, 2018-04-12:
[Realsense Customer Engineering Team Comment] 
hi @viprav07,

your D410 is PRQ unit, so it should be calibrated well and should able to update the FW.
so can you please provide your sceenshot of device manager so I can confirm it was detect correctly?
sometimes the type C cable have quality issue, so please check if your cable is USB certified.
also for calibration issue, please make sure the cable is good, then use this link for calibration.
https://downloadcenter.intel.com/download/27415/Intel-RealSense-D400-Series-Dynamic-Calibration-Software-Tools
please let me know how it works there.

viprav07, 2018-04-13:
Thank You @RealSense-Customer-Engineering 
I checked the cable details and it seems to be of a good quality. If you want to verify this please look at the link below:
https://www.bhphotovideo.com/c/product/1202446-REG/griffin_technology_gc41634_usb_type_c_to_usb.html
[image]

I compared with another cable and it is indeed showing USB2 
[image]

Next I tried the Calibration Software: 
[image]

I am waiting for another type c to type a cable to arrive. do you think that might solve the problem?  It says re calibrate using OEM tools. It might not be calibrated?

viprav07, 2018-04-18:
@RealSense-Customer-Engineering Having a type c to type cable didn't solve the issue either.

viprav07, 2018-04-19:
@dorodnic  @RealSense-Customer-Engineering  Can you help me here?

JasonChang13, 2018-04-25:
hi @viprav07,

the cable looks good to me, and this seems calibration table corrupted or version not match from the error message of calibration tool, unfortunately we can't fix it outside of intel.
can you please check with your intel contact window to ship the unit to us so we can fix it and send It back?
thanks.

Regards,
Jason
Intel RSG Customer Platform Engineering

closed 2018-04-25 by viprav07

antodi, 2018-08-03:
Hi all,
I have the exact same problem and I cannot calibrate the D435 using the Dynamic Calibration tool (it stops at step 10, giving me the resolution). I cannot update the firmware either (I have version 05.09.11.00). Is there anyway to fix it? Thanks for your help.