realsenseai/librealsense #1509: Cannot build example code rs-pointcloud.cpp due to example.hpp error

ljc19800331, 2018-04-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    |  4.13.0-38-generic  | 
| Platform                                 | PC|
| SDK Version                            |   | 

### Issue Description
Hello, I have successfully cmake and make the files with librealsense. When I want to build the rs-pointcloud.cpp with the command " g++ -std=c++11 rs-pointcloud.cpp -lrealsense2", it got the following error: rs-pointcloud.cpp:5:93: fatal error: example.hpp: No such file or directory
compilation terminated.

I have no idea how to check if this is a version problem or missing the example.hpp file. Any ideas or suggestions? Thanks.

dorodnic, 2018-04-10:
Well, if you clone the repository you can build all the examples using `-DBUILD_EXAMPLES=true`. 
Your way will also work, but you do have to isolate the dependencies used by our examples. This includes `stb_easy_font`, `glfw` and `imgui` for some examples. Alternatively, you can reduce the example to just SDK calls within console application, and it will also work. 
Alternatively, you can look at the [opencv examples](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv). These examples assume you have `opencv` installation on your system, so they rely on `opencv` for window management and rendering. 
The bottom line is that once you take some example code out of the SDK, it is up to you to figure out everything that is not strictly related to RealSense.

dorodnic, 2018-04-15:
Hi @ljc19800331 
Did you managed to compile the examples?

ljc19800331, 2018-04-16:
Hello@dorodnic , thanks for your help.

I can find the compiled and built-in example code in the Build folder in Librealsense. I am currently learning how to compile from scratch using the example.cpp and the other supporting file. Thanks.

closed 2018-04-16 by ljc19800331

D-aggie, 2019-06-19:
Hello @dorodnic , I am experiencing the same issue but I am having trouble following you solution. I have attempted to clone the example repository again adding the extra lines you have but I believe I am entering it the wrong way. Could you please explicitly show the lines you entered?
Thank you

dwalt123, 2019-10-31:
Hello, I had a similar experience using g++ to compile examples. I tried building the examples with cmake but I get:

RealSense error calling rs2_pipeline_start(pipe:0x558659fba0):
    The requested pixel format 'INZI' is not natively supported by the Linux kernel and likely requires a patch!
Alternatively please upgrade to kernel 4.12 or later. Last Error: Inappropriate ioctl for device

Is there a straightforward way to upgrade to kernel 4.12 or higher?