realsenseai/librealsense #1513: realsense  lack of cpuid.h

xiuruisi, 2018-04-10:
when I deploy the realsense sdk ,terminal  always reminder me that the computer lack of cpuid.h. How can i solve this problem?

dorodnic, 2018-04-10:
Hi @xiuruisi 
Could you please offer more details on the problem? What is the exact error and when does it occur? What version of Linux do you use, what Kernel and what is the underlying architecture?

xiuruisi, 2018-04-11:
@halton @dorodnic Hello. I am a developer of RealSense. When I follow the list in “ https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md ” to install the SDK I encounter some problem. 

 

My version of Linux is Ubuntu 16.04  LTS, version of Kernel is 4.4.55 and another is 4.4.77(I have two arm boards), and the underlying architecture is arm64.

 

The first problem appear when I make the project “/home/nvidia/librealsense/src/image.cpp:23:19: fatal error: cpuid.h: No such file or directory compilation terminated.”

 

And I can’t find the cpuid.h in my file system by myself.

 

The second, when I install the video4Linux, the program error in “cannot find the linux-header-4.4.55”, but I have “linux-header-4.4.0”. Meanwhile I can not find the suitable Linux-header for the video4Linux program in Internet.

xiuruisi, 2018-04-11:
@dorodnic  Could you tell more details about development/image.cpp:14? I don't understand.

dorodnic, 2018-04-11:
Actually I see now that the fix was also merged into latest master.
I think that on your machine the `(defined (__linux__) && !defined (__x86_64__))` will be true, and hence the line `#include <cpuid.h>` should not be compiled. 
Could you try again with the latest code and share if the problem continues?

xiuruisi, 2018-04-11:
@halton  I'am sorry that I want to recovery it so that I can relocation. But now I have a new problem : sudo update-grub (this can't use) ,after I go through the Internet ,I find the solution,but after set grub-commom
,terminal reminder me that "cannot find a GRUB drive for /dev/mmcblklp6, I feel terrible .How can I solve this problem?

xiuruisi, 2018-04-11:
@halton I'am sorry that I want to recovery it so that I can do a relocation. But now I have a new problem : sudo update-grub (this can't use) ,after I go through the Internet ,I find the solution,but after set grub-commom
,terminal reminder me that "cannot find a GRUB drive for /dev/mmcblklp6, I feel terrible .How can I solve this problem?

xiuruisi, 2018-04-11:
@dorodnic  sorry ,the arm has recovered , and product a new problem: sudo update-grub (this can't use) ,after I go through the Internet ,I find the solution,but after set grub-commom
,terminal reminder me that "cannot find a GRUB drive for /dev/mmcblklp6

xiuruisi, 2018-04-11:
@dorodnic  and reminder me  check your device.map

linzihao1999, 2018-04-11:
@dorodnic Hello. I am a student in China and study with @xiuruisi . I will try to explain this problem more detail. Now i am trying to use your latest master for cupid.h missing error and I will reply the latest situation. 

We want to run the RealSense D415 in the arm board, and its hardware is like @xiuruisi  said 

> the version of Linux is Ubuntu 16.04 LTS, version of Kernel is 4.4.55, and the underlying architecture is arm64. 

Earlier this week，i follow the step to the Video4Linux in arm board. When i run 

> ./scripts/patch-realsense-ubuntu-xenial.sh

the error is

> cp cannot stat 'usr/src/Linux-headers-4.4.77/.config': No such file or directory

I install pkg of linux-headers-4.4.0-generic, but i can not find the Linux-headers version consistent with the 'uname -r' (4.4.77) .  
Thank you for helping me to solve this problem.

dorodnic, 2018-04-15:
Hi @xiuruisi @linzihao1999 
The title of this issue is a bit confusing at this point, is there still issue related to the `cpuid`? 
As for your later question, I suspect that Ubuntu Xenial on ARM board would require different headers and we don't support this combination at the moment. Perhaps you could build the SDK with `-DFORCE_LIBUVC=true` CMake flag to bypass Kernel patching? Alternatively, you need to find the exact sources of the Kernel used and apply the patch manually.

RealSense-Customer-Engineering, 2018-04-17:
[Realsense Customer Engineering Team Comment] 
Can this ticket be close since no more cpuid.h issue?