realsenseai/librealsense #1514: Inappropriate ioctl for device

dlemke6, 2018-04-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |   Linux (OpenSuse Tumbleweed)  | 
| Kernel Version (Linux Only)    |  4.16.0-1                                         | 
| Platform                                 | PC |
| SDK Version                            |  2                          | 

### Issue Description
Realsense viewer 2.10.3 and examples won't connect to D435 camera. 

Error message:
 10/04 10:48:52,180 ERROR [139867025590016] (types.h:179) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Inappropriate ioctl for device

 10/04 10:48:52,181 WARNING [139867025590016] (rs.cpp:328) null pointer passed for argument "device"

dorodnic, 2018-04-10:
Hi @dlemke6 
There is indeed a problem running librealsense with 4.16 Kernel at the moment. We are working to fix it, but were hoping we have a bit more time to address it. 
Is downgrading to one of the supported Kernels an option for you? Just to clarify, this would be only a temporary measure until we fix it in librealsense. Long term, 4.16 is going to be the first Kernel that would not require kernel patches for any of the RealSense devices.

dlemke6, 2018-04-27:
Hi @dorodnic 
Sorry for not answering.
i'll try downgrading next week.
Is it possible to determine how long it will take?

Keep up the good work

dorodnic, 2018-04-29:
Hi @dlemke6 
Thank you. 
We are constrained by limited resources, but it is clear that we have to address this issue very soon as the Kernel becomes more "mainstream". 
To the best of my understanding, the new kernel exposes UVC devices with per-frame metadata as two video devices - one for frames and one for metadata. librealsense currently doesn't properly handle the "metadata" video device, hence the crash. As a first measure we will add check to ignore it as a work-around. Next, we will have to modify the library to take advantage of this new kernel feature.

fubsan, 2018-05-30:
Hi, any news on this? Using 4.16.12-1-ARCH I still get

` what():  set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Inappropriate ioctl for device`

RealSense-Customer-Engineering, 2018-05-31:
[Realsense Customer Engineering Team Comment]
@fubsan
The issue has been put into our sprint, should be finished by mid June. Sorry to keep you wait, please understand the limited resource.

fubsan, 2018-07-02:
Also reproduced on 4.17.3-1-ARCH

RealSense-Customer-Engineering, 2018-07-03:
[Realsense Customer Engineering Team Comment]
@fubsan
Thank you for the new information, I've already update it to team.

pumaking, 2018-07-09:
Also hitting this on Arch (4.17.2-1-ARCH). Looking forward to the fix!

RealSense-Customer-Engineering, 2018-07-11:
[Realsense Customer Engineering Team Comment]
@pumaking
Thank you for the update, fed information to the team again.

dorodnic, 2018-07-12:
We are still working on this (there were some complications).
@ev-mp, perhaps you could share some details on current status?

rimio, 2018-07-17:
Still present on 4.17.6-1-ARCH. Looking forward to a fix!

alexswerner, 2018-07-27:
Does any input from the user side help here? This makes the device  currently unusableas for me asother hardware requires kernel version >= 4.16. Do you think you will make it in this sprint?

RealSense-Customer-Engineering, 2018-08-02:
[Realsense Customer Engineering Team Comment]
@alexswerner
The problem is WIP, few items related to this issue has been fixed, but few still in progress. Sorry to keep you wait, but still need some time to finish it.

VileLasagna, 2018-08-16:
Hmmm, we're migrating to OpenSuSE and running 4.17.13 here at work and I was hoping I could hack this into working but, alas, it is no go. We're making do with some old ubuntu docker containers for the moment but that's pretty suboptimal.

Any updates on how this is going?

ev-mp, 2018-08-16:
Hello @VileLasagna , 
The work is being finalized and there is a pending [PR for Kernel 4.16 support](https://github.com/IntelRealSense/librealsense/pull/2252) that is being reviewed.
It includes essential UVC functionalities: enumeration, controls, streaming.
D400-proprietary metadata extensions which will be submitted separately.
You're welcomed to try it and provide a feedback.

In any case kernel 4.16 support will be part of the next week's release.

VileLasagna, 2018-08-17:
So I just gave the code on the PR a go and it seems to be working just fine. Did take a couple of tweaks in the kernel patching for the UVC driver, but pretty easy. Might make a PR with those myself once this is merged. good stuff

ev-mp, 2018-08-28:
Kernel 4.16 support is now part of the latest [v2.16](https://github.com/IntelRealSense/librealsense/releases/tag/v2.16.0) release.
It delivers HW timestamps for all UVC stream types without need of kernel patches.
Note that a [kernel patch](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites) is required for the extended metadata attributes (actual exposure, gain, etc')

dorodnic, 2018-08-30:
We seem to be in good shape with regards to new kernels (except the **last** last missing piece of auxiliary metadata fields, that is still being upstreamed, I was hoping the accepted patch would already include this...)
If you see any problems on 4.16+ please open a new issue.

closed 2018-08-30 by dorodnic

sudeepbanik1296, 2018-12-12:
| Required Info |  |
| ------ | ------ |
| Camera Model |SR300 |
| Operating System & Version | Ubuntu 14.04 |
| Kernel Version  | 4.16.0-1 |
| Platform| PC |
| SDK Version | 1.12.0 |

Issue Description:

BlasterX Senz3D(SR300) camera won't connect.

Error message:
```
terminate called after throwing an instance of 'rs::error'
  what():  VIDIOC_S_CTRL error 25, Inappropriate ioctl for device
Aborted (core dumped)
```