realsenseai/librealsense #1515: Problem with wait_for_frames in syncing two .bag files from two RealSense D415

LeoSarog, 2018-04-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (05.08.15.00) | 
| Operating System & Version |   {Win (10) |                       
| Platform                                 | PC |
| SDK Version                            |  { v2.0 }                          | 

### Issue Description
Hello, 
i saved two .bag files from two different RS depth cameras and i am trying to sync them. By syncing i mean pairing the frames of the first device with some of the second, based on the offset of their timestamps. At first, i retrieve the first frame of the first .bag file (with wait_for_frames) and find its pair with multiple wait_for_frames on the second .bag. The problem is that when i execute the next wait_for_frames for the next frame of the first .bag file, the frame i receive is not the next, but a frame with about 50 frames offset from the first one (checked it from the frame number variable of the frameset). All wait_for_frames are called inside a loop. Shouldn't wait_for_frames return me always the next frame? If yes, is there some depedency with wait_for_frames from different pipelines?
Thanks in advance!!

vladsterz, 2018-04-10:
@LeoSarog 
1)Doesn't wait_for_frames has a timeout? Have you investigated this?
2)Have you tried polling for frames?

LeoSarog, 2018-04-10:
@vladsterz  
1) The timeout is for passing the function if a frame does not arrive inside the timeout, right? The frames arrive normally on time, but they are not the correct ones (i mean the next one).
2) Poll will not return me the next frame for sure, so i don't know if the time offset of the right frame is checked for every iteration. Although, i will try to modify the implementation with poll_for_frames and check it.
Thanks :)

dorodnic, 2018-04-10:
Hi @LeoSarog 
If I understand the code correctly, the playback sensor will give you frames at the same rate as they were recorded. This means that calling `wait_for_frames` would not always give you the next pair. 
One direction I would investigate is pausing one pipeline while waiting for frames from the other. You can get the recorder from pipeline using `pipe.start(cfg).get_device().as<playback>()` to control playback. We are going to publish a dedicated code sample soon on the topic of recording.

LeoSarog, 2018-04-10:
Hi @dorodnic , 
yes, `wait_for_frames` won't give me the next pair, the pair is found through the timestamps (may need multiple `wait_for_frames` and timestamp check to find the next pair). The pause of the pipeline sounds really useful, i'm gonna investigate that! I am looking forward to your code sample.

LeoSarog, 2018-04-10:
`poll_for_frames` and the `pipe.start(cfg).get_device().as<playback>()` worked for me and the .bag files are now syncronized. Thanks a lot for the feedback!

RealSense-Customer-Engineering, 2018-04-16:
[Realsense Customer Engineering Team Comment] 
hi @LeoSarog,

do you need more help on this topic?
thanks.

LeoSarog, 2018-04-17:
Hi @RealSense-Customer-Engineering,
no thanks, the problem has been solved.

closed 2018-04-22 by dorodnic