realsenseai/librealsense #1518: C++ post-processing example not working

TheMikeyR, 2018-04-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Linux (Ubuntu 16.04.4)  | 
| Kernel Version (Linux Only)    |  4.14.13-041413-generic                                 | 
| Platform                                 | PC  |
| SDK Version                            |  v2.10.3                       | 
| Language                            |  C++                       | 

### Issue Description
When I'm running the post-processing example written in c++ I get frame didn't arrive within 5000, I've tried to disconnect and connect the device again. The other examples works fine (I've tested, pointcloud, sensor-control, measure, align. 

All the filters work fine in the realsense-viewer. The code is from the master branch in this repo and I've also tried to modify the code to change resolution on the depth stream and see if that will work out but it didn't. 

```
./rs-post-processing 
terminate called after throwing an instance of 'rs2::error'
  what():  Frame didn't arrived within 5000
[1]    4221 abort (core dumped)  ./rs-post-processing
```

dorodnic, 2018-04-10:
Hi @TheMikeyR 
I wish I could say something more useful, but are you sure? I can run this sample and there isn't anything special about it compared to say, `rs-capture`. Perhaps there are some sporadic connectivity issues with the device?

TheMikeyR, 2018-04-10:
@dorodnic same happens in rs-capture, seems to be connectivity issue as you say (I'm using an 5m active extender). When I activate all streams in realsense-viewer, I get incomplete frame detected when I turn them all on (just one frame), and then it works fine. The issue is probably due to the length of the cable. I guess I need to do some tricks to prevent it from chrashing the program.

closed 2018-04-10 by TheMikeyR