realsenseai/librealsense #1519: Artifacts in 1920x1080 IR image in D415

panjekm, 2018-04-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version | Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC |
| SDK Version                            |  v2.10.1                        | 

### Issue Description
When capturing the full resolution IR image from D415 camera I see some artifacts in the image (attached) but when resolution is set to 1280x720 (highest for depth capture) they do not appear.

[image]

[image]

dorodnic, 2018-04-10:
Hi @panjekm 
Please note that 1920x1080 is internal calibration format and is not intended for normal use. In particular, the images you see are not rectified, unlike the regular left/right Y8 images.

panjekm, 2018-04-10:
Ok. But i guess these artifacts are not connected to rectification. What I meant were those little square pattern bright spots in the white part of the chessboard which only appear on this 1920x1080 image.
These artifacts dont appear on the 1280x720 image. As I tried using those images for calibrating the camera it wasnt possible as the `Intel.Realsense.CustomRW` tool assumes that calibration parameters are for the 1920x1080 image. Setting the lower resolution in calibration.xml file doesnt seem to have any effect.

dorodnic, 2018-04-10:
What I meant is that 1920x1080 Y16 IR and 1280x720 Y8 IR differ in not just scale and the 1920x1080 is not intended for use by end customers. 
Perhaps someone else can answer how the data is aligned within the Y16 frame.

panjekm, 2018-04-10:
Thanks for you answer. 
But if I wanted to calibrate the camera on my own and use the provided tool to upload the calibration parameters, how can I do it? I need to somehow be able to access fullHD IR images to run calibration on them? Or is there some other way?

panjekm, 2018-04-11:
I did some more tests and turns out that if I use 1920x1080 Y8 image these artifacts dont show up. However, as you already mentioned, the Y8 image is probably rectified (see below) while Y16 isnt. This means that if I want to calibrate I need this Y16 unrectified image. It is strange that the artifacts disappear in Y8 image as rectification of Y16 image wouldnt just remove them. 

Also, on the other device (D435) these artifacts dont appear on Y16 raw 1280x800 image. So it seems like its a bug somewhere.

Do you know if these artifacts also appear on the newest firmware?

[image]

[image]

panjekm, 2018-04-11:
I updated the firmware to the latest (5.9.2) and the problem persists. What I found out is that it also appears at lower resolution (640x480) when using Y16 format (see image below).

[image]

This could be related to issue #450. Is there a way to get RAW16 images? The issue seems to be with the debayering. Is debayering done on the device directly?

panjekm, 2018-04-11:
Ok, seems to be related to debayering. I used the `cv::cvtColor` to convert the Y16 IR image into a debayered one.

closed 2018-04-11 by panjekm