realsenseai/librealsense #1521: Install on ubuntu14 : could not insert 'uvcvideo'

dan0910, 2018-04-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  SR300} | 
| Firmware Version                   | ??? | 
| Operating System & Version |   { Linux (Ubuntu 14)   | 
| Kernel Version (Linux Only)    |  (4.4.0-040400)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { ????}                          | 

### Issue Description
I am installing librealsense following the [doc](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)

I installed it step by step according to the instructions until the **4th step** in **Video4Linux backend preparation**: `./scripts/patch-realsense-ubuntu-xenial.sh` .

There came an error: 

> Handle uvcvideo :  Applying the patched module ... modprobe: ERROR: could not insert 'uvcvideo': Invalid argument
> Failed to insert the patched module. Operation is aborted, the original module is restored
> Verify that the current kernel version is aligned to the patched module version
> modprobe: ERROR: could not insert 'uvcvideo': Invalid argument

In the previous installation process, there are two things I'm not sure about. I don't know if this is related to the error.
1.  After updating Ubuntu, I reboot the computer and at Grub2 Boot Menu -> "Advanced Options for Ubuntu" , I selected the top one 'ubuntu Linux 4.4.0-040400', (I think the latest updated kernel version is probably at the top.) While there are some other kernel version  underneath it such as Linux 4.4.0-119xxx and 4.4.0-116xxx...Do I choose the right kernel version?
2.  When it runs `./scripts/patch-realsense-ubuntu-xenial.sh`, there are too many options I don't understand, for example,  Appraise integrity measurements (IMA_APPRAISE) [Y/n/?] 
I press the **Enter key** all the time and I think this is the default option. Is it right to do this ?

RealSense-Customer-Engineering, 2018-04-12:
[Realsense Customer Engineering Team Comment] 
Hi @dan0910,

You can try below instructions to install DKMS package directly without running the patch script. Ubuntu 16.04 is recommended.
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

dan0910, 2018-04-12:
Hi @RealSense-Customer-Engineering ,
Thank you for your reply.

My system is ubuntu14, and the kernel version is 4.4.0-040400. Need I upgrade my system to ubuntu16? (I need ubuntu14 for ros-indigo development…)

I follow the instructions to install DKMS package (ubuntu14) .  But when I run `realsense-viewer` , it comes:

> realsense-viewer: relocation error: /usr/lib/x86_64-linux-gnu/librealsense2.so.2: symbol _ZTVNSt7__cxx1119basic_istringstreamIcSt11char_traitsIcESaIcEEE, version GLIBCXX_3.4.21 not defined in file libstdc++.so.6 with link time reference

RealSense-Customer-Engineering, 2018-04-16:
[Realsense Customer Engineering Team Comment] 
Hi @dan0910,

Per my test configuration: Ubuntu 14.04.5 + Kernel 4.4.0-119 + DKMS package, to build the realsense-viewer app from the source code of librealsense 2.10.3, the app can run normally without your link issue. Please check if the library installed and linked  is correct.

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
Hi @dan0910,
If no further issue, will close this.

closed 2018-05-15 by RealSense-Customer-Engineering