realsenseai/librealsense #1523: Realsense ROS node stops publishing images

maciejjaskowski, 2018-04-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415, D435
| Firmware Version                   | 2.10.2
| Operating System & Version |  Linux (Ubuntu 16.04) | 
| Kernel Version (Linux Only)    |  4.13.0-38-generic, 4.13.0-36-generic | 
| Platform                                 | PC  |
| SDK Version                            |  2.10.2                          | 

### Issue Description
realsense ROS node stops publishing color images (didn't test with depth) on topic.

## Steps to reproduce
1. Start a realsense ROS node (as described in tutorial) publish color images on e.g. /camera1/image_raw
2. Start reading from the topic using e.g. `rostopic echo --noarr /camera1/image_raw`
3. Put your CPU under heavy pressure, by e.g. using ./cpuburn 

Now if you wait a couple of minutes, you will see at some point that there no new messages published on the topic.

Generally, it seems that the more realsense ROS nodes are on, the faster this phenomena occurs. 
The bigger CPU pressure, the faster this phenomena will occur. (I have 4 cores with hyperthreading, I usually put ./cpuburn -n 16)
It seems that this phenomena occurs faster if you keep subscribing to the topic:
`while [ 1 ]; do timeout 1 rostopic echo --noarr /camera1/image_raw`

I couldn't reproduce this on realsense-viewer.
The same phenomena occurs even if I set export OMP_NUM_THREADS=1
The same happens when instead of `rostopic echo` one uses python's rospy library.

lightningsoon, 2018-04-24:
I came across the same problem.
And now my computer seems not to find D415.
I doubt it was broken.

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
Hi @maciejjaskowski,
My test configurations as follows, run about 10 minutes, don't see this issue. How about your firmware version?
--------------------
kernel 4.13.0-41 + ros-wrapper build 2.0.3 + librealsense 2.10.2 and D415 FW 5.9.9

$ roslaunch realsense2_camera rs_camera.launch
subscribe below topics:
$ rostopic echo --noarr /camera/color/image_raw
$ rostopic echo --noarr /camera/depth/image_rect_raw
Stress test CPU:
$ burnP6 & burnP6 & burnP6 & burnP6

RealSense-Customer-Engineering, 2018-06-04:
[Realsense Customer Engineering Team Comment]
Hi @maciejjaskowaski,

any update? still saw this issue with the latest librealsense and firmware (ex: 5.9.11)?

RealSense-Customer-Engineering, 2018-06-19:
[Realsense Customer Engineering Team Comment]
I will close this first. If any concern, you can still reopen. Thanks.

closed 2018-06-19 by RealSense-Customer-Engineering

ryan123rudder, 2025-06-27:
Still having this issue with L515s depth images.

MartyG-RealSense, 2025-06-28:
Hi @ryan123rudder  Please create a new case for your problem by visiting the front page of this support forum at the link below and clicking the green 'New Issue' button.  Thanks!

https://github.com/IntelRealSense/librealsense/issues